zigpy/zha-device-handlers #1694: [BUG] TS0601 dimming turns off light

JoryHogeveen, 2022-08-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on light
2. Try change brightbess
3. Light turns off (not in HA but the switch turns off)

**Expected behavior**
A clear and concise description of what you expected to happen.
This used to work, I honestly don't know what version. I'm 100% sure this is broken in all 2022.8.x versions.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_dfxkcots",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

JoryHogeveen, 2022-08-15:
@javicalle

Thanks for the info at #1554
You mention that the device should turn on when I change the brightness.
This does not happen with my device. It turns on in HA but the device itself stays off.
I will add this to the bug report above.

Joepie007v2, 2022-08-22:
I recently changed from Deconz + Conbee II to ZHA + Conbee II.
I have the exact same problem and read all the topics I could find, but did not find a solution yet. 
With Deconz, I had no issues. I even bought a Tuya Gateway, just to be sure I have the latest firmware. When connected to the Tuya gateway (Tuya cloud integration), I could change the brightness without any problem.

I have no experience with ZHA before HA 2022.8. I also didn’t find other people with this problem until now. 
I would expect there has something changed what is causing this issue. 

I have 3 different Tuya Dimmers (all from brand Moes), and all have the same behavior with turning off the actual light (not the status of the light) when brightness is changed (slider in UI or change in HomeKit for instance).

types:
_TZE200_dfxkcots (I have a Moes Rotary knob dimmer and the Moes touch slider dimmer)
_TZE200_e3oitdyu (Moes 2 gang dimmer / 105B)

I hope there is a solution for this behavior. If I can help with some logs, I need some guidance what to do.

Eddio56, 2022-08-28:
I have a similar issue with dimmer TS0601 by _TZE200_dfxkcots and one month before it was OK. I can't say that updates cause this problem. But now then i try to change brightness, lamp is always turn off. On entity brightness number is changed, lamp brightness stay the same. When I change brightness with dimmer knob, it changed and brightness number on entity changed. I remove and reinstall the device with no effect.

dmulcahey, 2022-08-30:
Can someone with this issue turn on full debug logs and attach an unedited/unfiltered log of the interaction?

lotr, 2022-08-30:
`2022-08-30 17:48:20.794 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: setting transitioning flag to True
2022-08-30 17:48:20.802 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: starting transitioning timer for 1.25
2022-08-30 17:48:20.802 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: turned on: {'on_off': Default_Response(command_id=1, status=<Status.SUCCESS: 0>)}
2022-08-30 17:48:20.858 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: received onoff Bool.true while transitioning - skipping update
2022-08-30 17:48:22.054 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: transition complete - future attribute reports will write HA state
2022-08-30 17:49:02.895 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: setting transitioning flag to True
2022-08-30 17:49:02.914 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: starting transitioning timer for 0.25
2022-08-30 17:49:02.915 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-08-30 17:49:03.001 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: received onoff Bool.false while transitioning - skipping update
2022-08-30 17:49:03.079 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: received level 27 while transitioning - skipping update
2022-08-30 17:49:03.166 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: transition complete - future attribute reports will write HA state
2022-08-30 17:49:25.712 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: setting transitioning flag to True
2022-08-30 17:49:25.722 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: starting transitioning timer for 1.25
2022-08-30 17:49:25.722 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: turned off: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-08-30 17:49:25.778 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: received onoff Bool.false while transitioning - skipping update
2022-08-30 17:49:26.973 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: transition complete - future attribute reports will write HA state`

turned on, waited 4 a minute or something, changed the level, waited 4 another 20 seconds and turned off (the physical light was already off)

dmulcahey, 2022-08-30:
Where are the logs from the radio lib?

dmulcahey, 2022-08-30:
@TheJulianJES `2022-08-30 17:49:25.778 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: received onoff Bool.false while transitioning - skipping update` we may need to tweak how this works…

lotr, 2022-08-30:
```
2022-08-30 17:48:20.590 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x01C0:1:0x0006]: initializing channel: from_cache: False
2022-08-30 17:48:20.591 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x01C0:1:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-08-30 17:48:20.591 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x01C0:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-08-30 17:48:20.591 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x01C0:1:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[False]
2022-08-30 17:48:20.592 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x01C0:1:0x0006]: Reading attributes in chunks: ['on_off']
2022-08-30 17:48:20.593 DEBUG (MainThread) [zigpy.zcl] [0x01C0:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=100, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-08-30 17:48:20.594 DEBUG (MainThread) [zigpy.zcl] [0x01C0:1:0x0006] Sending request: Read_Attributes(attribute_ids=[0])
2022-08-30 17:48:20.595 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 100 under 101 request id, data: b'0064000000'
2022-08-30 17:48:20.595 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 101, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x01C0 endpoint=1>, 260, 6, 1, b'\x00d\x00\x00\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-30 17:48:20.596 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12c3001b001400650002c001010401060001050000640000000200
2022-08-30 17:48:20.598 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12c300090002002265
2022-08-30 17:48:20.599 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 101]
2022-08-30 17:48:20.794 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: setting transitioning flag to True
2022-08-30 17:48:20.794 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0x0006] Setting the NO manufacturer id in command: 1
2022-08-30 17:48:20.794 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0x0006] Sending Tuya Cluster Command... Cluster Command is 1, Arguments are ()
2022-08-30 17:48:20.795 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True, manufacturer=-1)
2022-08-30 17:48:20.795 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] get_dp_mapping --> found DP: 1
2022-08-30 17:48:20.795 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] from_cluster_data: 1, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
2022-08-30 17:48:20.796 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] ztype: Bool.true
2022-08-30 17:48:20.796 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] from_value: [1, 1]
2022-08-30 17:48:20.797 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] raw: b'\x01'
2022-08-30 17:48:20.797 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=102, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))
2022-08-30 17:48:20.798 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=103, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-08-30 17:48:20.800 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=102, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-08-30 17:48:20.801 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB00B:1:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-08-30 17:48:20.802 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: starting transitioning timer for 1.25
2022-08-30 17:48:20.802 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: turned on: {'on_off': Default_Response(command_id=1, status=<Status.SUCCESS: 0>)}
2022-08-30 17:48:20.804 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 103 under 104 request id, data: b'01670000660101000101'
2022-08-30 17:48:20.804 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (25, 104, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0xB00B endpoint=1>, 260, 61184, 1, b'\x01g\x00\x00f\x01\x01\x00\x01\x01', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
[... cut ...]
2022-08-30 17:48:49.055 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB00B](TS0601): Device seen - marking the device available and resetting counter
2022-08-30 17:48:49.056 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB00B](TS0601): Update device availability - device available: True - new availability: True - changed: False
```

Sry for not pasting all the logs, just briefly checking, if I'm not compromising any secrets... I'm not familiar with the full debug logs

lotr, 2022-08-30:
```
2022-08-30 17:48:51.972 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCF7E](TS0001): Device seen - marking the device available and resetting counter
2022-08-30 17:48:51.973 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xCF7E](TS0001): Update device availability - device available: True - new availability: True - changed: False
2022-08-30 17:48:53.953 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x961A](TS0002): Device seen - marking the device available and resetting counter
2022-08-30 17:48:53.953 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x961A](TS0002): Update device availability - device available: True - new availability: True - changed: False
2022-08-30 17:48:55.014 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x8C48](lumi.switch.l1aeu1): last_seen is 2921071.5650992393 seconds ago and ping attempts have been exhausted, marking the device unavailable
2022-08-30 17:48:55.015 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x8C48](lumi.switch.l1aeu1): Update device availability - device available: False - new availability: False - changed: False
2022-08-30 17:48:55.043 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x9E8F](TS0002): Device seen - marking the device available and resetting counter
2022-08-30 17:48:55.043 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x9E8F](TS0002): Update device availability - device available: True - new availability: True - changed: False
2022-08-30 17:49:01.094 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ed0000700aa00
2022-08-30 17:49:01.094 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-08-30 17:49:01.095 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-08-30 17:49:01.095 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17d0000800010001
2022-08-30 17:49:01.098 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17d0002d0026002202000001028f9e01040100000f0008e00a01002050e2ff2030e4ff200000af5ff36d1b00ae
2022-08-30 17:49:01.099 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [38, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x9e8f>, 1, 260, 0, b'\x08\xe0\n\x01\x00 P\xe2\xff 0\xe4\xff \x00', 0, 175, 95, 243, 109, 27, 0, -82]
2022-08-30 17:49:01.099 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Received ZCL frame: b'\x08\xe0\n\x01\x00 P\xe2\xff 0\xe4\xff \x00'
2022-08-30 17:49:01.101 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=224, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-30 17:49:01.103 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=80)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=48)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])
2022-08-30 17:49:01.105 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Received command 0x0A (TSN 224): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=80)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=48)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])
2022-08-30 17:49:01.105 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Attribute report received: app_version=80, 0xFFE2=48, 0xFFE4=0
2022-08-30 17:49:01.107 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=224, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-08-30 17:49:01.107 DEBUG (MainThread) [zigpy.zcl] [0x9E8F:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-08-30 17:49:01.108 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 224 under 109 request id, data: b'18e00b0a00'
2022-08-30 17:49:01.109 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x9e8f>, ep: 1, profile: 0x0104, cluster_id: 0x0000, data: b'08e00a01002050e2ff2030e4ff2000'
2022-08-30 17:49:01.109 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 109, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x9E8F endpoint=1>, 260, 0, 1, b'\x18\xe0\x0b\n\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-08-30 17:49:01.109 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12d1001b0014006d00028f9e010401000001050018e00b0a000200
2022-08-30 17:49:01.112 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12d10009000200226d
[... cut ...]
2022-08-30 17:49:25.797 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 124: 00
2022-08-30 17:49:25.798 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x7c 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xb00b endpoint=1>, status: 0x00
2022-08-30 17:49:26.973 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: transition complete - future attribute reports will write HA state
```

@dmulcahey rest of the logs 4 the timewindow, since somehow I didn't paste it all

gonzalezjj, 2022-09-03:
Hello everyone. I recently updated from HA 2022.04 to .08 and had the same issue with my dimmers when setting the level. I could fix it by reverting one change from pull request #1489.
The pull request changed zhaquirks/tuya/mcu/__init__.py and introduced a supposed handler (lines 400-413) for the move_to_level_with_on_off command. This handler expects a boolean parameter at args[1] containing whether the light should be on or off. I believe this is a misinterpretation of the actual function of the move_to_level_with_on_off command. According to the zcl specs, as I understand them, the command takes the same arguments as the non "with_on_off" command. It should then check whether the level is below the "min_level" of the dimmer and set on_off accordingly. It will never receive a boolean parameter if called according to the specs.
Since I'm fairly new to HA/zigpy and don't know the process to submit a pull request, I was wondering if someone could submit the revert for those lines.
Thanks.

shanelord01, 2022-09-12:
> move_to_level_with_on_off

Is this the section you mean? I'm not sure how to change it as this issue is driving me crazy and no-one is fixing it...

```
       # (move_to_level_with_on_off --> send the on_off command first)
        if command_id == 0x0004:
            cluster_data = TuyaClusterData(
                endpoint_id=self.endpoint.endpoint_id,
                cluster_attr="on_off",
                attr_value=args[1],
                expect_reply=expect_reply,
                manufacturer=manufacturer,
            )
            self.endpoint.device.command_bus.listener_event(
                TUYA_MCU_COMMAND,
                cluster_data,
            )
```

gonzalezjj, 2022-09-12:
> Is this the section you mean? I'm not sure how to change it as this issue is driving me crazy and no-one is fixing it...
>
> ```
>        # (move_to_level_with_on_off --> send the on_off command [...]

Yes, that's the section. After posting my reply, I realized that HA actually expects the on_off command to be sent and behaves weirdly if not. It assumes the light turns on and off based on the level, and does not send explicit on_off commands when using scenes, for example. **So I ended up leaving the code in and changing a single line**:
from:
>                 attr_value=args[1],
to:
>                 attr_value=args[0] > 0,

Basically, this fully implements the move_to_level_with_on_off assuming the device's min_level is 0. From my understanding, for a correct implementation, the min_level would need to be fetched from the device, but this is probably good enough for most use cases and fixes the current issue.

javicalle, 2022-09-12:
Ummm, I believe that this is a misunderstanding of the command ARGS meaning.
It seems that as @gonzalezjj points, the on_off command is detected from the brightness value.

`args[1]` it's the transition time and nothing to do with the on_off value

I will check it later at home, but I believe that the proposed code can do the job.

javicalle, 2022-09-12:
And here is the PR #1748

It will restore the 'normal' behavior, but...
The quirk that was intended to invoke the `on/off` command on non-compliant devices (most Tuya devices, I'd bet) should be implemented in HA by adding the device `manufacturer` value to the list:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L889-L897

javicalle, 2022-09-12:
In case someone wants to test the new version before next release, you can enable the local quirks and copy the file from here:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/mcu/__init__.py

lotr, 2022-09-13:
just a tiny bug remains - might be unrelated to this one and moved to a separate one - then the dimmer is turned off and I move the knob, the GUI shows, that the light turned on, but the physical light stays off

javicalle, 2022-09-13:
> just a tiny bug remains - might be unrelated to this one and moved to a separate one - then the dimmer is turned off and I move the knob, the GUI shows, that the light turned on, but the physical light stays off

Just to be sure, when you says "I move the knob" are you refering to the _knob of the physical device_ (and not from the GUI) right?
So the device has some separate mechanism to turn it on, maybe by pressing the knob?

Can you attach the HA logs from 'moving the knob' from the physical device?

gonzalezjj, 2022-09-13:
I'm pretty sure @lotr is describing the same behavior that I described as "HA behaves weirdly" (which, admittedly, is not very descriptive...). HA expects the move_to_level_with_on_off to be correctly implemented, so when the light is off and you increase the level using the slider in HA, HA automatically reports the light as on, since it has sent the move_to_level_with_on_off command. With the new code change, the light does not turn on, so there is a discrepancy between the actual state and what HA reports.

lotr, 2022-09-13:
@javicalle I mean the GUI knob, I'll attach the logs later today :)

shanelord01, 2022-09-13:
> In case someone wants to test the new version before next release, you can enable the local quirks and copy the file from here:
> 
> * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/mcu/__init__.py

How do I do that? Do I just add that file to the /config/zha_quirks folder and ensure this is in HA configuration.yaml?

zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks

lotr, 2022-09-13:
@shanelord01 what I've done, I've checked the pull request which contained only a single file, so I changed the file in the python venv lib folder

shanelord01, 2022-09-13:
> @shanelord01 what I've done, I've checked the pull request which contained only a single file, so I changed the file in the python venv lib folder

How do I access that?

lotr, 2022-09-13:
@shanelord01 depends on your installation - you have to find the exact file in the HA installation folder, try to grep it for example :)

shanelord01, 2022-09-13:
> @shanelord01 depends on your installation - you have to find the exact file in the HA installation folder, try to grep it for example :)

So I guess I'm screwed as I'm using Home Assistant OS... :(

lotr, 2022-09-13:
Here's the log. The physical device was off, the device in the GUI was off. I've moved the GUI knob from the most left position (min position) to some value. The physical device did not turn on, but the GUI was showing the device as it was turned on.

```
2022-09-13 16:58:01.414 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: setting transitioning flag to True
2022-09-13 16:58:01.415 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0x0008] Setting the NO manufacturer id in command: 4
2022-09-13 16:58:01.415 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0x0008] Sending Tuya Cluster Command. Cluster Command is 4, Arguments are (150, 0)
2022-09-13 16:58:01.416 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='current_level', attr_value=150, expect_reply=True, manufacturer=-1)
2022-09-13 16:58:01.417 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] get_dp_mapping --> found DP: 2
2022-09-13 16:58:01.417 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] from_cluster_data: 2, DPToAttributeMapping(ep_attribute='level', attribute_name='current_level', dp_type=<TuyaDPType.VALUE: 2>, converter=<function TuyaLevelControlManufCluster.<lambda> at 0xffff48732440>, dp_converter=<function TuyaLevelControlManufCluster.<lambda> at 0xffff487324d0>, endpoint_id=None)
2022-09-13 16:58:01.418 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] converted: 588
2022-09-13 16:58:01.418 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] ztype: 588
2022-09-13 16:58:01.418 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] from_value: [4, 0, 0, 2, 76]
2022-09-13 16:58:01.419 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] raw: b'\x00\x00\x02L'
2022-09-13 16:58:01.419 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=107, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x02L', *payload=1275199488))
2022-09-13 16:58:01.424 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=108, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-09-13 16:58:01.425 DEBUG (MainThread) [zigpy.zcl] [0xB00B:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=107, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x02L', *payload=1275199488)))
2022-09-13 16:58:01.426 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB00B:1:0x0008]: executed 'move_to_level_with_on_off' command with args: '(150, 0)' kwargs: '{}' result: Default_Response(command_id=4, status=<Status.SUCCESS: 0>)
2022-09-13 16:58:01.426 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: starting transitioning timer for 0.25
2022-09-13 16:58:01.427 DEBUG (MainThread) [homeassistant.components.zha.entity] light.livingroom_switch_light: turned on: {'move_to_level_with_on_off': Default_Response(command_id=4, status=<Status.SUCCESS: 0>)}
2022-09-13 16:58:01.428 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 108 under 109 request id, data: b'016c00006b020200040000024c'
2022-09-13 16:58:01.428 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (28, 109, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0xB00B endpoint=1>, 260, 61184, 1, b'\x01l\x00\x00k\x02\x02\x00\x04\x00\x00\x02L', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-09-13 16:58:01.429 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12c80023001c006d00020bb001040100ef010d00016c00006b020200040000024c0200
2022-09-13 16:58:01.439 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12c80009000200226d
2022-09-13 16:58:01.439 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 109]
2022-09-13 16:58:01.448 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ec9000700a600
2022-09-13 16:58:01.449 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-09-13 16:58:01.449 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-09-13 16:58:01.449 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04c90007000000
2022-09-13 16:58:01.453 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04c90013000c00226d020bb001010000000000
2022-09-13 16:58:01.453 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 109: 00
2022-09-13 16:58:01.454 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x6d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xb00b endpoint=1>, status: 0x00
2022-09-13 16:58:01.457 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0eca000700aa00
2022-09-13 16:58:01.458 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-09-13 16:58:01.459 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...]
2022-09-13 16:58:22.261 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 114, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=2 address=0x8191 endpoint=1>, 260, 0, 1, b'\x00q\x00\x04\x00', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2022-09-13 16:58:22.263 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12d4001b0014007200029181010401000001050000710004000200
2022-09-13 16:58:22.266 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12d400090002002272
2022-09-13 16:58:22.267 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 114]
```

javicalle, 2022-09-13:
> How do I do that? Do I just add that file to the /config/zha_quirks folder and ensure this is in HA configuration.yaml?
> 
> ```
> zha:
>   enable_quirks: true
>   custom_quirks_path: /config/zha_quirks
> ```
> 
> that doesn't seem to work...

Ummmm, I believe that would be a little more complex...

You will need to copy also the `ts0601_dimmer.py` file (in the same folder):
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_dimmer.py

And adapt (in the ts0601_dimmer.py file) all the `tuya/mcu/*` imports like this:
```python
from zhaquirks.tuya import NoManufacturerCluster, TuyaDimmerSwitch
from custom_zha_quirks import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
)
```

closed 2022-09-25 by dmulcahey

javicalle, 2022-09-25:
The fixed command probably will be included in the next HA release.
Remember to delete the local quirks once updated.

anttix, 2022-10-17:
I confirm my dimmers work correctly after upgrading to HA Core 2022.10.4

PR #1754 (2022-09-25): Add the `on_off` call in the `move_to_level_with_on_off` command
Implement the `on_off` call in the `move_to_level_with_on_off` command for the MCU Tuya devices. 
The previous implementation was removed in #1748. Originally introduced in #1489

After some discussion in https://github.com/zigpy/zha-device-handlers/pull/1748#issuecomment-1244931971 I really believe that the implementation must be done inside the handler library better than expect to add all the devices in the ZHA implementation. This way the handlers library would fully implement the `move_to_level_with_on_off` for the MCU devic [... cut]