zigpy/zha-device-handlers #1695: Unsupported Device Sylvania Lightify 72923 Motion and Temperature Sensor

lightning200mph, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Add device support for "Sylvania Lightify 72923 Motion and Temperature Sensor"

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0406",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "unk_manufacturer",
  "model": "unk_model",
  "class": "zigpy.device.Device"
}

```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
[... 344 lines cut ...]
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-08-17:
Have you tried to pair the device again? Try to keep the device awake during the pairing process just short pressing the pair button every second or so.

lightning200mph, 2022-08-17:
THANK YOU for that info!

It was successfully configured as "Motion Sensor-A by CentraLite" and it appears to be displaying everything correctly.

closed 2022-08-17 by lightning200mph