zigpy/zha-device-handlers #1696: [Device Support Request] CH10AX/SWITCH/1 by Schneider Electric

meremortals70, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
Device is Zigbee 3.0 certified, joined to network however doesn't include any controls only identify function.

**Describe the solution you'd like**
Please add a quick to add on/off/toggle. If other functions are available please add.

<details>

<summary>{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0b05",
        "0xff17"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0102"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Schneider Electric",
  "model": "CH10AX/SWITCH/1",
  "class": "zigpy.device.Device"
}</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
[... 467 lines cut ...]
      }
    }
  }
}</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

axellebot, 2022-08-26:
Can you please provide output file when calling `zha_toolkit.scan_device` service by installing [ZHA Toolkit](https://github.com/mdeweerd/zha-toolkit).

NB : Because of https://github.com/mdeweerd/zha-toolkit/issues/83, you will need to create `scans/CH10AX/SWITCH` or ` scans/CHAX/SWITCH` directories first.

alekgee, 2022-08-29:
{
  "ieee": "8c:f6:81:ff:fe:7b:00:00",
  "nwk": "0x26ce",
  "model": "CH10AX/SWITCH/1",
  "manufacturer": "Schneider Electric",
  "manufacturer_id": "0x4190",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0100",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "Basic",
          "name": "basic",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "zcl_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 3
            },
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "app_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 2
            },
            "0x0002": {
              "attribute_id": "0x0002",
              "attribute_name": "stack_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 6
            },
            "0x0003": {
              "attribute_id": "0x0003",
              "attribute_name": "hw_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 1
            },
            "0x0004": {
              "attribute_id": "0x0004",
              "attribute_name": "manufacturer",
              "value_type": [
                "0x42",
[... 753 lines cut ...]
            }
          },
          "commands_received": {},
          "commands_generated": {
            "0x01": {
              "command_id": "0x01",
              "command_name": "1",
              "command_args": "'not_in_zcl'"
            },
            "0x03": {
              "command_id": "0x03",
              "command_name": "3",
              "command_args": "'not_in_zcl'"
            },
            "0x06": {
              "command_id": "0x06",
              "command_name": "6",
              "command_args": "'not_in_zcl'"
            }
          }
        }
      }
    }
  ]
}``

alekgee, 2022-08-29:
Is detecting the domain type as light, not a switch

axellebot, 2022-08-30:
In order to guess how attributes affect device behavior, can you use `zha_toolkit.execute` with attributes from `OnOff` cluster and send us a feedback here ?

alekgee, 2022-08-30:
@axellebot could you share command and the options you like me to select when running `zha_toolkit.execute`

meremortals70, 2022-08-31:
Sorry I cannot at the moment due to the issue of not having any light controls with ZHA.

closed 2023-07-10 by github-actions[bot]