zigpy/zha-device-handlers #1697: [Device Support Request]

vbelloir, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
I got a [Moes Smart Button](https://www.moeshouse.com/products/ip55-waterproof-smart-zigbee-scene-switch-wireless-remote-dimmer?_pos=87&_sid=426664fbd&_ss=r). I can register it with ZHA, but I can create event for press, double press or long press.

I don't really understant the device reference.  
It reports it-self like a TS004F _TZ3000_kjfzuycl. TS004F is the reference, and TZ3000_kjfzuycl a kind of manufacturer ID?

There is already a quirk for [TS004F](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts004f.py), but it seems to be a 4 gang switch? with different manufacturer?

Is it really the same product, or do I have to try to make a news TS004F quirk?

**Describe the solution you'd like**
A would like to catch event on press, double press and long press, in order to make automations.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000",
        "0xe001"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_kjfzuycl",
  "model": "TS004F",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
I don't have any diagnostic button.
```
[image]
</details>

<details>
<summary>Additional logs</summary>

**with a 10s long press**
```
_TZ3000_kjfzuycl TS004F Stop event was fired with parameters: {'options_mask': None, 'options_override': None}
17:05:42 - Maintenant
_TZ3000_kjfzuycl TS004F Step event was fired with parameters: {'step_mode': <StepMode.Up: 0>, 'step_size': 85, 'transition_time': 10, 'options_mask': None, 'options_override': None}
17:05:41 - Maintenant
```

**with a single press**
```
_TZ3000_kjfzuycl TS004F Move To Level event was fired with parameters: {'level': 255, 'transition_time': 1, 'options_mask': None, 'options_override': None}
17:06:22 - Il y a 2 minutes
```

**with a double press**
```
_TZ3000_kjfzuycl TS004F 92aa42fe on_of  
éteint
Add any other context or screenshots about the feature request here.
17:06:16 - Il y a 2 minutes
```
</details>

MattWestb, 2022-08-17:
Try configure local quirk in HA and adding your device model info in this block:
https://github.com/zigpy/zha-device-handlers/blob/4fcf1c1cedc288f0fc80f3d7e4d308eb1c43cb13/zhaquirks/tuya/ts004f.py#L73-L77.
Restart HA and see if ZHA is loading the quirk for the device.

tuya was doing the dimmer switch with also "scene" support and the rotating knobs is using the same functions and is one dimmer for lights and can changing mode sending events for knob pressing and rotating but they is spamming new device model if this devices.

vbelloir, 2022-08-18:
OK, thanks for your answer.
Here is what I made:
- create a directory `config/zha_quirks.`
- copy/paste `ts004f.py` file with new line  ("_TZ3000_kjfzuycl", "TS004F"),
- edit `configuration.yam`l file with:
```
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
```

Then, I restarted HA.

In the device settings, There is no new entity or sensors:

[image]

Do I missed something?

MattWestb, 2022-08-18:
Looks you have doing all things OK !!

Look on the device signature (device card, device info, three dotc ( . . . ) Zigbee device signature). If "class": "zigpy.device.Device" then the quirk is not loaded, if getting TS004F and somthing its loaded.

If not loaded look if the device signature is the same as your device is having and fixing it in the quirk (normally tuya is adding strange / no standard cluster on endpoints).
Restart HA and look if the quirk is loaded (Zigbee device signature)
.
Zigbee controllers is sending events and you is getting them by listening to `zha_events` in development tools and if the quirk is loaded you is getting device automatons for the device.

vbelloir, 2022-08-18:
Here is the signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x1000",
        "0xe001"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_kjfzuycl",
  "model": "TS004F",
  "class": "zigpy.device.Device"
}
```

Looks like the quirk is not loaded, and the signature is the same as previously.  

> If not loaded look if the device signature is the same as your device is having and fixing it in the quirk (normally tuya is adding strange / no standard cluster on endpoints).
I don't really understand what you mean. Can you clarify?

> 
Zigbee controllers is sending events and you is getting them by listening to zha_events in development tools and if the quirk is loaded you is getting device automatons for the device.

I began to play with automation zha_event, and got a automation catchinh `move_to_level` event, and it works, but there is no event for double press or long press.

vbelloir, 2022-08-18:
There are several things I don't understand.
As far as I can see, the entity `switch.tz3000_kjfzuycl_ts004f_92aa42fe_on_off` toggles on double press, but there is no zha_event when I listen them "development tools > events > listen".

A single press doesn't affect the on_off entity, but generates a zha_event with move_to_level command.

A long press generates a lot of event with the following commands:
```
 egrep -r command Documents/zha_event.txt
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "step",
        "command": "checkin",
        "command": "move_to_level",
```

Here is the content of the event with `step` command:
```
        "command": "step",
        "args": [
            0,
            85,
            10
        ],
        "params": {
            "step_mode": 0,
            "step_size": 85,
            "transition_time": 10,
            "options_mask": null,
            "options_override": null
        }
```
Does it help?

vbelloir, 2022-08-18:
I also found that this switch as two modes `Dimmer switch` and `scene`.   
https://github.com/Koenkk/zigbee2mqtt/issues/12507

In fact, this is the `scene` mode I want to use.

Do you know how to change mode?

vbelloir, 2022-08-18:
OK, thanks.

I remove my ts004f file and replace it with your ROK2.py file.
Then I restarted HA.

The on_off siwth is still there, and the signature is still the same.  
[image]

I have a doubt. in zha_quirks directory, I only have the ROK2.py file. 
Should I have more of them?  https://github.com/zigpy/zha-device-handlers

MattWestb, 2022-08-18:
Put   ` zhaquirks: debug` in you HA configuration.yaml and also `homeassistant.components.zha: debug` is needed for catching more info then the quirk is loaded and testing commands (original ROK is sending toggle your is sending On and Off).

You shall getting somthing like this then HA is starting after all ZHA internal quirks have loading:
```
2022-08-13 07:37:49.665 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/custom_zha_quirks
2022-08-13 07:37:49.810 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts0601_trv_siterwell
```
And then ZHA is loading the device information you is also getting log for is if it was OK or failed.

vbelloir, 2022-08-18:
OK, I added this to my `configuration.yaml`.
```
logger:
  default: info
  logs:
    zhaquirks: debug
    homeassistant.components.zha: debug
```

Here is the `home-assistant.log` during HA startup.

```
cat homeassistant/home-assistant.log | grep -E "zhaquirks|homeassistant.components.zha"

2022-08-18 21:50:45 DEBUG (MainThread) [zhaquirks] Loading custom quirks from /config/zha_quirks
2022-08-18 21:50:45 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ROK2
....
2022-08-18 21:50:48 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0xD2EE](_TZ3000_kjfzuycl TS004F) restored as 'available', last seen: 1:13:44 ago, consider_unavailable_time: 21600 seconds
....
2022-08-18 21:50:49 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD2EE](TS004F): power source: Battery or Unknown
2022-08-18 21:50:49 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD2EE](TS004F): completed initialization

```

vbelloir, 2022-08-18:
In your ROK2.py file, I can't see my device, I added it 

Here is the complete result of previous command
[ha.log](https://github.com/zigpy/zha-device-handlers/files/9377399/ha.log)
:

MattWestb, 2022-08-19:
I was missing one debug setting and you need having `zigpy: debug` in HA.
Then is shall showing the quirk loading for devices like this:
```
2022-08-19 05:59:15.638 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden TRADFRI wireless dimmer (00:0b:57:ff:fe:8a:68:e9)
2022-08-19 05:59:15.638 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.dimmer.IkeaDimmer'>
2022-08-19 05:59:15.639 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:0b:57:ff:fe:8a:68:e9: <class 'zhaquirks.ikea.dimmer.IkeaDimmer'>
```
Your new device class is at line 189 `class TuyaSmartRemote004FROK2(CustomDevice):` and the "old" is at 71 `class TuyaSmartRemote004FROK1(CustomDevice):` so only changing in the "ROK2" class.

You looks good on reading code (im not) so the principle is that information in the device class signature **must** match the device or the quirk is not being loaded for the device. You can double check i have doing all right and not making typos in you new class (need little hex <-> dec conversions).

vbelloir, 2022-08-19:
Hi @MattWestb ,
thanks again for your support!

Sorry, I did not see that there were two classes in the file.
I have deleted the added line, activated zigpy debug logs.

Here is what I get :
```
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3000_kjfzuycl TS004F (38:5b:44:ff:fe:42:aa:92)
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ROK2.TuyaSmartRemote004FROK2'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ROK2.TuyaSmartRemote004F'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts004f.TuyaSmartRemote004F'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1}

```

Now, we can see that errors are located in endpoints and clusters.
It tries several call devices with errors, but we can't know what.

Device signature1 shows only one endpoint, with 7 input clusters, and 7 output clusters, but your `TuyaSmartRemote004FROK2` also.
```
        MODELS_INFO: [
            ("_TZ3000_kjfzuycl", "TS004F"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBExternalSwitchTypeCluster,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    Time.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },

```
For this class, you selected a DIMMER SWITCH. According to [tuya documentation](https://developer.tuya.com/en/docs/iot-device-dev/tuya-zigbee-lighting-dimmer-swith-access-standard?id=K9ik6zvlvbqyw) it seems ok.

But what I don't understant cluster numbers. In this documentation, it should have been 5 input clusters, and 8 output clusters.

Is there something I miss?

vbelloir, 2022-08-19:
OK, I did not noticed that for each class there is an original and a replacement definition.

On original, there are 7 input cluster and 7 output clusters. but for replacement one : **6** input and 8 .

I commented the line
`           #        LightLink.cluster_id,`
in order to not use it in replacement definition, but it doesn't help.

vbelloir, 2022-08-19:
I made two tables concerning clusters, in order to compare signature and tuya documentation

On for input:

name | Sig | Tuya doc | name
-- | -- | -- | --
Basic | 0x0000 | 0x0000 | Basic
PowerConfig | 0x0001 | 0x0001 | PowerConfig
Identity | 0x0003 | 0x0003 | Identity
Groups | 0x0004 | 0x0004 | Groups
  | 0x0006 | 0x1000 | TouchLink
TouchLink | 0x1000 |   |  
  | 0xe001 |   |  

On for output:

name | Sig | Tuya doc | name
-- | -- | -- | --
Identity | 0x0003 | 0x0003 | Identity
Group | 0x0004 | 0x0004 | Group
onoff | 0x0006 | 0x0005 | Scene
level | 0x0008 | 0x0006 | onoff
  | 0x000a | 0x0008 | level
OTA | 0x0019 | 0x0300 |  
TouchLink | 0x1000 | 0x1000 | TouchLink
  |   | 0x0019 | OTA

MattWestb, 2022-08-19:
```
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3000_kjfzuycl TS004F (38:5b:44:ff:fe:42:aa:92)
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ROK2.TuyaSmartRemote004FROK2'>
2022-08-19 08:20:08 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
```
its very true ZHA is not loading the quirk then somthing is wrong with the cluster in the signature.
The original is having one scene cluster as in that yours is not having (0x0005) and its having one extra tuya cluster as out (0xe001).

Good things is the local quirk is in the system and ZHA is trying using it !!!

I have looking one more time and i cant see the failing part (i have **not** cleaning my googles).

In this stage we can forgetting the replacement part then its not use then the signature is not matching and the quirk is not loaded for you device but the `TouchLink | 0x1000` shall not being remover in the future then it  can being used later if ZHA is getting more support for that cluster.

I think your tuya docks is for the "original ROK" or somthing that is not made in real.
Also some of the tuya dock is not true in reality (some tuya attributes is having  other function then the docks is saying).

I was looking one more time and the system is complaining that the input cluster is not OK = out cluster and the device property is OK.
And i was forgetting how to naming the cluster correct in the signature.
Can you updating the line 204 and putting in `TuyaZBExternalSwitchTypeCluster.cluster_id,` (only the last part is updated) but not braking the indent or some extra spaces and i think it shall loading OK !!

vbelloir, 2022-08-19:
You are right!

```
2022-08-19 11:59:36 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ROK2.TuyaSmartRemote004FROK2'>
2022-08-19 11:59:36 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 38:5b:44:ff:fe:42:aa:92: <class 'ROK2.TuyaSmartRemote004FROK2'>
```

[image]

And now, when I want to create an automation I can select lots of trigger!!

MattWestb, 2022-08-19:
### Great !!!
The ROK2 is having little different light commands (in command mode).
Best you is trying short press, double and long pres with release and looking for what `zha_event` you is getting then doing it and we is looking how we is doing the device automatons (DH) that working for the device. 

Also some combinations can sending more commands and we must finding the best for using in the DA.

ROK1 is having short press = toggle and no double press and ROK i think is ON and double is Off.

vbelloir, 2022-08-19:
I updated the comment above with logs for each action (press, double press, long press).

MattWestb, 2022-08-19:
I have cached:
Single: On = OK
Double: Off = OK

And is missing:
Long: ?
Long release: Level stop = OK
Rotating right  and left ?

MattWestb, 2022-08-19:
For single and double press updating line 137 with (2 lines same indent and no extra spaces !!!) :
```
        (SHORT_PRESS, BUTTON): {COMMAND: COMMAND_ON, ENDPOINT_ID: 1, CLUSTER_ID: 6},
        (DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_OFF, ENDPOINT_ID: 1, CLUSTER_ID: 6},
```
Restart HA and look what events you is getting.

MattWestb, 2022-08-19:
OPS !!!!
I was writing the wrong line number (for ROK1) !!
Do the changes at line 252.

MattWestb, 2022-08-19:
I have cooking little and i think the long press shall looking like this (with the short and double press):
```
    device_automation_triggers = {
        (SHORT_PRESS, BUTTON): {COMMAND: COMMAND_ON, ENDPOINT_ID: 1, CLUSTER_ID: 6},
        (DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_OFF, ENDPOINT_ID: 1, CLUSTER_ID: 6},
        (LONG_RELEASE, BUTTON): {
            COMMAND: COMMAND_STOP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
        },
        (LONG_PRESS, BUTTON): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
        },
```
At line 251 and until the rotating left/right that we is not having.

vbelloir, 2022-08-19:
Yes, it works quite good with that!
For short press and double press, it's perfect.

For long press, I configure a Telegram notification on `button released after long press`, And if I press less than 4s, it doesn't fire, but if I press 5s, I receive 5 notifications.

Here are all the zha_events generated

```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": [],
        "params": {
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-19T12:56:31.552785+00:00",
    "context": {
        "id": "01GAV2TCJ0ZZYCZ32ZW13WFJKR",
        "parent_id": null,
        "user_id": null
    }
}

Événement 92 déclenché 14:56:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": [],
        "params": {
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-19T12:56:30.835650+00:00",
    "context": {
        "id": "01GAV2TBVKXFXX2FJWYEWXE2CT",
        "parent_id": null,
        "user_id": null
    }
}

Événement 91 déclenché 14:56:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": [],
        "params": {
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-19T12:56:30.120938+00:00",
    "context": {
        "id": "01GAV2TB58JW4JABVBBJA4D2NP",
        "parent_id": null,
        "user_id": null
    }
}

Événement 90 déclenché 14:56:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": [],
        "params": {
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-19T12:56:29.407357+00:00",
    "context": {
        "id": "01GAV2TAEZWVJHSZMJ8MS86N9T",
        "parent_id": null,
        "user_id": null
    }
}

Événement 89 déclenché 14:56:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": [],
        "params": {
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-19T12:56:28.689989+00:00",
    "context": {
        "id": "01GAV2T9RHRKGHH2CB5Y6V45X4",
        "parent_id": null,
        "user_id": null
    }
}
```

Maybe I could filtered them in automation. I will look in nodered also.

MattWestb, 2022-08-19:
## Great !!
My experience is exactly 5 seconds for long press from the Dimmer switch and the original ROK but they is sending only one "stop command" so i think its one _personality of the device_ and all commands received is having different time stamps so its different command sent from the device.

Can you catching events the rotating the knob left and right (i think its light level for dimming up and down) ?

Also if you can trying the fixed device automatons by making automatons from the device card and look if they is working OK ?

We have near half of the device made and next on the list is "scene mode" commands :-)))

vbelloir, 2022-08-20:
> ## Great !!
>
> My experience is exactly 5 seconds for long press from the Dimmer switch and the original ROK but they is sending only one "stop command" so i think its one _personality of the device_ [...]

Yes, I think I will have to filter them.

> Can you catching events the rotating the knob left and right (i think its light level for dimming up and down) ?

This is only a button, this is not a rotating dimmer:https://www.domadoo.fr/en/smart-home/6232-moes-tuya-zigbee-smart-button-single-press-double-press-long-press-ip55.html

[image]
 
> Also if you can trying the fixed device automatons by making automatons from the device card and look if they is working OK ?
Sorry, but I don't really understand. I created three automations (one for each event), and they work pretty well.
[image]

What do you want me to do?

> We have near half of the device made and next on the list is "scene mode" commands :-)))
I know that the device has a scene mode, but I don't really understand what scene are. I have to read HA docs.

MattWestb, 2022-08-20:
All is OK i have looking on the Moes home page and finding its not one rotater.
I think the name " Smart Knob" is better then making on PR for adding it in ZHA.

We have done all light commands but if i understand is the device sending one other command after the first single click (on command) and is doing that until it have sending one double click (Off command).
Can you trying one single press for ON and then doing one more and looking what `zha_event` you is getting ?

I normally test DAs from the device card by the `Automations` plus sign for adding and selecting one DA and is adding Actions in the GUI  `zha_event` as event  (not device) and putting in one text like this (in yaml mode its looks like this)
```
action:
  - event: zha_event
    event_data:
      Rotated: Right Fast
```
Then we have getting the last command fixed we is looking for the "scene mode" (its no real scene in Zigbee only sending clicks and no Zigbee commands).

MattWestb, 2022-08-20:
I have doing some changes (device class name) and deleting rotating DA and adding the second single press from the logs you have posting (as `ALT_SHORT_PRESS, BUTTON`).

Can you testing if its working as expected ?
[ROK2,pyA.zip](https://github.com/zigpy/zha-device-handlers/files/9386611/ROK2.pyA.zip)

vbelloir, 2022-08-20:
OK, my your new file:

**Single press**
HA logs:
```
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Received ZCL frame: b'\x01\x1f\x00f\x01\x00'
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=31, command_id=0, *is_reply=False)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Decoded ZCL frame: LevelControl:move_to_level(level=102, transition_time=1, options_mask=None, options_override=None)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Received command 0x00 (TSN 31): move_to_level(level=102, transition_time=1, options_mask=None, options_override=None)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] No explicit handler for cluster command 0x00: move_to_level(level=102, transition_time=1, options_mask=None, options_override=None)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Received ZCL frame: b'\x01 \n\xdf\x00\x01\x00'
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=32, command_id=10, *is_reply=False)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Decoded ZCL frame: Color:move_to_color_temp(color_temp_mireds=223, transition_time=1, options_mask=None, options_override=None)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Received command 0x0A (TSN 32): move_to_color_temp(color_temp_mireds=223, transition_time=1, options_mask=None, options_override=None)
2022-08-20 12:39:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] No explicit handler for cluster command 0x0a: move_to_color_temp(color_temp_mireds=223, transition_time=1, options_mask=None, options_override=None)
```
Automation:
```
- id: '1660988199288'
  alias: SinglePress
  description: ''
  trigger:
  - device_id: f3102e7e98a48ce541ba5f3466f85e18
    domain: zha
    platform: device
    type: remote_button_short_press
    subtype: button
  condition: []
  action:
  - service: notify.ha_telegram
    data:
      message: short press with zha_quirk
  mode: single
```

It never trigs.

Here are event catched, immediatly after a short press:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_to_level",
        "args": [
            63,
            1
        ],
        "params": {
            "level": 63,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-20T10:44:16.875010+00:00",
    "context": {
        "id": "01GAXDMYVBZW9VS5WWBE1DH6YB",
        "parent_id": null,
        "user_id": null
    }
}
```
and
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0300",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_to_hue_and_saturation",
        "args": [
            169,
            63,
            1
        ],
        "params": {
            "hue": 169,
            "saturation": 63,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-20T10:44:17.176935+00:00",
    "context": {
        "id": "01GAXDMZ4RMEFM9W9P3GAMM0AE",
        "parent_id": null,
        "user_id": null
    }
}

```

MattWestb, 2022-08-20:
So the updated quirk is loading and working (with new name for the device) :-)))

You logs its little confusing then its sending the same command as long press :-(((

Can you start with one double press = OFF and then one single press = ON and then one more single press that shall being one other command if i understating the instruction from the Moes homepage.

By the way you filtered logs and events is great made !

MattWestb, 2022-08-20:
Can you trying updating the "second" double click DA with this ?
```
        (ALT_SHORT_PRESS, BUTTON): {
            COMMAND: COMMAND_MOVE_TO_SATURATION,
            ENDPOINT_ID: 1,
            CLUSTER_ID: 768,
            PARAMS: ("hue": 169, "saturation": 63},
        },
```
It shall triggering on the color move command.

vbelloir, 2022-08-20:
Double Press
```
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Received ZCL frame: b'\x01x\x00'
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=120, command_id=0, *is_reply=False)
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Decoded ZCL frame: TuyaSmartRemoteOnOffCluster:off()
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Received command 0x00 (TSN 120): off()
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] TS004X: send default response
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=120, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-08-20 16:27:35 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Sending reply: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
```

And zha_event:  
```

        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
```

On single press

```
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Received ZCL frame: b'\x01y\x01'
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=121, command_id=1, *is_reply=False)
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Decoded ZCL frame: TuyaSmartRemoteOnOffCluster:on()
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Received command 0x01 (TSN 121): on()
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] TS004X: send default response
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=121, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-08-20 16:28:45 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0006] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
```
```
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
```

Another single press:

```
2022-08-20 16:30:02 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Received ZCL frame: b'\x01z\x00\xff\x01\x00'
2022-08-20 16:30:02 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=122, command_id=0, *is_reply=False)
2022-08-20 16:30:02 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Decoded ZCL frame: LevelControl:move_to_level(level=255, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:30:02 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Received command 0x00 (TSN 122): move_to_level(level=255, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:30:02 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] No explicit handler for cluster command 0x00: move_to_level(level=255, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:30:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Received ZCL frame: b'\x01{\n\xbc\x00\x01\x00'
2022-08-20 16:30:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=123, command_id=10, *is_reply=False)
2022-08-20 16:30:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Decoded ZCL frame: Color:move_to_color_temp(color_temp_mireds=188, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:30:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Received command 0x0A (TSN 123): move_to_color_temp(color_temp_mireds=188, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:30:03 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] No explicit handler for cluster command 0x0a: move_to_color_temp(color_temp_mireds=188, transition_time=1, options_mask=None, options_override=None)
```
```
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_to_color_temp",
        "args": [
            188,
            1
        ],```
 and 
```
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_to_level",
        "args": [
            255,
            1
        ],```

MattWestb, 2022-08-20:
OK for getting the last post its have problem with not imported function and i doing one from the last log !!

MattWestb, 2022-08-20:
Can you looking wot args and prams the `"move_to_color_temp",` is having in the events ?
I need them pr the DA.

vbelloir, 2022-08-20:
Here is the whole zha_event logs:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0300",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_to_color_temp",
        "args": [
            188,
            1
        ],
        "params": {
            "color_temp_mireds": 188,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-20T14:30:03.019413+00:00",
    "context": {
        "id": "01GAXTJBGB8FY21VQN1K7XW4YX",
        "parent_id": null,
        "user_id": null
    }
}
```

MattWestb, 2022-08-20:
Updating the DA for the second single click like this:
```
        (ALT_SHORT_PRESS, BUTTON): {
            COMMAND: COMMAND_MOVE_COLOR_TEMP,
            ENDPOINT_ID: 1,
            CLUSTER_ID: 768,
            PARAMS: ("color_temp_mireds": 188},
        },
```
plus add this at line 33:
```
    COMMAND_MOVE_COLOR_TEMP,
```
Restarting HA and testing double (off), single (on) and the one second single (shall triggering on the color temp command).

vbelloir, 2022-08-20:
OK, file change (without the error **(** instead of **{**).

With the same sequence (double, single, single), here is what I get for the second single:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0300",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_to_color_temp",
        "args": [
            223,
            1
        ],
        "params": {
            "color_temp_mireds": 223,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-20T14:51:40.173022+00:00",
    "context": {
        "id": "01GAXVSY8DYG53R3Z4SHAGZCKH",
        "parent_id": null,
        "user_id": null
    }
}

Event 14 fired 4:51 PM:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "38:5b:44:ff:fe:42:aa:92",
        "unique_id": "38:5b:44:ff:fe:42:aa:92:1:0x0008",
        "device_id": "f3102e7e98a48ce541ba5f3466f85e18",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_to_level",
        "args": [
            102,
            1
        ],
        "params": {
            "level": 102,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
    },
    "origin": "LOCAL",
    "time_fired": "2022-08-20T14:51:39.871028+00:00",
    "context": {
        "id": "01GAXVSXYZMAM22TH2ZJMHXNY7",
        "parent_id": null,
        "user_id": null
    }
}
```
```
022-08-20 16:51:39 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Received ZCL frame: b'\x01\x03\x00f\x01\x00'
2022-08-20 16:51:39 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=3, command_id=0, *is_reply=False)
2022-08-20 16:51:39 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Decoded ZCL frame: LevelControl:move_to_level(level=102, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:51:39 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] Received command 0x00 (TSN 3): move_to_level(level=102, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:51:39 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0008] No explicit handler for cluster command 0x00: move_to_level(level=102, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:51:40 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Received ZCL frame: b'\x01\x04\n\xdf\x00\x01\x00'
2022-08-20 16:51:40 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=4, command_id=10, *is_reply=False)
2022-08-20 16:51:40 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Decoded ZCL frame: Color:move_to_color_temp(color_temp_mireds=223, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:51:40 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] Received command 0x0A (TSN 4): move_to_color_temp(color_temp_mireds=223, transition_time=1, options_mask=None, options_override=None)
2022-08-20 16:51:40 DEBUG (MainThread) [zigpy.zcl] [0xD2EE:1:0x0300] No explicit handler for cluster command 0x0a: move_to_color_temp(color_temp_mireds=223, transition_time=1, options_mask=None, options_override=None)
```

It seems not better.

MattWestb, 2022-08-20:
Its looks its sending different color changes on the second press so perhaps taking the color temp prams away but i think the move (its level / dim move) is being triggered then the long press 2 :-((

Can you doing the same sequence off, on and new single and one more new single and one more new single so we can see if its sending different color temps after the firs single press (on command).

vbelloir, 2022-08-20:
OK, I will do.

But there is something I don't understand. 
What color is supposed to change?

vbelloir, 2022-08-20:
Double Press
```
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
        "args": [],
        "params": {}
```

First Single Press
```
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": [],
        "params": {}
```

Second Single Press
```
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_to_level",
        "args": [
            102,
            1
        ],
        "params": {
            "level": 102,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_to_color_temp",
        "args": [
            223,
            1
        ],
        "params": {
            "color_temp_mireds": 223,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }
```

Third Single Press
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_to_level",
        "args": [
            63,
            1
        ],
        "params": {
            "level": 63,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }

        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_to_hue_and_saturation",
        "args": [
            169,
            63,
            1
        ],
        "params": {
            "hue": 169,
            "saturation": 63,
            "transition_time": 1,
            "options_mask": null,
            "options_override": null
        }

MattWestb, 2022-08-20:
The device is sending color temperature change and also one command for SATURATION = RBG color but we only need one of them for the "second" short press.

I trying getting the logic tuya is using so we can getting all working OK also more then one single press after the first one (on).

MattWestb, 2022-08-20:
Can you looking wot is sending with 4, 5 and 6 extra single press ?

vbelloir, 2022-08-20:
Here is the logs, with 10s between extra single press.
[zha_events.log](https://github.com/zigpy/zha-device-handlers/files/9387553/zha_events.log)

MattWestb, 2022-08-20:
M2

vbelloir, 2022-08-20:
I'm back.

MattWestb, 2022-08-20:
M2 !!
Looks first after one `off` is `on` and then its one pare of `move` (light level) and one `color move` change command that is Color temp or `hue_and_saturation`.
If using move CT its not working for all single press in the sequence then some time is hue and saturation move.
If using move level we must looking if its braking the long press and release.

Can you doing sequence with long press and release after have making one double press (off)  and one after one single press so we can see if braking the pattern for long press.

vbelloir, 2022-08-20:
Sorry for the delay.

Here are the two logs:

Long press after a off command (double press): [zha-event-after-off.log](https://github.com/zigpy/zha-device-handlers/files/9387930/zha-event-after-off.log)

Long press after an on command (single press):
[zha-event-after-on.log](https://github.com/zigpy/zha-device-handlers/files/9387932/zha-event-after-on.log)

MattWestb, 2022-08-21:
Im being confused ;-((
Both logs is starting with one `off` command if that  right or is is false mad ?
So long press is sending  off and then stop level that is repeated until the button is released.
Its make no sense then searing lights but its still one tuya device is its tuya logic we is dealing with.
If its the true we can using move level for second short press without args and we is getting triggering OK without braking the long press.
But we is missing log press / release and only getting the stop command = long pressed with repeating.

Do you have one light (no color is OK but better with CT or RGB) we can testing sending the command 2 ? 
Then you can see how its working in real and easier to understanding the (tuya) logic.

vbelloir, 2022-08-21:
Hi @MattWestb 
my message of this morning was not sent....

I was saying that yes, each log start with an OFF command, and this is what I wanted.
If I well understood, you wanted two logs of Long Press : one after an OFF command, and another after an ON command.

Getting an OFF command is easy with double Press, but it is not the same with ON command. Each single press does not fire an ON command, only one after an OFF command always fire an ON command. This is the reason why I realized the following sequence : OFF (double Press), ON (single Press) and then a long press.

In each logs, a long press is made by pressing 5s and then releasing.
In `zha_event-after-off ` log, we see the OFF command and then 10 STOP command sent in 3s....

In `zha-event-after-on` log, we see an OFF command (double press), an ON command (simple press), and then 5 STEP commands in 3s, and then only one STOP command, corresponding to the button release.

This is not what you wanted?

I don't  have zigbee bulb at home, but we got some at work, I will see if I can take one.

Thanks for your time, I really appreciate

MattWestb, 2022-08-21:
More or less then i like understanding the logic tuya have putting in the device.

Its in light command mode (we have only testing that and not the scene/event mode) is being "normal" after on off command and after one on command its changing function that making its tricky getting  the device automation working good for "normal users" that like have single, double and long press and not only long press then the device is is off mode and not in on mode.

I think i was not looking good enough in the on log and shall doing it later for see if i can getting commands that working well but im afraid that we is getting problem with commands after off.

We trying doing the best we can with the device in the light command mode then we is looking if e can getting the scene / event mode working OK.

vbelloir, 2022-08-21:
Ok, I will try to find a bulb tomorrow at work!

vbelloir, 2022-08-22:
I can  have this bulb : https://www.luminaire.fr/eglo-connect-e27-globe-dore-tunable-white.html

and a Philips Hue White and color IBRS  10461, 5600 VB, NL

MattWestb, 2022-08-22:
The EGLO is Bluetooth only i think (i dont understanding France :-(( ) so not working sending Zigbee commands to it.
The HUE With and color you can testing all functions in the light command mode but they is normal expensive and have some no zigbee standard stuff i dont like but shall working OK

vbelloir, 2022-08-22:
The EGLO bulb I have in my hand is clearly a Zigbee bulb, but you are right, my link is not with the good ref.
Here is its signature :
[eglo-sig.txt](https://github.com/zigpy/zha-device-handlers/files/9396162/eglo-sig.txt)
 
 
Right now, I don't manage to put the hue bulb in pairing mode...

MattWestb, 2022-08-22:
GOOOOD !!

If you have paring the EGLO and / or the HUE and can turning it on and off from the device card and also dimming it then it shall being OK.

On the device / integration card click on ZHA `configure` and the on the `group` tab.
Then click add group and adding the lights and the Smart switch (its not one light but tuya is adding groups this way on there TS004F light controllers) and wake up the smart  knob then clicking save so its not sleeping then sending commands to it.

If all works OK it the knob shall controlling the lights in the Zigbee light group.

vbelloir, 2022-08-22:
For both, I can turning on and off, but I can't change color or temp.
Both has entity named `_level_start_up_current_level`, but it does nothing.

 [image]

MattWestb, 2022-08-22:
Click on the name of the device / light (you is getting the name setting) then click on the setting icon (top right) then you is getting the GUI card for the device.
Then the light is off you can only toggle and light level.
Then the light is on you shall getting TC and Color controls on the card if the device is supporting it.

vbelloir, 2022-08-22:
Ok, I created a dashboard with some settings:

[image]

vbelloir, 2022-08-23:
Hi @MattWestb 
Is there something I can test?
(I wanted to test zigbee2mqtt, but my dongle is not supported)

MattWestb, 2022-08-23:
Try adding the lights in one Zigbee group and adding the remote to the group as i was writing in https://github.com/zigpy/zha-device-handlers/issues/1697#issuecomment-1222670677.

I think On and Off its working OK with the DA we is having.
We need getting the logic how long and double press is working then the last command sent was on and off so the automatons is working OK for both modes.

vbelloir, 2022-08-23:
Ok, sorry, I have missed this part of your #167 comment.

I created a group, and a both eglo and hue bulbs, but I can't add the switch. It is proposed by the GUI, but if I select it and click on `Add device` it is not added, and still in the Add device section:  
[image]

[83 more replies]

closed 2023-07-03 by TheJulianJES