zigpy/zha-device-handlers #1698: Tuya TS0201

Harvsch, 2022-08-17:
tuya.ts0201_neo.TemperatureHumidtyIlluminanceSensor.
It installed with no issues, but initialization never ended. The sensor properly sends luminance data, but temperature, humidity  status are both shown as unknown. Battery initially showed, but then went unknown.

The integration is HubZ Smart Home Controller - /dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_81300F1A-if01-port0, s/n: 81300F1A - Silicon Labs - 10C4:8A2A
Zigbee Home Automation
Running on an Odroid XU4

<details>
log
[zha-c2982df7e5d69e1181b908028d0307c3-LUMI lumi.sensor_motion-2448aeea293783dd0168f511a6a1fbb5.json.txt](https://github.com/zigpy/zha-device-handlers/files/9365486/zha-c2982df7e5d69e1181b908028d0307c3-LUMI.lumi.sensor_motion-2448aeea293783dd0168f511a6a1fbb5.json.txt)

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

``{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0106",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0400",
        "0xe002"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_qaaysllp",
  "model": "TS0201",
  "class": "zhaquirks.tuya.ts0201_neo.TemperatureHumidtyIlluminanceSensor"
}`yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2022-08-17:
Witch firmware version is running on your HubZ ?
You shall have EZSP 6.7.8.0 for getting the stick working well and old firmware is having some nasty bugs.

javicalle, 2022-08-18:
Have you tried to pair the device again? 
Try to keep the device awake during the pairing process just short pressing the pair button every second or so.

MattWestb, 2022-08-23:
I think its the original firmware it was shipped with.
One user have "collecting" firmware and one great firmware cooker have made one EZSP 6.7.8 that is the recommended stable version.
Firmware and instruction is https://github.com/walthowd/husbzb-firmware.
Its gets 2 firmware one with original` ncp-uart-sw-6.7.8.ebl` and one with higher baud rate `ncp-uart-sw-6.7.8-115k.ebl` but if using the latest you need update your setting for getting it working after the firmware update (how to that is posted in the git issues).

closed 2023-07-10 by github-actions[bot]