zigpy/zha-device-handlers #1700: [Device Support Request] Tuya Zigbee smart energy meter DDS238-2 Zigbee with a new manufacturer variant (_TZE200_bkkmqmyo)

mmaccaroni, 2022-08-21:
Hi, I purchased this device which was markes as working with zha, unfortunatrly I got one with a new manufacturer, and, upon pairing, no entities appear.
This is the signature

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_bkkmqmyo",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

Tried to adapt other quirks that were shared in the past  with no success

darckly, 2022-08-21:
Hello @mmaccaroni ,

I assume the new device is the single phase power metter, correct?
What you tried to change in the quirk, and how installed it?

mmaccaroni, 2022-08-21:
> Hello @mmaccaroni ,
> 
> I assume the new device is the single phase power metter, correct? What you tried to change in the quirk, and how installed it?

Hi darckly, thanks for your reply. Yes it is the single phase.
I set up a folder for custom quirks and pointed to in my configuration.yaml.
The change I made iwas the manufacturer string as per signature, but nothing happened. I truly have no idea what am I doing wrong.

darckly, 2022-08-21:
Hi @mmaccaroni 

Looks like you are doing the right things,
What you see in the Device Info screen?

If it loaded with using custom quirk, the scree should be like this:
[image]

mmaccaroni, 2022-08-21:
If I use the quirk I got thus error 

Logger: homeassistant.config_entries
Source: quirks/ts0601_din_power.py:28
First occurred: 13:36:46 (1 occurrences)
Last logged: 13:36:46

Error setting up entry Sonoff Zigbee 3.0 USB Dong - /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_9a4120891045ec118fb796fd6f14af06-if00-port0, s/n: 9a4120891045ec118fb796fd6f14af06 - ITead - 10C4:EA60 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 357, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 101, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/quirks/ts0601_din_power.py", line 28, in <module>
    class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 91, in __init_subclass__
    raise TypeError(
TypeError: `manufacturer_attributes` is deprecated. Copy the parent class's `attributes` dictionary and update it with your manufacturer-specific `attributes`. Make sure to specify that it is manufacturer-specific through the  appropriate constructor or tuple!

mmaccaroni, 2022-08-21:
This is the quirk I am loading for that error

`"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    manufacturer_attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t),
        TUYA_CURRENT_ATTR: ("current", t.int16s),
        TUYA_POWER_ATTR: ("power", t.uint16_t),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value / 10)
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value / 10)
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
            )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

    def voltage_reported(self, value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value)

    def power_reported(self, value):
        """Power reported."""
        self._update_attribute(self.POWER_ID, value)

    def current_reported(self, value):
        """Ampers reported."""
        self._update_attribute(self.CURRENT_ID, value)

class TuyaElectricalMeasurement(LocalDataCluster, Metering):
    """Custom class for total energy measurement."""

    cluster_id = Metering.cluster_id
    CURRENT_ID = 0x0000
    POWER_WATT = 0x0000

    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {0x0300: POWER_WATT}

    def energy_reported(self, value):
        """Summation Energy reported."""
        self._update_attribute(self.CURRENT_ID, value)

class TuyaPowerMeter(TuyaSwitch):
    """Tuya power meter device."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.switch_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        MODELS_INFO: [
            ("_TZE200_byzdayie", "TS0601"),
            ("_TZE200_ewxhg6o9", "TS0601"),
            ("_TZE200_bkkmqmyo", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51
            # device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
`

darckly, 2022-08-21:
@mmaccaroni 

Hmm, I don't update my instance of HA for several months, so looks like my configuration is not affected. But I see that a lot of dicsussions about this error: https://github.com/home-assistant/core/issues/69494

The possible solutions, as I see from the discussion:
- Remove /config/deps/lib/
- Reverting to version 2022.3.8
- Changing "manufacturer_attributes" to "attributes" 

Could you try the solutions from that tread?
Sorry, can't support you with testing, as temporary I haven't ability to update my configuration and check.

darckly, 2022-08-21:
@mmaccaroni 

I was not aware, but indeed, some changes were made in April to support ZCLv7 zigpy changes.
Please use as a source for modification this file: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py

Not the file attached in the issue #1298

mmaccaroni, 2022-08-21:
@darckly 

You are the man thanks now the quirk seems to be working when I rejoin the device!  I see the switch control and the electrical measurement entity. However none of them seems to be working. What do you need ? Is it because of clusters right?
[image]

darckly, 2022-08-21:
@mmaccaroni sometimes it need some time for sensors to appear (or restart HA again), and also connect the load to the power metter, e.g. iron or kettle.

mmaccaroni, 2022-08-21:
Ok i will try, in the meantime if you need me for debugging, if needed, I am at full disposal. Thank you again!

darckly, 2022-08-21:
Well, in parallel you can debug your device as desribed here: https://github.com/zigpy/zha-device-handlers/issues/1298#issuecomment-1018494493 and send me the logs

mmaccaroni, 2022-08-21:
[zhaquirks.log](https://github.com/zigpy/zha-device-handlers/files/9389505/zhaquirks.log)
@darckly thanks a lot, I enabled debugging as per instructions, let me kn ow if it is sufficient for you.

darckly, 2022-08-22:
@mmaccaroni Could you send me a link where you ordered this version of power meter?
I briefly looked into the log, and a bit puzzled. Looks like this is completelly new device with own attributes.
I want to check the specification of your device.

mmaccaroni, 2022-08-22:
@darckly this is the link where I ordered from Aliexpress:
https://a.aliexpress.com/_v1DOnz 
Nes device? Wow that was really unexpected.

darckly, 2022-08-22:
@mmaccaroni may I ask you to unpair the power meter, enable logging, pair the metter again (with load), turn on/off the device and send me the logs again.
I don't know what the reason, but I don't see the strings with "(command 0x0001)" ending in the log. Command 0x0001 is TUYA_GET_DATA, so after this command  the device should send the actual data to HA

darckly, 2022-08-22:
And do you have any other devices connected via ZHA, like termostat?

mmaccaroni, 2022-08-22:
Sure will do, power on and off from zha or manually from the device?
Yes I have other devices: Ikea Fyrtur covers with their remotes and repeaters, TS011F pkugs, Aqara Opple 6 button remote and my coordinator is a Sonoff zigbee 3.0. Everytings works perfectly with no issues.

darckly, 2022-08-22:
Please do the both: on and off from ZHA and manually.

mmaccaroni, 2022-08-22:
@darckly ok I performed the operations, the log is attached. I don't know if it is good but I enclose every ZHA related events, not only the quirk debug strings.
For your reference the device is joinded with Nwk: 0xc150.
When trying to operate upoon the zha switch entityi get the following error:

`[546990539824] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1713, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 89, in async_turn_on
    result = await self._on_off_channel.turn_on()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 342, in turn_on
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable`

it is also in the attached log. Events in the log are from the device join process, load attached since the beginning. Let me know if you need anything else and again thanks a lot for your support.
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9394695/home-assistant.log)

darckly, 2022-08-23:
@mmaccaroni Could you try this slightly modified quirk? At least to test on/off functionality

I'm trying to see what changed in other tuya quirks and apply the same changes..
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/9401523/ts0601_din_power.zip)
.

mmaccaroni, 2022-08-23:
@darckly thanks for your reply, loaded the quirk and repaired the device, the switch doe not work, but Idon't get the error message anymore. I attach the log for your reference
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9402177/home-assistant.log)

darckly, 2022-08-27:
@mmaccaroni unfortunatelly, can't find the way how to implement the support without having this device. I see that on zigbee2mqtt also has an issue with this manufacturing ID : _On the outside, there are no differences. This version is not recognised by zigbee2mqtt; apparently it uses another format._ (https://github.com/Koenkk/zigbee2mqtt.io/pull/1527/commits/e463498dcc49bdce1615d7d0132a02ca8872a6f1) 

Maybe someone else from contibutors can support... Also I'll continue to monitor Z2M forum.

mmaccaroni, 2022-08-28:
Thanks a lot @darckly really appreciated your support!

evlo, 2022-09-02:
> @mmaccaroni Could you try this slightly modified quirk? At least to test on/off functionality
> 
> I'm trying to see what changed in other tuya quirks and apply the same changes.. [ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/9401523/ts0601_din_power.zip) .

I tried your quirk for TS0601 by _TZE200_byzdayie
Quirk: ts0601_din_power.TuyaPowerMeter

My related comment here: https://github.com/zigpy/zha-device-handlers/issues/1298#issuecomment-1235701101

I'm  new to quirks and ZHA, but it seems that it now works, probably
[image]

gerrytb, 2022-09-12:
I am on the same situation, I bought this device because it was marked as **_"supported"_** on https://zigbee.blakadder.com/Hiking_DDS238-2.html 
So I followed the link of aliexpress and now I realized that is a total different product. I didn't receive it yet, so by the moment I cannot help.
I hope that it can be supported soon.

johndekroon, 2022-10-01:
Just received the same device. And since I'm here you can guess I also received the unsupported _TZE200_bkkmqmyo version, which was sold as "supported". I can pair it, but no entities are showing up.

boszenlaszlo, 2022-10-13:
I have the same device and i almost managed to work everything, except the current reading. Based on Zigbee2mqtt [https://github.com/Koenkk/zigbee2mqtt/issues/13767](url) i need to separate the voltage and current readings from one attribute, but i dont know how.
[image]

My custom quirks: [ts0601_din_power.txt](https://github.com/zigpy/zha-device-handlers/files/9774876/ts0601_din_power.txt)

johndekroon, 2022-10-13:
> I have the same device and i managed to work almost everything, except the current reading. Based on Zigbee2mqtt [Koenkk/zigbee2mqtt#13767](https://github.com/Koenkk/zigbee2mqtt/issues/13767) i need to separate the voltage and current readings from one attribute, but i dont know how. [image]
> 
> My custom quirks: [ts0601_din_power.txt](https://github.com/zigpy/zha-device-handlers/files/9774876/ts0601_din_power.txt)

Interessing. I tried your quirk and now it indeed shows readings. However: the readings are entirely different than on the screen of the device.

EstSoft87, 2022-10-20:
Any good news?

boszenlaszlo, 2022-10-29:
Current reading is works now. I put the some comment about the cluster data in the file. I put the reactive power in the code, but dont show up as sensor.
[image]

[image]

You can download the new version here: [ts0601_din_power_v2.zip](https://github.com/zigpy/zha-device-handlers/files/9894049/ts0601_din_power_v2.zip)

Dogmeat88, 2022-10-29:
> You can download the new version here: [ts0601_din_power_v2.zip](https://github.com/zigpy/zha-device-handlers/files/9894049/ts0601_din_power_v2.zip)

Thanks for your efforts @boszenlaszlo 
I have added the quirk you have been working on and all the values seem to be displaying fine. However the toggling the switch produces the error:  `'Status' object is not subscriptable`
Any idea what could be causing this?
Thanks in advance.

[image]

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/zha/core/channels/general.py:350
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 8:53:10 AM (3 occurrences)
Last logged: 9:48:39 AM

[139654927593296] 'Status' object is not subscriptable
[139654898783744] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 97, in async_turn_off
    result = await self._on_off_channel.turn_off()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 350, in turn_off
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
```

johndekroon, 2022-10-30:
I have a feeling that the current wattage is a bit off :)
[image]

flaktivated, 2022-11-01:
Hello, I have one of these devices and seem to be working on it in parallel. 

I have the following configured and displaying correct values in Home Assistant:
- Energy Summation
- RMS Voltage
- RMS Current
- Active Power
- AC Frequency
- Power Factor
[image]

The switch successfully toggles the Meter, but get the error message as well. @javicalle  hopes that the following PR will fix the error message: https://github.com/zigpy/zha-device-handlers/pull/1859

With the following attributes correctly mapped but not showing as sensors:
- Reactive Power
- Sum of Reactive Power

[image]
[image]

Has anyone here got the reactive power to display under sensors?

javicalle, 2022-11-01:
> 1. How to increase the resolution of sensors (i.e. Total Reactive Power, would be nice to have 2 Decimal Places)

The `Sensor` class have a `decimal` attribute:
* https://github.com/home-assistant/core/blob/914ccdbc4f931603aebc009983315ed15e695a59/homeassistant/components/zha/sensor.py#L122-L130

flaktivated, 2022-11-01:
Updated the total reactive power sensor definition as follows:

```python
@MULTI_MATCH(channel_names=CHANNEL_ELECTRICAL_MEASUREMENT)
class ElectricalMeasurementTotalReactivePower(ElectricalMeasurement, id_suffix="total_reactive_power"):
    """Total Reactive Power measurement."""

    SENSOR_ATTR = "total_reactive_power"
    _attr_device_class: SensorDeviceClass = SensorDeviceClass.REACTIVE_POWER
    _attr_should_poll = False  # Poll indirectly by ElectricalMeasurementSensor
    _unit = POWER_KILO_VOLT_AMPERE_REACTIVE_HOUR
    _decimals: 2
```

Restarted HA and checked sensor dashboard and still shows 0dp:
[image]

Thought my quirk might have a bug, checked configuration for a division error:
```python
elif attrid == HIKING_TOTAL_REACTIVE_ATTR:
            self.endpoint.electrical_measurement.reactive_energy_reported(value / 100)
```
[image]
Meter shows 0.2kvar on display so all looks good with the above.

Changes the divisor from `100` to `10` to test and the value for the sensor shows the value `2`. So the quirk doesn't seem to be the issue.

javicalle, 2022-11-01:
I believe that the `total_reactive_power` attribute is a integer:
* https://github.com/zigpy/zigpy/blob/8a2c7548065367d7e94ddf5857652ca1cd087e70/zigpy/zcl/clusters/homeautomation.py#L190

So, the `23/100=0,23` --> `0`

The problem is that `total_reactive_power` don't configure any `multiplier` or `divisor` that can allow the decimal values.

Maybe you can keep the value as is in the `_update_attribute` and overwritte the `formatter` in the `ElectricalMeasurementTotalReactivePower`:
```python
    def formatter(self, value: int) -> int | float:
        """Return 'normalized' value."""
        value = float(value) / 100
        if value < 100:  # only round tinny values
            return round(value, self._decimals)
        return round(value)
```

flaktivated, 2022-11-01:
That works perfectly! Will merge this change into the Quirk PR to make it ready for display in the future.

javicalle, 2022-11-02:
> ```python
> # Reactive power units
> ...
> POWER_KILO_VOLT_AMPERE_REACTIVE_HOUR: Final = "kvarh"
> ```

According to the specification the units would be `kVAr`:

[image]

[image]

Reference:
* https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf

flaktivated, 2022-11-02:
Thanks for picking that up, I just copied what came with the meter instructions.

Keen to get this merged into core is there any etiquette or best practices to follow? Or just fork and PR?

javicalle, 2022-11-04:
Let see if I can recall all the info.

The current device classes in HA are listed here:
* https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes

There is a `reactive_power` type and the supported units are `var` (in lowercase 🤷🏻‍♂️)

Usually, any change of types and units will require a architecture discussion.
The `reactive_power` was discused here:
* https://github.com/home-assistant/architecture/discussions/688
and implemented here:
* https://github.com/home-assistant/core/pull/62010

After that, a couple of discussions have been arise:
* https://github.com/home-assistant/architecture/discussions/724
* https://github.com/home-assistant/architecture/discussions/729

The first one (724) talks about the reactive energy and its units `kvarh` (so your device measure `reactive_power` or `reactive_energy`?) . Also reference some specifications and integrations that use it.

So, what now?
I guess that the path to follow would be the architecture discussion. Can be a slow way but I believe that is the right one.
You can expose your case there and see if that drives you somewhere.

marcoNieddu, 2022-12-03:
No possible to change the teporting frequency

javicalle, 2022-12-08:
Can someone confirm that #1878 fix that issue?

closed 2023-07-10 by github-actions[bot]

jovanyeo, 2024-02-21:
> Current reading is works now. I put the some comment about the cluster data in the file. I put the reactive power in the code, but dont show up as sensor.
> [image]
>
> [image]
>
> You can download the new [...]

Hello,

Does this quirk works for this: _TZE204_cjbofhxw too?