zigpy/zha-device-handlers #1707: [BUG] Tuya Air Quality gives incoherents values

lhanneus, 2022-08-27:
The Tuya air quality detector located here zhaquirks/tuya/air/ts0601_air_quality.py gives incoherents values for one of the 2 devices : **149,9°C temperature and 105,% of humidity**.
Quirk :  zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2SensorGPP

The 2 sensors are installed in the same environment and positioned 20cm apart .
[image]

javicalle, 2022-08-27:
Can you share the device's signature for both devices?
Please, be sure to format the code putting it inside the ``` marks.

interlock, 2022-09-22:
I am also seeing swapped values for CO2 and Formaldehyd

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_dwcarsat",
  "model": "TS0601",
  "class": "zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2SensorGPP"
}
```

drmulligan, 2022-10-05:
#1406 and #1563 are related.  Variants of this device is not properly supported yet with missing or sensors mixed up.  

In my case my zigbee signature is identical to the one posted by @interlock above and as they reported CO2 and Formaldehyde are reversed but also Voclevel is actually reporting PM2.5 values and the VOC is missing.

pamatika, 2022-10-28:
Any updates on this issue?
Is there someway to help in debugging?

lhanneus, 2022-10-30:
The hardware is badly designed and give incoherent data. **Anyone should avoid buying it.**

Oliviakrkk, 2022-10-30:
Are there any alternatives? Heiman HS2AQ is there, but no one is selling it...

marinswk, 2023-01-19:
Is this being investigated? is there a way in which a fix can be pushed? it basically renders a whole set of devices unusable on ZHA, i think there is already a solution on Z2QMTT integration.

drmulligan, 2023-07-18:
I can confirm that this is still an issue for _TZE200_dwcarsat.

closed 2024-01-21 by github-actions[bot]