zigpy/zha-device-handlers #1714: [Device Support Request] https://github.com/zigpy/zha-device-handlers/issues/new?assignees=&labels=&template=device-support-request.md&title=%5BDevice+Support+Request%5D-ZZZZZZZAAAAAA--+-+-+

alekgee, 2022-08-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

axellebot, 2022-08-30:
Can you please provide output file when calling `zha_toolkit.scan_device` service by installing [ZHA Toolkit](https://github.com/mdeweerd/zha-toolkit).

alekgee, 2022-08-31:
{
  "ieee": "2c:11:65:00:00:00:00:00",
  "nwk": "0xdb05",
  "model": "CH2AX/SWITCH/1",
  "manufacturer": "Schneider Electric",
  "manufacturer_id": "0x4190",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0100",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "Basic",
          "name": "basic",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "zcl_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 3
            },
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "app_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 2
            },
            "0x0002": {
              "attribute_id": "0x0002",
              "attribute_name": "stack_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 6
            },
            "0x0003": {
              "attribute_id": "0x0003",
              "attribute_name": "hw_version",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "REPORT|READ",
              "access_acl": 5,
              "attribute_value": 1
            },
            "0x0004": {
              "attribute_id": "0x0004",
              "attribute_name": "manufacturer",
              "value_type": [
                "0x42",
[... 1520 lines cut ...]
              "command_id": "0x00",
              "command_name": "0",
              "command_args": "'not_in_zcl'"
            },
            "0x01": {
              "command_id": "0x01",
              "command_name": "1",
              "command_args": "'not_in_zcl'"
            },
            "0x02": {
              "command_id": "0x02",
              "command_name": "2",
              "command_args": "'not_in_zcl'"
            }
          }
        }
      }
    },
    {
      "id": 242,
      "device_type": "0x0061",
      "profile": "0xa1e0"
    }
  ]
}

axellebot, 2022-08-31:
Can you edit your json comments and please surround long json output (like in the exemple when creating a issue) :

This example :

> \<details>
>   \<summary>My output\</summary>
> 
> \```json
> My long json
> \```
> 
> \</details>

Will be shown like this :
<details>
  <summary>My output</summary>
 
```json
My long json
```
</details>

This way it will be readable and avoid mentioning developers that are are listed in the json

alekgee, 2022-09-01:
@axellebot updated the formatting

TheJulianJES, 2023-01-04:
See: https://www.home-assistant.io/integrations/zha/#modifying-the-device-type

PR #4151 (2025-07-14): Add Schneider Electric CH2AX/SWITCH/1 support
## Proposed change
Checked that signature looks identical to `NHPB/SWITCH/1`

Fixes #4150
Fixes #1714

## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2023-07-10 by github-actions[bot]