zigpy/zha-device-handlers #1717: [BUG] TS0210 Tuya won't reset state

AyoKeito, 2022-08-29:
**Is your feature request related to a problem? Please describe.**
Pretty much the same problem as in #963 
Home Assistant is getting "Detected" but won't reset it, unless physical button is pressed on the device. Then, the actual data is received. Shaking the device also changes state to "Detected" but it won't reset.

**Describe the solution you'd like**
The sensor state should reset to false after timeout.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_bmfw9ykl",
  "model": "TS0210",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.8.7",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Moscow",
    "os_name": "Linux",
    "os_version": "5.15.0-43-generic",
    "run_as_root": true
  },
  "custom_components": {
    "average": {
      "version": "2.3.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.26.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    }
  },
  "integration_manifest": {
[... 266 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
Obvious changes between #963 and my device signature are different manufacturer and "0x000a" moved from "in_clusters" to "out_clusters".
I've tried putting #963 quirk to custom_zha_quirks but it didn't work, probably because of different manufacturer.
Running latest HA in Docker, updated today just in case.

Device is moved 3 hours ago and physical button pressed 20 minutes ago:
[image]

javicalle, 2022-08-29:
Copy the `ts210.py` file in your local quirk folder:
* https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0210.py

Add this class at the end of the file:
```python

class TuyaVibration_TO(CustomDevice):
    """Tuya vibration sensor (TO)."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.motion_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODEL: "TS0210",
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0402",
            # "in_clusters": ["0x0000", "0x0001", "0x0500"],
            # "out_clusters": ["0x000a", "0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: IAS_VIBRATION_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    VibrationCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

AyoKeito, 2022-08-29:
Oh! That was fast. 
Thank you, it worked!

PR #1720 (2022-08-30): Add `TuyaVibration_TO` devices
Add model:TS0210 devices with the Time as `OUTPUT_CLUSTERS`

Fixes: #1717
closed 2022-08-30 by dmulcahey