zigpy/zha-device-handlers #1722: [Device Support Request] _TZE200_pay2byax windows/door sensor not fully supported

bsafh, 2022-08-30:
**Is your feature request related to a problem? Please describe.**
This device shows up in ZHA but it does not show the battery level correctly and it seems to have the wrong class/values for iaszone. Opening the windows does not change the state shown.

**Describe the solution you'd like**
I would like to have this sensor fully supported as windows/door sensor.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_pay2byax",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.8.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.10.0-17-amd64",
    "supervisor": "2022.08.5",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.26.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "openweathermap_all": {
      "version": "0.0.1",
      "requirements": [
[... 356 lines cut ...]
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

javicalle, 2022-09-04:
Duplicate of #1502 

Maybe you can try with the proposed quirk:
* https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1118930342

dmulcahey, 2022-09-04:
Closing duplicate

closed 2022-09-04 by dmulcahey