zigpy/zha-device-handlers #1726: [Device Support Request] Danfoss Ally eTRV0100 - other features than just temperature and setpoint

OZ1SEJ, 2022-08-30:
**Is your feature request related to a problem? Please describe.**
Currently, the Danfoss Ally integration supports a temperature sensor (what's the current temperature at the valve?) and a temperature control (setpoint).

**Describe the solution you'd like**
According to https://zigbee.blakadder.com/Danfoss_014G2461.html the device supports

- % opening
- child lock
- external temperature sensor
- local temperature battery
- open window detection
- ota
- thermostat

...of which only "local temperature battery" (I suppose) and "thermostat" are currently supported in ZHA. The other parameters seem to be available by creating template sensors, but I believe these ought to be directly available.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4678, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0020",
        "0x0201",
        "0x0204",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0000",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Danfoss",
  "model": "eTRV0100",
  "class": "zhaquirks.danfoss.thermostat.DanfossThermostat"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Copenhagen",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
    "supervisor": "2022.08.5",
[... 398 lines cut ...]
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```
N/A

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

N/A

mhaas, 2022-09-01:
Would it be possible to map the non-functional "off" state to the lowest possible temperature instead?

OZ1SEJ, 2022-09-03:
That certainly sounds like a good idea.

stachjankowski, 2022-10-08:
If anyone would be interested in implementing these changes, here is the documentation.
https://assets.danfoss.com/documents/202524/AU417130778872en-000101.pdf

Caius-Bonus, 2023-01-29:
I have implemented all attributes mentioned in the file and have opened pull requests in the relevant repositories:
https://github.com/home-assistant/core/pull/86907
https://github.com/zigpy/zha-device-handlers/pull/2150
Note that just installing the quirk is not enough to get entities in home assistant.

jkozera, 2023-08-02:
#86907 is still not merged. I was testing #86907 over the winter by patching it into my HA instance together with https://github.com/home-assistant/core/pull/86907, and it seemed to work well.

OZ1SEJ, 2024-01-30:
Due to this issue, I switched from ZHA to Zigbee2MQTT a long time ago. If this has since been remedied in ZHA, please feel free to close this issue...

closed 2024-08-04 by github-actions[bot]