zigpy/zha-device-handlers #1728: [Device Support Request] Initializing with no signature?

bulldog5046, 2022-08-31:
**Is your feature request related to a problem? Please describe.**
I'm currently trying to overcome some specific challenges adding standardised zigbee devices (Solar inverters) and the processing of their broadcast packets.

**Describe the solution you'd like**
I have 3 main issues to overcome.

1. I need the network to always permit associations as the inverters/zigbee radios expect to join the network on each boot cycle. ie. when the sun is out.
2. I need to understand and be given some steer on if Quirks can be used to initialize devices that do not response to node descriptor requests? I don't currently understand if/how it's possible.
3. I would need a method to capture broadcasts on the network and process the payload data for it to be reported into HA. 

<details>

<summary>Device signature</summary>

This isn't currently possible as the device won't respond to the node descriptor. I've included a pcap of the zigbee traffic flow.
[enecsys_join.pcapng.zip](https://github.com/zigpy/zha-device-handlers/files/9462803/enecsys_join.pcapng.zip)

And additionally, a failing flow to a Zigpy controlled network.
[enecsys_fail.zip](https://github.com/zigpy/zha-device-handlers/files/9462817/enecsys_fail.zip)

</details>

MattWestb, 2022-08-31:
By Zigbee standard some very basic things is one must for getting it certificated for having some security in the network.
Its looks your device is not supporting basic device information and also is not saving the network data in NVR so on real world its not one Zigbee device its one device that is using **some** Zigbee functionality and not being one Zigbee device.

If the device is not responding on node description request from the network its not possible identifying witch device it is and how it working for the system = the system cant handling the device at all. Perhaps its possible patching ZHAs database with the missing information but its not recorded trying that.

Have the network open for rejoining 24/7 is the same as putting your house key on the outside the enter door and with instruction how to using it  for all that is passing by = zero security.

Recommending sending back the controller back and demanding getting one Zigbee certificated product or one with cable for connecting the system with one "normal"  protocol that HA can using (HA is supporting very many so it shall not being one problem).

bulldog5046, 2022-09-01:
Hi Matt,

I agree with your points and perhaps the use of quirks isn't the way to go (or, even possible).

This stems from hardware that was produced by a Company called Enecsys that went out of business in 2015. At that point in time the solar systems they deployed used their own gateway device and proprietary monitoring system. When they went into administration the community found a way to repurpose the existing gateway device by collecting logs from its HTTP server, parsing them and forwarding to third party monitoring solutions.

Now, between a mix of the gateway devices failing and more used inverter units appearing on the market and being there is more of a need for a solution to gather the data directly.

Going a bit off-topic but i would welcome any idea's for potential solutions. It's sounding like a modified version of Zigpy might be required to do what i need, I guess this could maybe be dockerised into some sort of virtual gateway to run on HA?

MattWestb, 2022-09-01:
I dont knowing if its possible then its depend if the system can getting enough information from the device.
If i was you i shall using one Zigbee coordinator for this devices and not putting all you devices in one production system (if you like having more "normal" device make one second system for then that is secure).
You can making one automation that is sending permit joining every X seconds (254 is max for one Zigbee 3 coordinator) so you is getting one open network so you device can joining if the is being power retested.

Then you cant getting all information from the device the system cant handling it OK. I was thinking adding one "normal" device like one light and then patching the `zigbee.db` in your HA configuration directory (then HA is stopped).
ZHA is adding devices and many more data from the device then paring and you can changing device MWK and IEEE to your "problem device" and then adding endpoints and cluster to them and last attributes. Its dependency between the tables but i diónt knowing haw the code is doing it so it being one try and error.

The most important is that the device is getting the network key so it can communicating with the system. Also if you can getting information with endpoints and cluster its using so and adding them from the back door.

Perhaps its possible getting it working or not, i dont knowing.

Perhaps more user and devs is having more ideas how to getting them working in your system but i think luck of information is the largest problem for us.

By the way if the device is made 2015 or even earlier it can being that they is using the "secret" universal key and perhaps can patching ZHA for allowing using it in your system (its blocked in Zigbee 3 for security reasons).

closed 2023-03-07 by github-actions[bot]