zigpy/zha-device-handlers #1733: [Device Support Request] _TZE200_byzdayie DDS238-2 - total power consumed in kWh possible?

evlo, 2022-09-02:
Hi,

this is my first time working with ZHA and first time with quirks, have this device

https://zigbee.blakadder.com/Hiking_DDS238-2.html

In HA it shows as

```
TS0601
by \_TZE200\_byzdayie
Quirk:  ts0601\_din\_power.TuyaPowerMeter
```

Only thing working was the switch

I have put this

```
zha:
enable_quirks: true
custom\_quirks\_path: /config/zha_quirks
```

into `configuration.yaml`

and then into /config/zha_quirks this file

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601\\\_din\\\_power.py

I have restarted HA

Now it shows

```
Electricalmeasurement
0 W
Electricalmeasurementpowerfactor
Unknown
Electricalmeasurementrmscurrent
0.0 A
Electricalmeasurementrmsvoltage
241 V
```

switch still works, pretty much all good.

I was wondering if there was a way to get the total power in kWh as a sensor reading in HA?

I have tried https://github.com/zigpy/zha-device-handlers/issues/1700#issuecomment-1223842781 but it did not get total power (in kWh) as a sensor

evlo, 2022-09-03:
This value shows "null" until meter measures some about 0.2 amount of kWh consumed.

closed 2022-09-03 by evlo

thoj, 2023-02-04:
Hi everyone :)

I have 3 hiking TZE200_byzdayie devices. I have used them before on the Z2M platform but now I am testing it on ZHA after getting a SkyConnect Dongle. I have written some converters for Z2M but I need to learn the syntax of zha before I can resolve any issues on my own.

The quirk file works almost as expected, but i have 2 issues:
1. The switch status is incorrect. When I turn the switch on, the physical device changes to "on" and the status shows "on." However, after a minute the status changes to "off" even though the device is still physically "on." The device accurately displays the correct Wattage and Amperage. 
w. The voltage is approximately double what it should be. I tried adjusting the calculation in the quirks file but it did not fix the problem. I'm probably changing it in the wrong place. Hehe. 

Any Tips?