zigpy/zha-device-handlers #1736: [Device Support Request]

foXaCe, 2022-09-07:
**Is your feature request related to a problem? Please describe.**
Valve controller frankever

**Describe the solution you'd like**
home assistant does not create a switch

   

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

`1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_1n2zev06",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
`

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

[... 377 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
[image]
`

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2022-09-07:
Is it one TRV or somthing else ?
If TRV or thermostat look if the same model is already supported with one other manufacture ID from the tuya trv matrix https://github.com/zigpy/zigpy/discussions/653#discussioncomment-314395.

Also one link to manufacturer or if not "found" link to the seller of the product.

foXaCe, 2022-09-07:
Hello this model

https://frankever.com/fk-v02-zig

MattWestb, 2022-09-08:
OK its one water valve and not one thermostat.
Look in the other device requests is some is using the same commands and try adding your device / manufacture ID to the quirk and see is its working.
@javicalle have helping some users getting some of them working but its needs some work with testing and making logging if the command the device is using.

foXaCe, 2022-09-08:
thank you for your help
but i have no idea i have to do this
I have a lot of zigbee devices and this is the first time I have encountered this problem. 😒

closed 2023-07-10 by github-actions[bot]