zigpy/zha-device-handlers #1738: [BUG] TS011F (by _TZ3000_u5u4cakc) can't set power recovery state

smiba, 2022-09-08:
**Describe the bug**
There is no way for me to change the default power behaviour, the plug is by default set to power off on restore, meaning that the devices connected to it will not power back on without intervention. This is an issue as I use this plug for my freezer too, meaning that on a few seconds of electricity disruption, my freezer will defrost unless I manually enable the plug again on time.

**To Reproduce**
Steps to reproduce the behavior:
1. Unplug plug
2. It will not turn on 

OR:
1. In Clusters select the cluster Tuyz ZBOnOffAttributeCluster (0x0006), try to change the power_on_state (0x8002) attribute. Right now this says "PowerOnState.Off", which means the plug defaults to Off. 0x02 should be "Restore" and 0x01 "On". 
2. Submit either 1 or 2 to the cluster attribute, no change can be observed after getting the attribute. The following error is shown in the ZHA log:

```
[0xB69B](TS011F): set: PowerOnState.Off for attr: 32770 to cluster: 6 for ept: 1 - res: Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)
Set attribute for: cluster_id: [6] cluster_type: [in] endpoint_id: [1] attribute: [32770] value: [PowerOnState.Off] manufacturer: [2] response: [Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)]
```

**Expected behavior**
The plug should have a user friendly option in ZHA for this, zigbee2mqtt does have it for this plug. If no user friendly option is available, I should at least be able to configure The power_on_state attribute manually.

javicalle, 2022-09-09:
That log is from a try to set `PowerOnState.Off` (literally) as value (and is failing). I suppose that you have already tried to use '2' or '1' as value, right? Can you attach the logs trying to set numeric values? 

Can you try the same but with the `start_up_on_off` attribute?

smiba, 2022-09-09:
I tried it another time with the numeric value "2" and the log appears the same as the one I posted

```
[0xB69B:1:0x0006] Failed to convert attribute 0x8002 from PowerOnState.Off (<class 'str'>) to type <enum 'PowerOnState'>: invalid literal for int() with base 10: 'PowerOnState.Off'
[0xB69B:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), manufacturer=2, tsn=228, command_id=<GeneralCommand.Write_Attributes: 2>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xB69B:1:0x0006] Sending request: Write_Attributes(attributes=[])
[0xB69B:1:0x0006] Received ZCL frame: b'\x0c\x02\x00\xe4\x0b\x02\x80'
[0xB69B:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), manufacturer=2, tsn=228, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xB69B:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)
[0xB69B](TS011F): set: PowerOnState.Off for attr: 32770 to cluster: 6 for ept: 1 - res: Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)
```

Here is me retrieving start_up_on_off (0x4003) and then submitting "2" as attribute:

```
[0xB69B:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), manufacturer=2, tsn=54, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xB69B:1:0x0006] Sending request: Read_Attributes(attribute_ids=[16387])
[0xB69B:1:0x0006] Received ZCL frame: b'\x1c\x02\x006\x01\x03@\x86'
[0xB69B:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), manufacturer=2, tsn=54, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xB69B:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x4003, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)])
Read attribute for: cluster_id: [6] cluster_type: [in] endpoint_id: [1] attribute: [16387] manufacturer: [2] response: [None] failure: [{16387: <Status.UNSUPPORTED_ATTRIBUTE: 134>}],
[0xB69B:1:0x0006] Failed to convert attribute 0x4003 from None (<class 'str'>) to type <enum 'StartUpOnOff'>: invalid literal for int() with base 10: 'None'
[0xB69B:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), manufacturer=2, tsn=56, command_id=<GeneralCommand.Write_Attributes: 2>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xB69B:1:0x0006] Sending request: Write_Attributes(attributes=[])
[0xB69B:1:0x0006] Received ZCL frame: b'\x0c\x02\x008\x0b\x02\x80'
[0xB69B:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), manufacturer=2, tsn=56, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xB69B:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)
[0xB69B](TS011F): set: None for attr: 16387 to cluster: 6 for ept: 1 - res: Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)
Set attribute for: cluster_id: [6] cluster_type: [in] endpoint_id: [1] attribute: [16387] value: [None] manufacturer: [2] response: [Default_Response(command_id=2, status=<Status.MALFORMED_COMMAND: 128>)] 
```

javicalle, 2022-09-09:
Ummm, your logs shows:
* invalid literal for int() with base 10: 'PowerOnState.Off'
* manufacturer=2

Are you sure that you are setting the attribute value and not the "Manufacturer code override" ("Anulación del código del fabricante" in spanish))?

[image]

MattWestb, 2022-09-09:
The 0x400x attributes is Zigbee standard but then tuya is not  doing real Zigbee devices (only some few is certificated) then is using there own attributes and its implanted here:
https://github.com/zigpy/zha-device-handlers/blob/b98948e7b6538ab31d87330bba1793e0eb65de11/zhaquirks/tuya/__init__.py#L864-L871

Also you have potting the attribute value in the wrong feeld (i have doing the same many times) `manufacturer=2`.

Try reading the 0x8002 attribute and if you is getting somthing that  is not none try writing 0, 1 or 2 and reading it back.

smiba, 2022-09-09:
> Are you sure that you are setting the attribute value and not the "Manufacturer code override" ("Anulación del código del fabricante" in spanish))?

Oh oh, I feel a little bit stupid right now. That fixed it. 
It's now reporting PowerOnState.LastState

So sorry for wasting everyones time!

closed 2022-09-09 by smiba

javicalle, 2022-09-09:
Don't worry. With an UI control it would be more easy, but not implemented yet.
There is still a lot to improve, but it is a lot of work and few hands.