zigpy/zha-device-handlers #1739: [Device Support Request] New manufacturerName for Zigbee Energy Meetering Device RTX ZCB16-2P (_TZE200_fsb6zw01)

tomkusz, 2022-09-08:
I have a problem with the correct recognition of Tuya Energy Device - RTX ZCB16-2P (device signature: _TZE200_fsb6zw01) in ZHA. After adding the device to ZHA, only the Lqi and Rssi entities are recognized. 

I would like to receive only the results of the device measurements like in another meetering device, such as _TZE200_ewxhg6o9.
I do not care about the ability to control the switch.

[image]

Device signature:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_fsb6zw01",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Diagnostic information:
[zha-c05bf04a53c93d36e3c8cf73f1e975a5-_TZE200_fsb6zw01 TS0601-e0909940fd8bb3308f3bd290c68b70a3.json.txt](https://github.com/zigpy/zha-device-handlers/files/9531093/zha-c05bf04a53c93d36e3c8cf73f1e975a5-_TZE200_fsb6zw01.TS0601-e0909940fd8bb3308f3bd290c68b70a3.json.txt)

javicalle, 2022-09-09:
It seems a variant from the `TuyaPowerMeter`.

There is a possible quirk for that device:
```python
class TuyaPowerMeter_TI(TuyaSwitch):
    """Tuya power meter device."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.switch_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [
            ("_TZE200_fsb6zw01", "TS0601"),
        ],
        ENDPOINTS: {
            # "profile_id": 260, "device_type": "0x0051",
            # "in_clusters": ["0x0000", "0x0004", "0x0005", "0x000a", "0xef00"],
            # "out_clusters": ["0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    Time.cluster_id,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }
```

Enable the custom quirks configuration, copy the current `ts0601_din_power.py` inside the folder and edit it adding the proposed class at the end of the file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py

Save changes, delete cache and restart.
Check if the device signature has changed to the new `TuyaPowerMeter_TI` and look to any relevant info in the logs.

tomkusz, 2022-09-11:
Thanks a lot for the assistance. Now it's much better then before, but some entities not working propely yet ...  
[image]

tomkusz, 2022-09-11:
"Nieznany" means unknown ...

javicalle, 2022-09-11:
Some power metering entities needs to be working some time with some load to register values.
If not working as expected, attach the debug logs from the device. Also the device specs or capabilities (for example the vendors site) would be helpful to know what is reporting the device.

javicalle, 2022-10-22:
That is the device diagnostic, what are needed is the related HA logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

It can be very verbose so try to identify the device reports. One thing that you can do is to switch on and off the plug when you are trying to catch the logs. Usually device reports its values when doing.

From the attached device diagnostic attached I can see that Metering and ElectricalMeasurement clusters have values. You don't see any value in HA?