zigpy/zha-device-handlers #1743: [Device Support Request] Various HUE & Paulmann devices

popy2k14, 2022-09-11:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
Infos about if the following devices are compatible with ZHA,
and if not, it's an device request.

<details>

<summary>Device signature</summary>

those are just device id's from deconz API, hope this helps to identify the devices
```
LCT015
LTW012
LTW010
LCT007
LTW013
LWG004
LCL001
500.48
Switch Controller (Paulmann)
```
</details>

**Additional context**
Planing to move from deconz to ZHA, and want to check compatibility first.
Already checked: https://zigbee.blakadder.com/zha.html and removed the compatible ones from the list.

kryanth, 2022-09-13:
The Philips Hue devices (LCT*, LWT*, LTW*) just work. 

LCL001 is listed on blakadder as working, it probably just does.

popy2k14, 2022-09-13:
ok, thx.
ANy hints about the paulmann devices:

500.48
500.43 Switch Controller (Paulmann)

?

jerrm, 2022-09-15:
If you have more than a handful of devices, do yourself a favor and get a second coordinator.  Keep both running in parallel and move devices at your leisure instead of a rushing to do it all at once.

popy2k14, 2022-09-15:
thx for the hint.
already don this way and got an sonoff 3.0 dongle plus.

TheJulianJES, 2023-01-04:
Like mentioned, basically all Hue devices are supported.
No idea about the Paulmann devices. If they follow standards, they'll work.

If you have issues with those devices, open a new issue and provide the requested information (for which you need to have the device in ZHA).

closed 2023-01-04 by TheJulianJES