zigpy/zha-device-handlers #1746: TS0041 TS0044 TS0042 - missing presses and LED flashing

evlo, 2022-09-12:
**Describe the bug**
TS0041 TS0044 TS0042
 - sometimes does not react and run associated automation on button press, even thought LED flashes
 - LED(s) flashing at some interval about once and hour or so even when no button is pressed

TS004F does not have that issue

**To Reproduce**
have TS0041 TS0044 TS0042

**Expected behavior**
react on press all the time and do not flash leds Lrandomly"

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_dfgbtub0",
  "model": "TS0042",
  "class": "zhaquirks.tuya.ts0042.TuyaSmartRemote0042TO"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

</details>

Seems to be announce issue? Same behaviour with zigbee2mqtt. With tuya gateway it sometimes do not react on press, or rather reacts with huge delay, but never fleshes leds.

MattWestb, 2022-09-12:
The TS004F is having problem with default response that must being sent in the wrong direction for EP 2-4 for getting the communication working 100% and its not fixed in ZHA and i think the other TS004X can having the same problem if not getting the DR not OK.

I dont have and TS004X devices only TS 004F that is working little different.

If you having the hardware for sniffing it shall being great if getting it sniffed so can see how it working with tuya ZBGW !!

evlo, 2022-09-12:
I Havel no idea how to sniff ZigBee communication. Maybe sdr? Is there some guide somewhere? Maybe fixing this And the eink devices would be lesa pain then spelling And replacing with others tahat might od might not work :)

MattWestb, 2022-09-12:
Some info: https://github.com/zigpy/zigpy/wiki/Sniffing-Zigbee-traffic
Also Z2M is having some and also deCONZ with sniffing firmware (EZSP is using the same firmware as the normal coordinator).

evlo, 2022-09-13:
https://github.com/Koenkk/zigbee2mqtt/issues/8072 < linking this here as there are some wireshark caps. already

MattWestb, 2022-10-01:
I think that the quirk is need being updated to sending tuya magic for the device working OK wot we have seen in the linked Z2M issue and sniff from the tuya ZBGW.

cordvision, 2023-02-17:
I'm having the same issue with those switches (I have two TS0044). After inactivity for several hours, the first pressing is ignored. It is as if the device goes into a sleep mode and you have to press a button to wake it up.
There seams to be a sulution available for Zigbee2MQTT (see here: https://community.home-assistant.io/t/tuya-ts0044-first-action-ignored/475940)
I hope this can be resolved as this switch is very popular and affordable. It's very reliable besides the mentioned issue...

MattWestb, 2023-02-17:
@javicalle Is it working putting `EnchantedDevice` here 
https://github.com/zigpy/zha-device-handlers/blob/8122e9283589f4fff585e4cf03892bf87587e345/zhaquirks/tuya/ts0044.py#L27
or must is being added in all device classes ?
And how to do it right ?
https://github.com/zigpy/zha-device-handlers/blob/8122e9283589f4fff585e4cf03892bf87587e345/zhaquirks/tuya/ts0044.py#L46

javicalle, 2023-02-17:
> @javicalle Is it working putting `EnchantedDevice` here
> 
> https://github.com/zigpy/zha-device-handlers/blob/8122e9283589f4fff585e4cf03892bf87587e345/zhaquirks/tuya/ts0044.py#L27
> 

I bet that can be done adding in the parent class:
```python
class Tuya4ButtonTriggers(EnchantedDevice):
    """Tuya 4-button remote device triggers."""
```

Maybe even we can define the `CustomDevice` in the parent class and remove from the extended ones, but must be double checked to be sure.
<details>

```python
class Tuya4ButtonTriggers(EnchantedDevice, CustomDevice):
    """Tuya 4-button remote device triggers."""

```

```python
class TuyaSmartRemote0044TI(Tuya4ButtonTriggers):
    """Tuya 4-button remote device with time on in cluster."""

```

```python
class TuyaSmartRemote0044TO(Tuya4ButtonTriggers):
    """Tuya 4-button remote device with time on out cluster."""

```
```
...
```

</details>

MattWestb, 2023-02-18:
@cordvision can you testing this quirk for TS0044 ?
You need configure local quirks and unzipping the quirk in the directory and restart HA.
[ts0044.zip](https://github.com/zigpy/zha-device-handlers/files/10773655/ts0044.zip) 
 Instruction https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

Then ZHA is started and stable deleting one of your remotes from the device card and waiting one minute and adding it new (resetting the remote).

The trick is the system is sending tuya magic only then the device is not known for the system so its must neing one new paring not updating one in the system known device.

Pleas testing and reporting back !!

@javicalle i finding the same but its needs very mush testing and we dont have enough device for doing it safely and it can be that we is needing it for some devices that is not liking getting the magic. 
I also thinking the most importing is getting all sleepers getting the magic and then adding it for router devices then its needed (we knowing that some routers is also needing it but very likely most or all sleepers is doing it).

cordvision, 2023-02-20:
@MattWestb 
Thank you... I'm testing it as we speak and I will report back....

cordvision, 2023-02-20:
@MattWestb 
I already have a custom quirk running successfully. After adding th ts0044 quirk, all my zigbee devices stopped running and I'm getting the following error: 

Logger: homeassistant.config_entries
Source: zha_quirks/ts0044.py:47
First occurred: 19:26:21 (1 occurrences)
Last logged: 19:26:21

Error setting up entry ConBee II - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2492391-if00, s/n: DE2492391 - dresden elektronik ingenieurtechnik GmbH - 1CF1:0030 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/zha_quirks/ts0044.py", line 47, in <module>
    class TuyaSmartRemote0044TI(CustomDevice, Tuya4ButtonTriggers):
TypeError: Cannot create a consistent method resolution
order (MRO) for bases CustomDevice, Tuya4ButtonTriggers

After deleting the quirk and re-starting again... everything is back to normal with no errors.

TheJulianJES, 2023-02-20:
`Tuya4ButtonTriggers` now inherits `EnchantedDevice` (which shouldn't be the case IMO) and that also inherits from `CustomDevice`.
Yet `TuyaSmartRemote0044TI` and all other quirks also inherit `CustomDevice` again. This doesn't work.

Either `CustomDevice` can be removed everywhere where `Tuya4ButtonTriggers` is also used, or just remove `EnchantedDevice` from `Tuya4ButtonTriggers` again (they are device automation triggers and not a device after all).
Then, replcae `CustomDevice` with `EnchantedDevice` in all those quirks.

cordvision, 2023-02-20:
@TheJulianJES 
I have to be honest, I don't understand a word you said.... I don't even know where to start. I'm very much a beginner with home assistant (I can follow step by step instructions, that is how I installed another quirk for a motion sensor I got, but I don't understand at all what you were describing).

TheJulianJES, 2023-02-20:
It was more directed towards MattWestb/javicalle, but here's a modified version that *might work*. You still need to remove + re-pair the Tuya devices. Let me know if this works or crashes again.

Try this: [ts0044_custom.py.txt](https://github.com/zigpy/zha-device-handlers/files/10787208/ts0044_custom.py.txt)
(Remove the `.txt` from the file name, so it just has `.py` at the end.)

cordvision, 2023-02-20:
@TheJulianJES 
Thank you! I just uploaded it and re-added one of my switches. So far everything works as expected. I'll report back in a few days (I can't easily replicate the sleep issue as it sometimes takes 12+ hours of inactivity for the device to go to "sleep"). Fingers crosse :-)

javicalle, 2023-02-20:
Thanks Julian. 
Sorry for all the troubles. 😓

TheJulianJES, 2023-02-20:
No worries, happy to help where I can (with Tuya stuff)

cordvision, 2023-02-22:
@TheJulianJES
I just wanted to report back: Unfortunately it didn't fix the issue. In fact, it actually got worse. Not only was the first button press still ignored after some time of inactivity, but I had the switches stop responding entirely a few times. 

On I side-note: I also noticed that the "battery level" reporting (on both of my switches) started jumping from 100% to 70% and back (see here: https://prnt.sc/E-u-KtouHNRQ). This coincided on both switches with the time I loaded the "custom quirk".  The switches are both less than 3 weeks old and I did put new batteries in when I got them.

Any idea what the problem might be? Also, can somebody confirm that the switch indeed works properly with Zigbee2MQTT?

cordvision, 2023-03-25:
Just wanted to report back as promised: So far my battery values have not updated. They remain the same.... not sure if this reflects the actual amount of battery I have left.

TheJulianJES, 2023-03-25:
Possibly related:
- https://github.com/zigpy/zha-device-handlers/issues/2227
- https://github.com/zigpy/zha-device-handlers/pull/2288

Rojoss, 2023-09-13:
Hi, I'm curious if there is any solution for this issue since it's still happening for me and it's super anoying.
I have quite a lot of these switches so replacing them all will be quite expensive.
I have the same issue as evlo where the buttons are missing presses when the devices go to sleep after some time.

I have the following switches which all have the same problem.
`TS0041`, `TS0042`, `TS0043` & `TS0044`

When I expand Zigbee info on the device it does say that there's a quirk.
Does this maybe mean that it's a known issue or does this mean there is some sort of patch available?
[image]

<details><summary>Device signature</summary>
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0001",
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_5e235jpa",
  "model": "TS0042",
  "class": "zhaquirks.tuya.ts0042.TuyaSmartRemote0042TO"
}
```
</details>

<details><summary>Diagnostics</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.08.3",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "dwains_dashboard": {
[... 311 lines cut ...]
        "device_type": {
          "name": "REMOTE_CONTROL",
          "id": 6
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0006": {
            "endpoint_attribute": "TS004X_cluster",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

MattWestb, 2023-09-13:
@Rojoss Pleas enabling debug loging for ZHA and zigpy and posting the log then its happening so the devs can looking if somthing going wrong.
Also make sure running latest HA the some fixes is implanted for this devices that looks have 2 "working modes" 1=Deep sleeping and missing events or 2=Good responding but draining  batteries.

closed 2024-03-18 by github-actions[bot]

sanderlv, 2024-12-04:
Is this going te be resolved? I have the axact same issues after update of Z2m with 4 of these ts0044 devices...

sanderlv, 2025-01-03:
Indeed, I have 4(!) of these and they always worked fine. Now all 4 have this exact same issue...

provinzio, 2025-03-31:
I had the same issues for months and replaced 2032 batteries with fresh 2025 batteries. It's currently working as expected.

sanderlv, 2025-03-31:
Batteey is. Not my issue.

Since z2m 2.x all chamged and using mew outputs from device it started to work again.

It needed some change.