zigpy/zha-device-handlers #1747: Moes Model ZSS-ZK-THL - TS0222 _TYZB01_kvwjujy9

evlo, 2022-09-12:
I was wondering if anyone managed to device above mentioned reliably?

https://zigbee.blakadder.com/Moes_ZSS-ZK-THL.html
https://www.zigbee2mqtt.io/devices/ZSS-ZK-THL.html
https://github.com/zigpy/zha-device-handlers/issues/961

In my experience it works a bit better with zigbee2mqtt, also with tuya hub it still sometimes read zeros.

Since there is E1 reported as temperature on display or the battery icon, or exclamation mark on the display, and display flashing "black/white" even with fresh battery etc. I wonder if it might be super sensitive to batteries ore something.

Also do you think that in the compatibility databases devices like this should really be considered unsupported since the behaviour is so bad?

almico, 2022-11-14:
I keep giving this device a chance every few months. Both with Tuya gateway, and with ZHA.
Every time, like this time, I have to take the battery out and put the device back in its box. Hoping for some future miracle 😭

evlo, 2022-11-14:
It is still mindboggling that devices like this get into production.

igorsantos07, 2023-02-24:
I was never able to get it to work properly with the HA Zigbee mesh, but since this is pretty much my only way to get lux without messing with ESPHome, I decided to throw more money onto AliExpress and tried to buy a Tuya gateway. Pairing them together with my Tuya account does the trick - and yep, I now have a second Zigbee network running on my office, _just for this freaking sensor_ 🙄 and data do go through Tuya cloud 🫠 but at least it's pretty much instant.

Ddetails were discussed confusingly to exhaustion at https://github.com/zigpy/zha-device-handlers/issues/961

evlo, 2023-02-24:
And battery life is ok on tuya gw for you?

igorsantos07, 2023-02-24:
Can't say for sure... I popped in a used battery and I just started using
it 2 days ago, so...

Igor Santos
-- Desenvolvedor Web
[enviado do meu celular]

closed 2023-08-30 by github-actions[bot]

evlo, 2023-09-01:
I do only use LYWSD03MMC and CGG1-Mijia now. I'm sharing this as those are in my test best devices of this kind.