zigpy/zha-device-handlers #1750: [BUG] Moes MS-105Z 1 Gang Dimmer module not dimming as expected via Home Assistant (TS0601 _TZE200_la2c2uo9)

wutr, 2022-09-13:
**Describe the bug**
I am using an EFR32MG21 based Sonoff Zigbee 3.0 USB Dongle plus ZBDongle-E with ZHA in Home Assistant. I have added a 1 gang [MoesGo Smart Dimmer Module MS-105Z](https://www.amazon.co.uk/MoesGo-ZigBee3-0-Required-Wireless-Control/dp/B09724V6YB) with a switch and one dimmable LED connected for testing. I think it is the same as this: https://zigbee.blakadder.com/Moes_MS-105Z.html It is discovered as a light named TS0601 _TZE200_la2c2uo9 and it has a switch and a dimmer slider in the Home Assistant UI.

I can manually control the brightness by keeping the physically connected switch on until the desired brightness is reached, so the LED dimming circuit and the attached LED bulb works.

However, when I try to change the brightness through Home Assistant, the following happens:

- When the switch entity in HA is already on and I change the brightness slider, the light turns off with a short transition. The switch entity state remains 'ON'
- When I then turn off the switch entity, and turn it on again, the light turns on at the brightness that is currently on the slider (I think there might be a transition, but I'm not sure as it is so bright at higher brightness levels)
- Clicking at a higher level on the brightness slider turns off the light and most of the time also the switch entity
- Dragging the brightness slider turns the light off, but leaves the switch entity 'ON'

**To Reproduce**
See description of the problem

**Expected behavior**
The dimmer's brightness should be directly controllable via the brightness slider. It should not require turning the switch entity off and on again. Any interaction with the slider should leave the switch entity in its previous state.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_la2c2uo9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.60",
    "supervisor": "2022.08.6",
[... 276 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

wutr, 2022-09-24:
It works as expected with zigbee2mqtt

alitz, 2022-09-28:
I am having the exact same issue with ZHA. Please help.

steve088, 2022-10-05:
I am also having this exact issue with ZHA.

dobromir-hristov, 2022-10-10:
Same happens for the two gang version, just installed one.

alitz, 2022-10-15:
I can confirm that the issue was resolved with the latest update of home assistant version 2022.10.4

steve088, 2022-10-20:
I can also confirm that that this issue has been resolved with an update to Home Assistant 2022.10.x

javicalle, 2022-12-11:
Reported as fixed.
Please, reopen if the problem persists.

alitz, 2022-12-11:
With the latest version of home assistant 2022.12.1, the issue is back.

javicalle, 2022-12-11:
Someone able to test this fix?:
* https://github.com/zigpy/zha-device-handlers/issues/1554#issuecomment-1345426616

NdK73, 2022-12-11:
I could test the fix, but I'd need some guidance (still don't grock containers very well).

closed 2022-12-12 by dmulcahey

NdK73, 2022-12-12:
> as completed via #1999

Just updated to latest core. Not better than before :( Still can't dim,
just switch.

dmulcahey, 2022-12-12:
Will be in .4 it isn’t in .3

PR #1999 (2022-12-12): Revert PR #1984
Users are reporting bad behavior with the dimming function after PR #1984:
* https://github.com/zigpy/zha-device-handlers/issues/1554#issuecomment-1343399518

Reverting to fix issues.

Fix: #1996
Fix: #1750