zigpy/zha-device-handlers #1752: [BUG] E1812 Ikea shortcut button does not fire zha_event

NdK73, 2022-09-13:
**Describe the bug**
Similar to #1516 but different context.

I'm using supervised HA where I confugured ZHA to access a [ZigStar LAN Gateway](https://zig-star.com/projects/zigbee-gw-lan/).
I can succesfully bind a Sonoff SNZB-04 (door/window sensor) and a RSH-ZigBee-SC04 4-button scene switch. Both devices report events to HA.
But both the Ikea E1812 I tested seems to have troubles both during discovery and after binding. I once saw events while testing a CC2531-based bridge, but only for a short time. (Actually, while collecting debug info, one of the buttons started generating events... hope it continues)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ZHA integration
2. Put E1812 in pairing mode (4 clicks on bind button, red LED starts blinking)
3. Click on add device
4. The logs report:
 ```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
New device 0xd62c (8c:f6:81:ff:fe:2b:c6:6f) joined the network
[0xd62c] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD62C ieee=8c:f6:81:ff:fe:2b:c6:6f is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81,\xd6o\xc6+\xfe\xff\x81\xf6\x8c\x80'
[0xd62c:zdo] ZDO request ZDOCmd.Device_annce: [0xD62C, 8c:f6:81:ff:fe:2b:c6:6f, 128]
Tries remaining: 3
[0xd62c] Requesting 'Node Descriptor'
Tries remaining: 2
[0xd62c] Extending timeout for 0x8d request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD62C ieee=8c:f6:81:ff:fe:2b:c6:6f is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x8d\x00,\xd6\x02@\x80|\x11RR\x00\x00,R\x00\x00'
[0xd62c] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xd62c] Discovering endpoints
Tries remaining: 3
[0xd62c] Extending timeout for 0x8e request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD62C ieee=8c:f6:81:ff:fe:2b:c6:6f is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x8e\x00,\xd6\x01\x01'
[0xd62c] Discovered endpoints: [1]
[0xd62c] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xd62c:1] Discovering endpoint information
Tries remaining: 3
[0xd62c] Extending timeout for 0x8f request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xD62C ieee=8c:f6:81:ff:fe:2b:c6:6f is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x8f\x00,\xd6"\x01\x04\x01 \x08\x01\x06\x00\x00\x01\x00\x03\x00\t\x00 \x00\x00\x10\x07\x03\x00\x04\x00\x06\x00\x08\x00\x19\x00\x02\x01\x00\x10'
[0xd62c:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2080, device_version=1, input_clusters=[0, 1, 3, 9, 32, 4096], output_clusters=[3, 4, 6, 8, 25, 258, 4096])
[0xD62C:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=144, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xD62C:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0xd62c] Extending timeout for 0x90 request
[0xD62C:1:0x0000] Received ZCL frame: b'\x18\x90\x01\x04\x00\x00\x42\x0E\x49\x4B\x45\x41\x20\x6F\x66\x20\x53\x77\x65\x64\x65\x6E\x05\x00\x00\x42\x17\x54\x52\x41\x44\x46\x52\x49\x20\x53\x48\x4F\x52\x54\x43\x55\x54\x20\x42\x75\x74\x74\x6F\x6E'
[0xD62C:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=144, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xD62C:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='IKEA of Sweden')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TRADFRI SHORTCUT Button'))])
[0xd62c] Read model 'TRADFRI SHORTCUT Button' and manufacturer 'IKEA of Sweden' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, alarms:0x0009, poll_control:0x0020, lightlink:0x1000] out=[identify:0x0003, groups:0x0004, on_off:0x0006, level:0x0008, ota:0x0019, window_covering:0x0102, lightlink:0x1000] status=<Status.ZDO_INIT: 1>>
[0xd62c] Discovered basic device information for <Device model='TRADFRI SHORTCUT Button' manuf='IKEA of Sweden' nwk=0xD62C ieee=8c:f6:81:ff:fe:2b:c6:6f is_initialized=True>
Device is initialized <Device model='TRADFRI SHORTCUT Button' manuf='IKEA of Sweden' nwk=0xD62C ieee=8c:f6:81:ff:fe:2b:c6:6f is_initialized=True>
Checking quirks for IKEA of Sweden TRADFRI SHORTCUT Button (8c:f6:81:ff:fe:2b:c6:6f)
Considering <class 'zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn2'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn'>
Found custom device replacement for 8c:f6:81:ff:fe:2b:c6:6f: <class 'zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn'>
'button' component -> 'ZHAIdentifyButton' using ['identify']
'sensor' component -> 'Battery' using ['power']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 0xD62C:8c:f6:81:ff:fe:2b:c6:6f entering async_device_initialized - is_new_join: True
device - 0xD62C:8c:f6:81:ff:fe:2b:c6:6f has joined the ZHA zigbee network
[0xD62C](TRADFRI SHORTCUT Button): started configuration
[0xD62C:ZDO](TRADFRI SHORTCUT Button): 'async_configure' stage succeeded
[0xD62C:1:0x0000]: Configuring cluster attribute reporting
Error handling '_save_attribute' event with (8c:f6:81:ff:fe:2b:c6:6f, 1, 0, 5, 'TRADFRI SHORTCUT Button') params: FOREIGN KEY constraint failed
[0xD62C:1:0x1000] Received ZCL frame: b'\x19\x93\x41\x01\x00\x00'
[0xD62C:1:0x1000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=147, command_id=65, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xD62C:1:0x1000] Decoded ZCL frame: LightLinkCluster:get_group_identifiers_rsp(total=1, start_index=0, group_info_records=[])
[0x0000:2] Cannot add 0x0000 group, no groups cluster
[0x0000:1] Cannot add 0x0000 group, no groups cluster
[0xD62C:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xD62C:1:0x0001] Received ZCL frame: b'\x08\x97\x07\x00'
[0xD62C:1:0x0001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=151, command_id=7, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xD62C:1:0x0001] Decoded ZCL frame: PowerConfiguration1CRCluster:Configure_Reporting_rsp(status_records=[ConfigureReportingResponseRecord(status=<Status.SUCCESS: 0>)])
[... 802 lines cut ...]
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0102",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI SHORTCUT Button",
  "class": "zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn"
}
```

</details>

MattWestb, 2022-09-14:
`<Status.NWK_NO_ROUTE: 205>` and `Extending timeout for 0x9d request` = the device is sleeping.
Try waking  the device up then its paring by pressing the button every 2 seconds then its looks its failing in sleep and not receiving the configuration OK.

Also if its paired OK do one reconfigure this device from the device card and not  forgetting waking  it up then sending  things to it.

I can see you is having more sleepers but do you also have some routers that can helping the network preforms better ?
Also wot coordinator is you using ?

NdK73, 2022-09-15:
Seems the anwer via mail got lost. But I now have news.

Just before re-pairing, I retested the button that yesterday didn't work and now it's working. The only change was a system upgrade (including a HA update to 2022.9.4) and a reboot. None of those changes was about the bridge, and previously I didn't see button presses in bridge's console. I'm really confused. But maybe the hint could help others: update and reboot! And wait the next day before trying do pair again.

Tks and sorry for the noise.

NdK73, 2022-10-11:
I already tried pressing the button at regular intervals w/ no luck :(
The (currently...) working one did not require any presses, and its log is
the one I attached at the end of the report.

I also read that newer fw on the button only allows binding to another
device, so I also tried binding it to the coordinator from the device page
in HA before retrying pairing.

The coordinator is a ZigStar Lan Gateway (link at the beginning of the
report).

Tks,
Diego

TheJulianJES, 2023-01-04:
Is there still an issue here? IKEA devices often need to be woken up again after re-pairing to reconfigure them.

So, open the device page -> click reconfigure device and shortly after starting the reconfiguration, press a button on the remote to wake it up.

NdK73, 2023-01-04:
issue seems device-dependant.
One of the buttons is working w/o issues. The other one gets paired and then disappears. Sometimes it works for some time but suddenly stops.
Also tried pressing the action button, start reconfiguration, press the button again every few seconds to keep it awake: reconfiguration failed anyway.
**BUT** if I single-press the pairing button just before starting reconfiguration, it fails very early **and events start to come thru**.

MattWestb, 2023-01-04:
Its can being problems if paring direct with the coordinator or one router (depends of the coordinator type).
And IT coordinator is not setting up them OK then having them as direct children.
Try paring thru on router and its very likely working OK and also reconfigure after its being paired.

NdK73, 2023-01-04:
Actually I couldn't pair it with various routers, but only with coordinator. I now moved it where it does have to go thru 2 or 3 routers and seems it's still working. But pairing is a nightmare.

aykevl, 2023-02-07:
Had a similar issue. This did the trick for me:
- remove the battery
- press the button a few times (probably unnecessary but that's what I did)
- re-insert the battery
- reconfigure, press the button right after starting reconfiguring

The reconfigure went really fast, exited with an error, but it works nonetheless. I've tried removing/repairing/reconfiguring many times and removing the battery was the thing that finally did the trick for me.

donoghdb, 2023-02-08:
@aykevl I have the same issue, even got IKEA to send another one but neither of them i can detect events. I have tried custom quirks but nothing is working :(

aykevl, 2023-02-10:
I've now switched over to zigbee2mqtt and they got recognized instantly, no issues whatsoever.

donoghdb, 2023-02-10:
Thank you @aykevl. I now have two working buttons :). ZHA is the issue.

malfurion, 2023-03-06:
Hey there,

i have the same issue.

i've switched vom deconz to zha with my conbee 2 stick.

i use 3 tradfri shortcut buttons but only 1 is working after i tried hours of reconfiguration and so on but the 2 other buttons won't work.

does someone figured out a method to get them work ?

MattWestb, 2023-03-06:
After getting the the device paired you shall doing one reconfigure from the device card and waking  the device up by pressing one key so its not sleeping and can receiving the commands.

It can being tricky getting it working and with the new 24.5 firmware for IKEA controllers its the best way taking the battery out and waiting one half minute and putting it back then start the reconfigure at the **same time** (I know its not easy, _im a Man_ ;-).

radian997, 2023-03-13:
I had the same issue with pairing Shortcut Button with my ZHA and CC2652P. I made several attempts to pair the button. The successful attempt was when I pressed the button regularly (every 1-2 seconds) during pairing and ~2 minutes after pairing. Now I see button events in the Home Assistant logbook

MattWestb, 2023-03-13:
@radian997 I think TI is having one bug that is making problem adding all Zigbee 3 controllers / sleeping end devices **direct** to the coordinator. If having some router devices try joining them thru them and is very likely working better.

NdK73, 2023-03-13:
Already tried both workarounds (also combined):
 - press button every second: 50% of the times binding or configuration fails
 - bind to a router: I have some Tuya switches and a reprogrammed CC2531 dongle, no change

Other devices (including a Tuya 4-button scene switch) work w/o issues.
One of the two buttons also worked for some months then decided to not fire events.
My Z2M-managed dongle (also CC2652P) w/ no routers attached ( = buttons are bound to the coordinator) sees events from both buttons. That makes me think it's a ZHA/zigpy issue, maybe in a quirk.

NdK73, 2023-05-20:
Seems the only procedure that seems to work quite often is:
1) delete shortcut button
2) **restart HA**
3) re-bind the button

Without step 2, events do not appear.

HIH.

MattWestb, 2023-05-20:
After new paring IKEA controller always do one reconfigure for getting the reporting working OK and also dot forgetting binding the group for older for firmware / devices for latest versions if searing lights with them.

closed 2023-11-23 by github-actions[bot]