zigpy/zha-device-handlers #1757: [Device Support Request] aqara Smart Radiator Thermostat E1

garv3, 2022-09-15:
**Is your feature request related to a problem? Please describe.**
The device offers some functionality that is not supported.
climate.turn_off does not switch the device into the off state which closes the valve (aka summer mode). 
The device probably does not support most havac modes since it's just a radiator valve but these modes are displayed.
Linking the device to another thermometer is possible but not supported by zha.
Battery state is unknown.
and many more

**Describe the solution you'd like**
All features should be supported.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0201",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0201",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.airrtc.agl001",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Cannot be downloaded (works for other devices). See logs!
```

</details>

<details>
<summary>Additional logs</summary>

```
Logger: aiohttp.server
Source: components/zha/diagnostics.py:153
First occurred: 18:20:42 (14 occurrences)
Last logged: 18:25:01

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 234, in get
    data = await info[sub_type.value](hass, config_entry, device)
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 101, in async_get_device_diagnostics
    device_info[CLUSTER_DETAILS] = get_endpoint_cluster_attr_data(zha_device)
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 123, in get_endpoint_cluster_attr_data
    ATTR_IN_CLUSTERS: {
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 126, in <dictcomp>
    **get_cluster_attr_data(cluster),
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 152, in get_cluster_attr_data
    UNSUPPORTED_ATTRIBUTES: {
  File "/usr/src/homeassistant/homeassistant/components/zha/diagnostics.py", line 153, in <dictcomp>
    f"0x{cluster.find_attribute(u_attr).id:04x}": {
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 188, in find_attribute
    return self.attributes[name_or_id]
KeyError: 900

```

</details>

**Additional context**
Nothing to add. Thanks in advance!

garv3, 2022-10-14:
No news on this?

TobiasBales, 2022-11-01:
Afaik Zigbee2MQTT has better support (but I still have not really understood zigbee 😓) so it might be able to port some of the config?
I am willing to help/debug/provide whatever is needed but I have no idea where to get started really :/

xatr0z, 2022-11-17:
FYI, I bought this device because [Home assistant explicitly recommends](https://www.home-assistant.io/blog/2022/10/03/short-term-solutions-save-energy-and-money-europe/) this specific device:
`This TRV can be controlled directly from Home Assistant using Zigbee or via the Aqara hub. `

Not that its any of your business what Paulus writes in his blogs, but I'm expecting more people with this issue...

hraschan, 2022-11-27:
Is there a documentation on how to implement this stuff or a new device support? I would like to help and write the implementation.

dmulcahey, 2022-11-27:
The readme in this repo

garv3, 2022-12-08:
Still nothing. What a shame. I think, I will switch to zigbee2mqtt as soon as the stick for matter arrives...

jzielke84, 2022-12-30:
I'd really appreciate this, since I have multiple thermostats - all without battery status.

xF4m3, 2023-01-01:
I do have to say i wanted to use zha since it seems to be the newer go to thing and also looked more refined gui wise, but I switched to zigbee2mqtt like many others did and the range and stability as well as the device support is just way better as of now. I doubt this will change that soon.

jymdman, 2023-01-11:
I have created a quirk that is working fine for me. I have uploaded a gist so more people can try it out and if its stable maybe this can be included in offical build.

[https://gist.github.com/jymdman/3761b343fe15deb222c430b93e44b08a](https://gist.github.com/jymdman/3761b343fe15deb222c430b93e44b08a)

No settings is showing up in Home Assistant, but you can control it with cluster commands.
I also have created a blueprint for using external sensor temperature with the thermostat.

[https://gist.github.com/jymdman/2ff42543ae4b10fae3217259ad97ad6b](https://gist.github.com/jymdman/2ff42543ae4b10fae3217259ad97ad6b)

garv3, 2023-01-13:
> I have created a quirk that is working fine for me. I have uploaded a gist so more people can try it out and if its stable maybe this can be included in offical build. [...]

What exactly is working for you? I have added the quirk, restarted HA an reconfigured the device but nothing changed. Is it necessarry to relink the device?

jymdman, 2023-01-13:
@garv3 Yes, you have to remove and add the device for the quirk to work.

ostojan, 2023-01-17:
@jymdman Does turning trv off work for you? I tried your quirk but turning thermostat off doesn't work. Status in HA is set to off, but device doesn't turn off.

garv3, 2023-01-18:
> @jymdman Does turning trv off work for you? I tried your quirk but turning thermostat off doesn't work. Status in HA is set to off, but device doesn't turn off.

At least for me this does not work using the TRV option. You could change the TRV mode using the OppleCluster (in). You will have to set the system_mode attribute to 0 to turn the valve off (summer mode) or 1 to turn it on again (normal operation).
This cannot be done from the UI. You will have to use an automation, node-red or a script.

crypticviper, 2023-01-25:
Just have a question. Is it possible to set the target temperature for this device based on the time trigger? I don't get any options to do this with automation (from the HA UI).

I get mode options in the UI and I don't think these modes even work. And battery is still unknown for me even after using the quirk mentioned above.

[image]

> At least for me this does not work using the TRV option. You could change the TRV mode using the OppleCluster (in). You will have to set the system_mode attribute to 0 to turn the valve off (summer [...]

MattWestb, 2023-01-25:
Then doing the automation you must selecting `service` and `Climate: Set temperature` as action.
With `device` you dont have the climate functions only normal switching functions.

msolomos, 2023-01-29:
It's a shame such a kind of product don't have a full support from ZHA. 
I don't really mind for the rest of the modes that are showing...as I just don't use them, but the battery state is a problem.
Thumbs Up for battery level!

TheJulianJES, 2023-01-29:
@msolomos Did you check with Home Assistant 2023.2.0 betas?

msolomos, 2023-01-29:
> @msolomos Did you check with Home Assistant 2023.2.0 betas?

I haven't no. I don't really try beta versions to be honest as I had a bad experience at the past. But does it worth?  
Has it got a fix ?? Does this applies also to Aqara E1 curtain driver?

margau, 2023-01-29:
> > @msolomos Did you check with Home Assistant 2023.2.0 betas?
> 
> I haven't no. I don't really try beta versions to be honest as I had a bad experience at the past. But does it worth?  
> Has it got a fix ?? Does this applies also to Aqara E1 curtain driver?
> 

In this thread, the merged https://github.com/zigpy/zha-device-handlers/pull/2068 is mentioned, adding Support for battery but only the thermostat.

msolomos, 2023-01-29:
> In this thread, the merged #2068 is mentioned, adding Support for battery but only the thermostat.

I have seen this threat but I think I will wait for the official realize.

msolomos, 2023-02-02:
> @msolomos Did you check with Home Assistant 2023.2.0 betas?

I tried the 2023.2.0 last night. It didn't make any difference. 
As another bug came up with my tablet's webview I had to revert to previous version. 

I will keep looking into this, but until now for me it doesn't work.

ostojan, 2023-02-02:
> > @msolomos Did you check with Home Assistant 2023.2.0 betas?
> 
> I tried the 2023.2.0 last night. It didn't make any difference. As another bug came up with my tablet's webview I had to revert to previous version.
> 
> I will keep looking into this, but until now for me it doesn't work.

I haven’t had time yet to test new version and I’ll be testing it maybe this evening, but have you removed device from ZHA and added it back again?

rahulpdev, 2023-02-08:
> I have created a quirk that is working fine for me. I have uploaded a gist so more people can try it out and if its stable maybe this can be included in offical build. [...]

Could this blueprint be tweaked to work with https://zigbee.blakadder.com/houseiq_GTZ02.html that is essentially the same hardware as the Aqara TRV? I have one of each device and they're identical!

msolomos, 2023-02-13:
it is such a shame that OFF state is still not supported.  I never thought to switch to Zigbee2Mqtt and to be honest I know that  nobody cares if I do :-)

But living on a big house with 12 aqara radiators leaving me with no many options.

And even though battery was fixed, which gave us something to wait for, the most important is still not fixed.

So please, once more, we need the OFF state...
Because, at the end of the day, if you can't turn them off, what is the meaning of multi zone heating??

anyrandomguy92, 2023-02-13:
> it is such a shame that OFF state is still not supported. I never thought to switch to Zigbee2Mqtt and to be honest I know that nobody cares if I do :-)
>
> But living on a big house with 12 aqara [...]

if you check #2109 looks like the off function is being implemented. in the meantime if you wanna turn off a zone you can just use the cluster comand or, as i do, just set a very low temperature even if i'm not shure is a good thing to do

msolomos, 2023-02-14:
> if you check #2109 looks like the off function is being implemented. in the meantime if you wanna turn off a zone you can just use the cluster comand or, as i do, just set a very low temperature even if i'm not shure is a good thing to do

@anyrandomguy92 thank you for your prompt reply. I noticed #2109 and I am very excited. 
Do we know when it will be available ??

jzielke84, 2023-02-14:
I’ll also switch to MQTT anyway since the battery status is properly exposed there. I’m tired of all this quirk stuff which sometimes relies on just one person.

msolomos, 2023-02-14:
> I’ll also switch to MQTT anyway since the battery status is properly exposed there. I’m tired of all this quirk stuff which sometimes relies on just one person.

if you are going to switch to MQTT just for the battery state I would say not, because it is already supported. 
If you want for the rest features I would say to you just wait for the #2109

TheJulianJES, 2023-04-27:
The issue in the original post was fixed in HA 2023.4.0 with:
- https://github.com/zigpy/zha-device-handlers/pull/2238
- https://github.com/home-assistant/core/pull/90158

For implementing the native scheduling option of the device, see this issue/feature request:
- https://github.com/zigpy/zha-device-handlers/issues/2359

closed 2023-04-27 by TheJulianJES