zigpy/zha-device-handlers #1760: [Device Support Request] LEDVANCE Panel TW Z3

alexdelprete, 2022-09-16:
**Is your feature request related to a problem? Please describe.**
I'm unable to set the power on state on LEDVANCE Panel TW Z3. I have 3 of them in the bedrooms and when power goes off and on during the night they turn on and I must make sure they remain off, otherwise my wife and kids get mad at me and my "smart" automation stuff...:)

**Describe the solution you'd like**
Quirk to add the `0x01` (`save_defaults`) command in the `0xfc01` cluster to save the default power on state. Like in #459 (commented in #449) by @TheJulianJES. That _seems_ to be working for all LEDVANCE lights.

I'd like to do a PR starting from #459, but I need a little bit of guidance regarding the light type differences, how does the quirk match the ID of the device, etc. would love to learn how to do it, if you have the patience to guide me a little bit.

Thanks for any help on this.

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4489, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0b05",
        "0x1000",
        "0xfc0f"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LEDVANCE",
  "model": "Panel TW Z3",
  "class": "zigpy.device.Device"
}
```
</details>

TheJulianJES, 2022-09-16:
Thanks for creating the dedicated issue.

The LEDVANCE "Z3" products are (thankfully) entirely different from the old OSRAM/LEDVANCE Lightify series.
It looks like your LEDVANCE Panel still has a a custom "LEDVANCE cluster" ``0xfc0f``:
https://github.com/zigpy/zha-device-handlers/blob/a072ca3a99acc3744e74ef0b9d40d30d86adeead/zhaquirks/ledvance/__init__.py#L8-L18

However, I know that some LEDVANCE products still support setting standard Zigbee 3 attributes but don't allow reading of them at all.
Go to your device's config page, click on the little arrow under ``Zigbee info``, select ``Manage clusters``, select the ``OnOff`` cluster in the first dropdown menu. In the second dropdown menu, select ``start_up_on_off``.
When clicking ``Read Zigbee attribute``, the first field will probably populate with ``None``.
Remove that and enter a ``0`` instead, then click ``Set Zigbee attribute`` (leave the second text field empty).
Wait ~20 seconds and then unplug the light. Wait another ~10 seconds and then re-plug the light and see if it stays off.

The values are:
* ``0`` = turn off on start-up
* ``1`` = turn on on start-up
* ``2`` = toggle on start-up
* ``255`` = restore last state on start-up

If this works, great! Ignore the rest of my comment then (and let me know that it worked).
Otherwise, continue reading for trying the custom "LedvanceCluster" approach with a custom quirk (for now).

---

It's also possible that your device still supports the ``save_defaults`` command, but I don't know.

I've quickly thrown together a custom quirk (could be completely broken/not working at all because I haven't had time to properly write or test it but maybe you can fix the remaining issues if there are any).
​

To install it:

1. Take a backup of HA
2. In your ``configuration.yaml``, add this (if you don't already have it):
```yaml
zha:
  custom_quirks_path: /config/zha_custom_quirks
```
(assuming your HA config directory is at ``/config`` (it is if you're running Home Assistant OS))

3. In the ``/config`` directory, create a folder called ``zha_custom_quirks``, put the ``ledvance_panel_tw_z3.py`` file from this extracted ZIP file in it: [ledvance_panel_tw_z3.py.zip](https://github.com/zigpy/zha-device-handlers/files/9580336/ledvance_panel_tw_z3.py.zip)
4. Then restart Home Assistant

Through the ``Manage clusters`` option for the device, you should now see an ``LedvanceLightCluster `` now (assuming everything worked correctly).
Select that and in the "commands" section, you should see the ``save_defaults`` command.
I guess you can turn the light off and then execute that command. The button *should* turn green briefly.
After ~20 seconds, disconnect the device from mains powered, wait another ~10 seconds and reconnect it and see what happens.
If it still turns on, try to set the light to a very dimmed state and call the ``save_defaults`` command again. (Maybe it doesn't work with an entirely off-state?) Then re-plug the panel to test if that worked.

alexdelprete, 2022-09-16:
Julian, first of all thank you very much for the prompt reply. :)

I had spent all evening trying with the standard cluster/attributes, but it didn't work, that's why I then looked on the repo for other solutions, when I found your issue. I also made a script trying to set the `start_up_on_off` attribute to 255, but it didn't work on any panel.

```yaml
service: zha.set_zigbee_cluster_attribute
data:
  value: 0xff
  ieee: f0:d1:b8:00:00:13:d6:9b
  endpoint_id: 1
  cluster_id: 0x0006
  cluster_type: in
  attribute: 0x4003
```

So I downloaded your v2 zip, and configurez ZHA. Unfortunately on restart I have this error regarding `GreenPowerProxy` not defined. ZHA won't start, so I have no zigbee control now. 

```log
Logger: homeassistant.config_entries
Source: zha_custom_quirks/ledvance_panel_tw_z3.py:61
First occurred: 05:44:37 (1 occurrences)
Last logged: 05:44:37

Error setting up entry Tube EFR32 Pro for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 357, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 101, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/zha_custom_quirks/ledvance_panel_tw_z3.py", line 29, in <module>
    class PanelTW(CustomDevice):
  File "/config/zha_custom_quirks/ledvance_panel_tw_z3.py", line 61, in PanelTW
    OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
NameError: name 'GreenPowerProxy' is not defined
```

TheJulianJES, 2022-09-16:
If that happens, you can always remove the custom quirk (the ``.py``file) and restart HA.
But yeah, I didn't test that quirk at all (or wrote it properly). Try this:

Change the imports at the top to this (notice the added ``GreenPowerProxy``).
```python
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
```

Or just download the updated quirk file here: [ledvance_panel_tw_z3.py version 3.zip](https://github.com/zigpy/zha-device-handlers/files/9580572/ledvance_panel_tw_z3.py.version.3.zip)

alexdelprete, 2022-09-16:
> If that happens, you can always remove the custom quirk (the `.py`file) and restart HA.

I simply commented the quirks line in zha section and restarted. :)

`# custom_quirks_path: /config/zha_custom_quirks`

Ok, now I modified the quirk adding the line and it started fine. I opened up the clusters and I don't see `LedvanceLightCluster` but `OsramLightCluster`:

[image]

Should I execute the command anyway?

alexdelprete, 2022-09-16:
I tried it anyway, it didn't work, the panel turns on when power comes back on. I guess `0xfc0f` doesn't work on LEDVANCE. Should be `0xfc01`, right?

TheJulianJES, 2022-09-16:
Ah, that was the first thing I changed (in the "v2 version"). I originally had the LedvanceCluster but that has a different cluster id. The signature of your device reports ``0xfc0f`` which is the "Osram clsuter". The command is the same anyway.

Can you check if it saves some defaults at all? Like, turn the brightness to 1% (but leave the light on), execute ``save_defaults``, wait and replug?
When executing the command, do errors appear anywhere?

alexdelprete, 2022-09-16:
> The signature of your device reports `0xfc0f` which is the "Osram cluster". The command is the same anyway.

Maybe since the fw was updated in late 2021, they switched to Osram clusterId instead of ledvance?

> Can you check if it saves some defaults at all? Like, turn the brightness to 1% (but leave the light on), execute `save_defaults`, wait and replug?

Set at 2%, powered off, powered on, and it turned on full bright. :(

There's no chance to "read" what is the power on config? It's so annoying...

> When executing the command, do errors appear anywhere?

No errors...

TheJulianJES, 2022-09-16:
> Maybe since the fw was updated in late 2021, they switched to Osram clusterId instead of ledvance?

Can you click the "Add device" button in ZHA to permit joins and then reset the LEDVANCE device (so it re-joins and re-interviews and reports all possible "new clusters").

Also, you might have tried this too but does setting ``LevelControl`` / ``start_up_current_level`` to ``0`` or ``2`` do anything? (You can try through the UI if you haven't already. Just make sure to leave the second text field with the manufacturer code empty)

TheJulianJES, 2022-09-16:
Also try to execute the ``save_defaults`` command with ``0x110c`` or ``0x1189`` in the second (manufacturer code) text field.

alexdelprete, 2022-09-16:
> Can you click the "Add device" button in ZHA to permit joins and then reset the LEDVANCE device (so it re-joins and re-interviews and reports all possible "new clusters").

Reconfiguration (button on the left, under device info) doesn't do exactly that? not reliable?

> Also, you might have tried this too but does setting `LevelControl` / `start_up_current_level` to `0` or `2` do anything?

I think I tried it...it didn't work, but I'll try again.

TheJulianJES, 2022-09-16:
> Reconfiguration (button on the left, under device info) doesn't do exactly that? not reliable?

I'm pretty sure it doesn't re-interview the device (but only does binding and reporting). It would be nice if you could reset one panel to see if the signature changes when re-pairing after the firmware update.

alexdelprete, 2022-09-16:
> `4489`

Tried both manuf. codes, doesn't work. Tried also initializing one of the three panels and rejoined the zigbee network, made no difference.

I'm starting to lose hope to solve this one...

TheJulianJES, 2023-01-04:
Likely not fixable, as the device doesn't seem to have any attributes for saving power-on-state.

closed 2023-01-04 by TheJulianJES