zigpy/zha-device-handlers #1764: [Device Support Request]

Fierydwarf, 2022-09-17:
**Is your feature request related to a problem? Please describe.**
After joining the Zigbee plugs, TS011F by _TZ3000_okaz9tjs to ZHA the numbers being pulled are incorrect, yet when using adding the same device using Zigbee2MQTT the Amps show as 2.1A, which is correct and the PolingSmartEnergy is 0

**Describe the solution you'd like**
Fix the numbers.

<details>
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}

```yaml
N/A
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.4",
    "dev": false,
    "hassio": true,
[... 500 lines cut ...]
      }
    }
  }
}
[image]

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

puddly, 2022-09-17:
In what way are the numbers incorrect?

Fierydwarf, 2022-09-17:
[image]

javicalle, 2022-09-23:
Please, edit your first post and try to fix your code format.

Put the code inside the ` ``` ` marks and don't remove any breakline:
````
<details>
<summary>Device signature</summary>

```yaml
Here your code
```

</details>
````

If you can get it, just put the code between a pair of ``` marks

Fierydwarf, 2022-09-25:
I think ive formatted it correctly

javicalle, 2022-09-25:
I believe that the current (`A`) can be fixed just setting a value to the `ac_current_divisor (id: 0x0603)` from the `ElectricalMeasurement (Endpoint id: 1, Id: 0x0b04, Type: in)` cluster.
First check if there is 
First validate if the attribute has a value already assigned (pushing the 'Get Zigbee attribute' button).
If it has no value assigned, set the value `100`.
If it already has a value assigned, apply the x100 correction to the value it has. For example, if the current value were 10, you would assign a value of 1000.

The `PolledSmartEnergy` is an incremental value. 
It will continue to grow and I don't know if there is any way to reset it. Maybe the best will be remove the device and pair it again.

javicalle, 2022-10-08:
You would need enable the local quirk configuration. There are some info for example here:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once configured, copy the current `` file inside the `custom_zha_quirks` folder:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py

Add the proposed quirk at the end of the file. That would be my proposal:
```python
class Plug_v2(CustomDevice):
    """Another TS011F Tuya plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0051",
            # "in_clusters": ["0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04", "0xe001"],
            # "out_clusters": []
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBE000Cluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

Once added you need to restart HA, remove the device and pair it again.
Look at the device signature, it must change to the new quirk (check the `class` part).

dmulcahey, 2022-10-30:
@Fierydwarf did this work?

Fierydwarf, 2022-10-30:
@dmulcahey Sorry, i missed the update. Ive trid ti tonight and no it didnt work. The values are still the same.The Class within Signature displays as "class": "Zigpy.device.Device"

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

javicalle, 2022-10-30:
Upssss, my fault. 

Change the `TuyaZBE000Cluster` for the `TuyaZBExternalSwitchTypeCluster`. Already updated in my comment.

Fierydwarf, 2022-11-06:
No luck. i think i need to delete the files, and zigby_quirks and start fresh.

'''
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
'''

javicalle, 2022-11-06:
Double check if there is a `__pycache__` folder in your local quirk folder and delete it before restart HA.

Fierydwarf, 2022-11-06:
Progress!

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "ts011f_plug.Plug_v2"
}```

Fierydwarf, 2022-11-06:
Progress! I deleted the ZHA quirks option in HA, deleted the files and folders and started fresh. Still may need a bit of tweaking but its much closer than it was previously 

[image]

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "ts011f_plug.Plug_v2"
}
```

javicalle, 2022-11-06:
We will need more information about what is working and what isn't.
At least the diagnosis information from the device and the current and expected values for the rogue entities.

Fierydwarf, 2022-11-12:
Appoligise for the slow reply. life. So the values are pretty much there, enought "there" for me. Thankyou!.  Im not not able to add the entity to the Home Assistant Energy dashboard. Im assuming this would need raising a seperate tickt for?

javicalle, 2022-11-13:
Is the quirk from https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1272390644 the right version? Is there something else to fix? 
The quirk must be merged to the mainstream to be included in next versions of HA.

> Im not not able to add the entity to the Home Assistant Energy dashboard. Im assuming this would need raising a seperate tickt for?

Yes, it will be the best.

Fierydwarf, 2022-11-27:
Quirks resolved the issue

closed 2022-11-27 by Fierydwarf

jesus51998, 2024-02-08:
Hi, I have 4 device kind “_TZ3000_okaz9tjs”, "TS011F", I followed the thread, but my device, don’t work well, the energy and current are multiplier for 100 or 1000.

Somebody can help my.

I do..
1.- Delete device.
2.- delete folder  __pycache__
3.- copy file https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py in my folder /config/custom_zha_quirks/
4.- In configuration.yaml I insert:
zha:
   enable_quirks: true
   custom_quirks_path: /config/custom_zha_quirks/
5.- Reboot HA.
6.- Add device
But my device don’t change… 

‘’’{

  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",

  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0xe000",
        "0xe001"
      ],
     "output_clusters": []
    }
  },
  "manufacturer": "_TZ3000_okaz9tjs",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}’’’

I changed the "ts011f_plug.py" file, with comments from some forum threads, and then I did all the steps from 1 to 6 but my device doesn't change.

TeaMoe, 2024-02-15:
@jesus51998 

Hi, last week i got some _TZ3000_okaz9tjs Plugs from AliExpress and i also noticed that the current and energy readings are off by 100s. Additionally these plugs identify themselves as lights so that none of the existing quirks match. I solved the issue for me by creating the following quirk, which is essentially just a copy of the `Plug_v2` but with an ON_OFF_LIGHT as DeviceType. I'll do a proper pull request later.

```python
"""TS011F plug."""

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.quirk_ids import TUYA_PLUG_ONOFF
from zhaquirks.tuya import (
    EnchantedDevice,
    TuyaNewManufCluster,
    TuyaZB1888Cluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBMeteringClusterWithUnit,
    TuyaZBOnOffAttributeCluster,
)

class Plug_v2l(EnchantedDevice):
    """Another TS011F Tuya plug. First one using this definition is _TZ3000_okaz9tjs."""

    quirk_id = TUYA_PLUG_ONOFF

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0100",
            # "in_clusters": ["0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04", "0xe001"],
            # "in_clusters": ["0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x000a", "0x0702", "0x0b04", "0x1000", "0xe000", "0xe001"],
            # "out_clusters": []
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    LightLink.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                    TuyaZBMeteringClusterWithUnit,
                    TuyaZBElectricalMeasurement,
                    LightLink.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

jesus51998, 2024-02-15:
Hi, it works really good, I proved a lot of thing but nothing did work. In this moment work perfect.
thank you

cordvision, 2024-02-28:
I could use some help. I bought a "smart breaker" (not a plug) as you can see  here:
[image]

It shows up under the same name as the plug: "TS011F
by _TZ3000_zrm3oxsh"

The current (A) is off by a factor of x1000 and the Summation delivered is off by about a factor of 100x.

I was hoping that the posted custom quirk would fix the problem.

However, I can't get it to load the quirk for the device. There is a good chance that I'm doing something wrong (I'm quite new at this, but I do already have other custom quirks running successfully). 
Here is a screenshot of the Zigbee device. I don't see a path to the custom quirk  in "Zigbee Info":
[image]

First of all, do I need to edit the posted code at all and add the signature of my device? The only thing I changed is the following: # "device_type": "0x0051", to # "device_type": "0x010a"
I did this because  if I go to "Manage Zigbee Device" -->  "Signatur", I see the following

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_zrm3oxsh",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}

```

Quirk I'm using.
This is the code that resides here: /homeassistant/zha_quirks/ts011f_plug.py

```yaml
"""TS011F plug."""

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
[... 1559 lines cut ...]
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

```

Any chance somebody can tell my what I'm doing wrong?

alka79, 2024-03-03:
I too was struggling with this plug _TZ3000_okaz9tjs (brand ELIVCO).
[Your quirk](https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958) above fixes it. Thank you @TeaMoe  :-)

kitor, 2024-04-04:
> Hi, last week i got some _TZ3000_okaz9tjs Plugs from AliExpress and i also noticed that the current and energy readings are off by 100s. Additionally these plugs identify themselves as lights so [...]

This also works fine for me, thanks!

Btw: I noticed there's no OTA cluster on that device. Isn't that weird?

JeromeSi, 2024-04-21:
with https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958
I have this issue :
`File "/config/custom_zha_quirks/ts011f_plug.py", line 30, in <module>
    from zhaquirks.tuya import (
ImportError: cannot import name 'EnchantedDevice' from 'zhaquirks.tuya' (/usr/local/lib/python3.11/site-packages/zhaquirks/tuya/__init__.py)
2024-04-21 16:00:25.562 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'9cd7c77f21539874f5ded383f47b1ef7eacfd77e' (b'cfd7' != b'68e9')`
with https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1272390644
I have this issue :
`File "/config/custom_zha_quirks/ts011f_plug.py", line 30, in <module>
    from zhaquirks.tuya import (
ImportError: cannot import name 'EnchantedDevice' from 'zhaquirks.tuya' (/usr/local/lib/python3.11/site-packages/zhaquirks/tuya/__init__.py)
2024-04-21 15:50:33.896 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'9cd3c77f21539874f5de8383f47b1ef0ea3ddc7e' (b'3ddc' != b'9b3b')`
I'm a rookie in HA. Maybe I'm make something wrong. I don't know

WeterPeter, 2024-11-04:
> @jesus51998
>
> Hi, last week i got some _TZ3000_okaz9tjs Plugs from AliExpress and i also noticed that the current and energy readings are off by 100s. Additionally these plugs identify themselves as [...]

Ah, this worked for me. I was in the proces returning the plugs but with this quirk not needed! 
Thanks! 👍🏻

mhelff, 2025-01-20:
> [@jesus51998](https://github.com/jesus51998)
>
> Hi, last week i got some _TZ3000_okaz9tjs Plugs from AliExpress and i also noticed that the current and energy readings are off by 100s. Additionally [...]

@TeaMoe Did you open a PR for that? I ran in the same issue, and had to use your custom quirk - the current ZHA implementation still seems to be buggy.

Badbond, 2025-02-04:
> Btw: I noticed there's no OTA cluster on that device. Isn't that weird?

I noticed the same for the [new variant](https://github.com/zigpy/zha-device-handlers/issues/3268) of `_TZ3000_okaz9tjs`. Unfortunately, I am also facing unexpected behavior in that the plugs power off, for which several sources indicate it may be good to update the firmware. @kitor were you able to progress on OTA updates for these plugs that are missing the cluster?

PR #2069 (2023-01-10): Add support for TS011F by _TZ3000_okaz9tjs
Fixes #1764, fixes #1966 

Note, I doubt this is fully ready, but it is functional.

Potential discussions are if the new Plug class named correctly, is adding _v2 appropriate?

And should class TuyaZBMeteringCluster_v2() be in this file or say, in __init__.py