zigpy/zha-device-handlers #1765: [Device Support Request] Aqara LUMI.weather

paulstimpson, 2022-09-17:
**Is your feature request related to a problem? Please describe.**
I have purchased an Aqara temperature/humidity/pressure sensor. Home Assistant says it has paired and configured it successfully. Its entities exist and can be used. The attributes update immediately when paired but then don't continue to update. The device appears in the visualization but doesn't have any connections and after a short time it becomes unavailable and falls off the mesh. Pressing the button on the device while it's connected causes it to update in HA. I've tried adding it directly to the coordinator and via two different model Ikea bulbs as repeaters, the result is identical.

**Describe the solution you'd like**
The device should pair then its values automatically update periodically in HA, like my Sonoff temperature sensors do.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0403",
        "0x0405",
        "0xffff"
      ],
      "out_clusters": [
        "0x0000",
        "0x0004",
        "0xffff"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.weather",
  "class": "zhaquirks.xiaomi.aqara.weather.Weather2"
}

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.67",
    "supervisor": "2022.08.6",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "nodered": {
      "version": "1.0.7",
[... 319 lines cut ...]
          }
        }
      }
    }
  }
}

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```
LUMI lumi.weather Identifybutton became unavailable
5:06:14 PM - Yesterday
LUMI lumi.weather Zha Event was fired
5:06:14 PM - Yesterday

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2022-09-30:
I have one IKEA CWS1 600 Lm (first gen) and its working very good with my Aqara weather on  the balcony for 3 years.
Most of my lumi weather is connected thru IKEA 2 gen devises like the outlet and is working well.

I have some 3 gen IKEA lights and they is having not so good radio performance  as first and 2 gen devices (Silabs have putting the TX power down in the module) and some is the antenna very bad places (up side down and is shielded of the metal in the socket) and can being tricky getting good radio performance.

How hi / low is your LIQ on the link between the routers and also the link the end device is having to its parent ?

My production network is one device having 96 i LIQ and all the other is having 116 or higher and is normally very stable only my Philips HUE motion sensors is "jumping around".

monsieurhannes, 2022-10-27:
I also have problems with my aqara "weather" sensor `lumi.weather`, having the sensor just 1m from a `TRADFRI bulb E27 WW 806lm by IKEA of Sweden`  and another one roughly 1m from **6** `TRADFRIbulbE14WScandleopal470lm` - though that's through a wooden floor up to the attic.
They work just fine when I connect them. But after a couple of status updates they become unresponsive.
My network looks like this, I'd say it's not too crowded, but who knows what the limit is... I use a ConBee II and the two circles on the left are the aqara sensors.
[image]

I have one aqara TVOC sensor which has no problem staying connected.

dmulcahey, 2022-10-30:
These work fine when the parents don't drop them. This is a widely known issue w/ older aquaria devices and there is nothing that any integration can do to prevent this. This is purely caused by the mix of devices you have on your network. See this discussion for more details: https://community.smartthings.com/t/original-aqara-xiaomi-zigbee-sensors-contact-temp-motion-button-outlet-leak-etc/113253

closed 2022-10-30 by dmulcahey

monsieurhannes, 2022-10-30:
Thanks for the reply. I’ll try to hard-bind them to the coordinator so they don’t lose connection 👍