zigpy/zha-device-handlers #1768: [Device Support Request]

TommisGR, 2022-09-18:
**Is your feature request related to a problem? Please describe.**
tuya Din Rail Power Meter recognized as 
TS0601 by _TZE200_lsanae15
Only entities created are LQI and RSSI

**Describe the solution you'd like**
To be able to report Power usage

<details>

<summary>Device signature</summary>

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_lsanae15",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.8.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Athens",
    "os_name": "Linux",
    "os_version": "5.15.32-v8",
    "supervisor": "2022.08.6",
    "host_os": "Home Assistant OS 8.5",
    "docker_version": "20.10.14",
    "chassis": "embedded",
    "run_as_root": true
  },
[... 305 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.
[image]
https://www.aliexpress.com/item/1005004399475951.html

TommisGR, 2022-09-20:
Tryied adding custom quirk 'ts0601_din_power.py' with the below change:
```
MODELS_INFO: [
            ("_TZE200_lsanae15", "TS0601"),
        ],
```
but it doesnt get recognized am I doing something wrong
quirk folder works the'__pycache__' is created after I delete it and restart HA

TommisGR, 2022-09-21:
can someone please guide me with adding a custom quirk.

javicalle, 2022-09-21:
It seems to be the `TuyaPowerMeter` with the GPP cluster.

Can you try with that quirk?:
```python
class TuyaPowerMeter_GPP(TuyaSwitch):
    """Tuya power meter device."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.switch_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x000a, 0x0019]
        MODELS_INFO: [
            ("_TZE200_lsanae15", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51
            # device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

You will need to import the `GreenPowerProxy`:
```python
from zigpy.zcl.clusters.general import GreenPowerProxy
```

TommisGR, 2022-09-22:
used the "ts0601_din_power.py" from the 0.0.79 and replaced from the
class TuyaPowerMeter(TuyaSwitch):
and to the bottom  with the
class TuyaPowerMeter_GPP(TuyaSwitch): (all the page)
additionaly added on top
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time , GreenPowerProxy

[image]
some improvement but still cant view current consumsion

javicalle, 2022-09-22:
I think you need to remove the device from HA and re-pair again.

If there are no improvements, the debug logs would be necessary to know what could be happening:
* https://www.home-assistant.io/integrations/zha/#debug-logging

PS: No need to remove the `TuyaPowerMeter` class, you can keep both classes in the file. 😉

TommisGR, 2022-09-22:
NO improvements and log files sims long what part should I grep for?

javicalle, 2022-09-22:
You can try to grep the device UID `0x73d2`.
Any warning of "no handler for..." o similar will be usefull.

TommisGR, 2022-09-22:
the `grep "0x73d2" home-assistant.log` returns lots of the below nothing about "no handler"
anything else I could try?
```
2022-09-22 13:36:02.775 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of a4:c1:38:cb:35:b5:0e:3b/0x73d2 device
2022-09-22 13:36:03.093 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32817, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=13), 148, -63, 0x73d2, 255, 255, b'f\x00\x04\x00\x02\x98\xf4\xab\xe3\xcc\xcc\xcc\xccQ\xf6\x08\xfe\xff#\xa4`\x00\x00\x04\x02\x00d\x98\xf4\xab\xe3\xcc\xcc\xcc\xcc\xdf\xc9\xca$\x00K\x12\x00wL%\x02\x01*']
2022-09-22 13:36:03.102 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] request status: Status.SUCCESS. response: Neighbors(entries=4, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=60:a4:23:ff:fe:08:f6:51, nwk=0x0000, device_type=<DeviceType.Coordinator: 0>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Parent: 0>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=0, lqi=100), Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=00:12:4b:00:24:ca:c9:df, nwk=0x4C77, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Sibling: 2>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=42)])
2022-09-22 13:36:04.372 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] Querying next starting at 2
2022-09-22 13:36:04.732 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32817, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=14), 148, -63, 0x73d2, 255, 255, b'h\x00\x04\x02\x02\x98\xf4\xab\xe3\xcc\xcc\xcc\xcc\xd7\x18\xc2$\x00K\x12\x00\x1e\xd6%\x02\x01!\x98\xf4\xab\xe3\xcc\xcc\xcc\xccw\x90\xc5$\x00K\x12\x00\xad\xaf%\x02\x01\x12']
2022-09-22 13:36:04.740 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] request status: Status.SUCCESS. response: Neighbors(entries=4, start_index=2, neighbor_table_list=[Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=00:12:4b:00:24:c2:18:d7, nwk=0xD61E, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Sibling: 2>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=33), Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=00:12:4b:00:24:c5:90:77, nwk=0xAFAD, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Sibling: 2>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=18)])
2022-09-22 13:36:04.742 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] Done scanning. Total 4 neighbours
2022-09-22 13:36:53.433 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=15), 132, -67, 0x73d2, 255, 255, b'\t\xef\x02\x01\x02\x06\x00\x00\x08\x08\xe4\x00\x04j\x00\x00\xb1']
2022-09-22 13:36:53.452 DEBUG (MainThread) [zhaquirks.tuya] [0x73d2:1:0xef00] Received value [8, 228, 0, 4, 106, 0, 0, 177] for attribute 0x0006 (command 0x0002)
2022-09-22 13:36:53.636 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=16), 148, -63, 0x73d2, 255, 255, b'\t\xf0\x02\x01\x03\x01\x02\x00\x04\x00\x00\x0c\x0f']
2022-09-22 13:36:53.657 DEBUG (MainThread) [zhaquirks.tuya] [0x73d2:1:0xef00] Received value [0, 0, 12, 15] for attribute 0x0201 (command 0x0002)
2022-09-22 13:38:05.125 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=17), 136, -66, 0x73d2, 255, 255, b'\x08\xf1\n\x01\x00 F\xe2\xff 6\xe4\xff \x00']
2022-09-22 13:38:53.440 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=18), 144, -64, 0x73d2, 255, 255, b'\t\xf2\x02\x01\x04\x06\x00\x00\x08\x08\xeb\x00\x04\xe7\x00\x00\xc1']
2022-09-22 13:38:53.465 DEBUG (MainThread) [zhaquirks.tuya] [0x73d2:1:0xef00] Received value [8, 235, 0, 4, 231, 0, 0, 193] for attribute 0x0006 (command 0x0002)
2022-09-22 13:38:53.644 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, optio [... cut ...]
2022-09-22 13:40:53.608 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=21), 148, -63, 0x73d2, 255, 255, b'\t\xf5\x02\x01\x07\x01\x02\x00\x04\x00\x00\x0c\x10']
2022-09-22 13:40:53.626 DEBUG (MainThread) [zhaquirks.tuya] [0x73d2:1:0xef00] Received value [0, 0, 12, 16] for attribute 0x0201 (command 0x0002)
```

javicalle, 2022-09-22:
I was expecting some of this:
```
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Received ZCL frame: b'\t\x97\x02\x00\xdf\x06\x01\x00\x01\x00'
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=151, command_id=2, *is_reply=True)
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Received command 0x02 (TSN 151): set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))
2022-06-25 15:19:25 WARNING (MainThread) [zigpy.zcl] [0xC958:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=151, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
```

Any warning that could be relevant to us?

TommisGR, 2022-09-22:
`grep "0x73d2" home-assistant.log | grep "zigpy"`

```
2022-09-22 09:33:04.644 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
2022-09-22 09:33:04.645 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 5 value: TS0601
2022-09-22 09:33:05.728 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
2022-09-22 09:33:05.729 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 5 value: TS0601
2022-09-22 09:33:05.729 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0b04] Attribute id: 1539 value: 1000
2022-09-22 09:33:05.729 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0b04] Attribute id: 1538 value: 1
2022-09-22 09:33:05.730 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0702] Attribute id: 768 value: 0
2022-09-22 09:33:05.730 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 1 value: 70
2022-09-22 09:33:05.730 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 65506 value: 54
2022-09-22 09:33:05.730 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 65508 value: 0
2022-09-22 09:33:05.730 DEBUG (MainThread) [zigpy.appdb] [0x73d2:1:0x0000] Attribute id: 65503 value: S�*��*e��*
2022-09-22 13:36:02.775 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of a4:c1:38:cb:35:b5:0e:3b/0x73d2 device
2022-09-22 13:36:03.102 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] request status: Status.SUCCESS. response: Neighbors(entries=4, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=60:a4:23:ff:fe:08:f6:51, nwk=0x0000, device_type=<DeviceType.Coordinator: 0>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Parent: 0>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=0, lqi=100), Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=00:12:4b:00:24:ca:c9:df, nwk=0x4C77, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Sibling: 2>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=42)])
2022-09-22 13:36:04.372 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] Querying next starting at 2
2022-09-22 13:36:04.740 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] request status: Status.SUCCESS. response: Neighbors(entries=4, start_index=2, neighbor_table_list=[Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=00:12:4b:00:24:c2:18:d7, nwk=0xD61E, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Sibling: 2>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=33), Neighbor(extended_pan_id=cc:cc:cc:cc:e3:ab:f4:98, ieee=00:12:4b:00:24:c5:90:77, nwk=0xAFAD, device_type=<DeviceType.Router: 1>, rx_on_when_idle=<RxOnWhenIdle.On: 1>, relationship=<RelationShip.Sibling: 2>, reserved1=0, permit_joining=<PermitJoins.Unknown: 2>, reserved2=0, depth=1, lqi=18)])
2022-09-22 13:36:04.742 DEBUG (MainThread) [zigpy.neighbor] [0x73d2] Done scanning. Total 4 neighbours
2022-09-22 14:28:54.085 DEBUG (MainThread) [zigpy.device] [0x73d2] Invalid state on future for 0x37 seq -- probably duplicate response
```

javicalle, 2022-09-22:
Can you pair the device again and attach the logs from the pairing process?

TommisGR, 2022-09-22:
Will do but probably tomorow not at home atm

RealKanashii, 2022-09-22:
> 

I tried the quirk but I got an error: 

 File "/config/custom_zha_quirks/ts0601_din_power.py", line 3, in <module>
    class TuyaPowerMeter_GPP(TuyaSwitch):
NameError: name 'TuyaSwitch' is not defined

TommisGR, 2022-09-22:
deleted device and cache restarted then added device again
new UID 
`grep "0xb6e9" home-assistant.log`
``` 
2022-09-22 22:59:02.463 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0xb6e9, a4:c1:38:cb:35:b5:0e:3b, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2022-09-22 22:59:02.465 INFO (MainThread) [zigpy.application] New device 0xb6e9 (a4:c1:38:cb:35:b5:0e:3b) joined the network
2022-09-22 22:59:02.468 DEBUG (MainThread) [zigpy.device] [0xb6e9] Scheduling initialization
2022-09-22 22:59:02.479 INFO (MainThread) [zigpy.device] [0xb6e9] Requesting 'Node Descriptor'
2022-09-22 22:59:02.481 DEBUG (MainThread) [zigpy.device] [0xb6e9] Extending timeout for 0x29 request
2022-09-22 22:59:02.484 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for a4:c1:38:cb:35:b5:0e:3b/0xb6e9
2022-09-22 22:59:02.497 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=32), 180, -55, 0xb6e9, 255, 255, b'\x04\xe9\xb6;\x0e\xb55\xcb8\xc1\xa4\x8e']
2022-09-22 22:59:02.500 INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0xb6e9, a4:c1:38:cb:35:b5:0e:3b
2022-09-22 22:59:02.501 INFO (MainThread) [zigpy.application] Device 0xb6e9 (a4:c1:38:cb:35:b5:0e:3b) joined the network
2022-09-22 22:59:02.502 DEBUG (MainThread) [zigpy.device] [0xb6e9] Scheduling initialization
2022-09-22 22:59:02.503 DEBUG (MainThread) [zigpy.device] [0xb6e9] Canceling old initialize call
2022-09-22 22:59:02.505 DEBUG (MainThread) [zigpy.zdo] [0xb6e9:zdo] ZDO request ZDOCmd.Device_annce: [0xB6E9, a4:c1:38:cb:35:b5:0e:3b, 142]
2022-09-22 22:59:02.524 INFO (MainThread) [zigpy.device] [0xb6e9] Requesting 'Node Descriptor'
2022-09-22 22:59:02.526 DEBUG (MainThread) [zigpy.device] [0xb6e9] Extending timeout for 0x2b request
2022-09-22 22:59:02.557 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for a4:c1:38:cb:35:b5:0e:3b/0xb6e9
2022-09-22 22:59:02.748 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32770, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=33), 184, -54, 0xb6e9, 255, 255, b'+\x00\xe9\xb6\x01@\x8eA\x11BB\x00\x00*B\x00\x00']
2022-09-22 22:59:02.758 INFO (MainThread) [zigpy.device] [0xb6e9] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
2022-09-22 22:59:02.761 INFO (MainThread) [zigpy.device] [0xb6e9] Discovering endpoints
2022-09-22 22:59:02.875 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32773, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=34), 184, -54, 0xb6e9, 255, 255, b'-\x00\xe9\xb6\x02\x01\xf2']
2022-09-22 22:59:02.884 INFO (MainThread) [zigpy.device] [0xb6e9] Discovered endpoints: [1, 242]
2022-09-22 22:59:02.885 INFO (MainThread) [zigpy.device] [0xb6e9] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
2022-09-22 22:59:02.886 INFO (MainThread) [zigpy.endpoint] [0xb6e9:1] Discovering endpoint information
2022-09-22 22:59:02.989 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32772, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=35), 180, -55, 0xb6e9, 255, 255, b'/\x00\xe9\xb6\x14\x01\x04\x01Q\x00\x01\x04\x04\x00\x05\x00\x00\xef\x00\x00\x02\x19\x00\n\x00']
2022-09-22 22:59:03.004 INFO (MainThread) [zigpy.endpoint] [0xb6e9:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
2022-09-22 22:59:03.012 INFO (MainThread) [zigpy.endpoint] [0xb6e9:242] Discovering endpoint information
2022-09-22 22:59:03.138 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32772, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTIO [... cut ...]
2022-09-22 23:10:52.479 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 45, 0, 6, 168, 0, 1, 54] for attribute 0x0006 (command 0x0002)
2022-09-22 23:10:52.619 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 13, 82] for attribute 0x0201 (command 0x0002)
2022-09-22 23:12:52.500 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 36, 0, 6, 150, 0, 1, 49] for attribute 0x0006 (command 0x0002)
2022-09-22 23:12:52.688 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 13, 83] for attribute 0x0201 (command 0x0002)

```

javicalle, 2022-09-22:
> > 
> 
> I tried the quirk but I got an error:
> 
> ```
>  File "/config/custom_zha_quirks/ts0601_din_power.py", line 3, in <module>
>     class TuyaPowerMeter_GPP(TuyaSwitch):
> NameError: name 'TuyaSwitch' is not defined
> ```

You need to add the proposed quirk at the bottom of the existing `ts0601_din_power.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py

RealKanashii, 2022-09-22:
Thanks, same result that @TommisGR , switch appears, no power metrics.... unknow. Thanks for the effort. My logs:

```
2022-09-22 23:38:58.668 INFO (MainThread) [zigpy.application] New device 0x8ffa (a4:c1:38:f0:77:0f:4a:6d) joined the network
2022-09-22 23:38:58.669 DEBUG (MainThread) [zigpy.device] [0x8ffa] Scheduling initialization
2022-09-22 23:38:58.671 DEBUG (MainThread) [zigpy.zdo] [0x8ffa:zdo] ZDO request ZDOCmd.Device_annce: [0x8FFA, a4:c1:38:f0:77:0f:4a:6d, 142]
2022-09-22 23:38:58.673 INFO (MainThread) [zigpy.device] [0x8ffa] Requesting 'Node Descriptor'
2022-09-22 23:38:58.673 DEBUG (MainThread) [zigpy.device] [0x8ffa] Extending timeout for 0x42 request
2022-09-22 23:38:58.774 INFO (MainThread) [zigpy.device] [0x8ffa] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
2022-09-22 23:38:58.775 INFO (MainThread) [zigpy.device] [0x8ffa] Discovering endpoints
2022-09-22 23:38:58.905 INFO (MainThread) [zigpy.device] [0x8ffa] Discovered endpoints: [1, 242]
2022-09-22 23:38:58.906 INFO (MainThread) [zigpy.device] [0x8ffa] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
2022-09-22 23:38:58.908 INFO (MainThread) [zigpy.endpoint] [0x8ffa:1] Discovering endpoint information
2022-09-22 23:38:59.012 INFO (MainThread) [zigpy.endpoint] [0x8ffa:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
2022-09-22 23:38:59.015 INFO (MainThread) [zigpy.endpoint] [0x8ffa:242] Discovering endpoint information
2022-09-22 23:38:59.129 INFO (MainThread) [zigpy.endpoint] [0x8ffa:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
2022-09-22 23:38:59.223 INFO (MainThread) [zigpy.device] [0x8ffa] Read model 'TS0601' and manufacturer '_TZE200_lsanae15' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2022-09-22 23:38:59.224 INFO (MainThread) [zigpy.device] [0x8ffa] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_lsanae15' nwk=0x8FFA ieee=a4:c1:38:f0:77:0f:4a:6d is_initialized=True>
2022-09-22 23:39:09.053 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [8, 248, 0, 12, 139, 0, 2, 143] for attribute 0x0006 (command 0x0002)
2022-09-22 23:39:09.200 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [0, 0, 0, 170] for attribute 0x0201 (command 0x0002)
2022-09-22 23:39:09.369 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [22, 9, 22, 9, 0, 0, 0, 170] for attribute 0x0003 (command 0x0002)
2022-09-22 23:39:09.509 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [9, 23, 9, 23, 0, 0, 0, 170] for attribute 0x0004 (command 0x0002)
2022-09-22 23:39:09.646 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [1, 0] for attribute 0x000b (command 0x0002)
2022-09-22 23:39:09.798 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [1, 1, 0, 80] for attribute 0x0011 (command 0x0002)
2022-09-22 23:39:09.967 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [1, 18, 3, 0, 8, 50, 50, 48, 56, 49, 55, 53, 49] for attribute 0x0110 (command 0x0002)
2022-09-22 23:40:41.070 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [8, 248, 0, 12, 65, 0, 2, 129] for attribute 0x0006 (command 0x0002)
2022-09-22 23:40:41.212 DEBUG (MainThread) [zhaquirks.tuya] [0x8ffa:1:0xef00] Received value [0, 0, 0, 171] for attribute 0x0201 (command 0x0002)
```

RealKanashii, 2022-09-23:
Tell me if you need more logs or test something.

javicalle, 2022-09-23:
The `TuyaPowerMeter_GPP` quirk definitely doesn't work for your device.

The quirk handles DPs: 
```
TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101
```

And your device seems to make use of DPs (the attribute IDs):
```
Received value [22, 9, 22, 9, 0, 0, 13, 75] for attribute 0x0003
Received value [9, 22, 9, 22, 0, 0, 2, 194] for attribute 0x0004
Received value [9, 44, 0, 7, 33, 0, 1, 80] for attribute 0x0006
Received value [1, 0] for attribute 0x000b
Received value [1, 1, 0, 80] for attribute 0x0011
Received value [1, 18, 3, 0, 8, 50, 50, 48, 55, 49, 48, 50, 52] for attribute 0x0110
Received value [0, 0, 13, 75] for attribute 0x0201
```

A new quirk is necessary to be able to make it work. It will be necessary to match the IDs (and data types) with the magnitudes.

TommisGR, 2022-09-23:
is there a guide or some sort info on how to go about it ourselfs?

javicalle, 2022-09-23:
If you want to try there is my proposal:
```python
"""Tuya Din Power Meter."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOff,
)

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    # def _update_attribute(self, attrid, value):
    #     super()._update_attribute(attrid, value)
    #     if attrid == TUYA_TOTAL_ENERGY_ATTR:
    #         self.endpoint.smartenergy_metering.energy_reported(value / 100)
    #     elif attrid == TUYA_CURRENT_ATTR:
    #         self.endpoint.electrical_measurement.current_reported(value)
    #     elif attrid == TUYA_POWER_ATTR:
    #         self.endpoint.electrical_measurement.power_reported(value / 10)
    #     elif attrid == TUYA_VOLTAGE_ATTR:
    #         self.endpoint.electrical_measurement.voltage_reported(value / 10)
    #     elif attrid == TUYA_DIN_SWITCH_ATTR:
    #         self.endpoint.device.switch_bus.listener_event(
    #             SWITCH_EVENT, self.endpoint.endpoint_id, value
    #         )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
[... 128 lines cut ...]
    }
```

Delete the previous quirk file and put all the code inside a new file (with a new name).
That quirk is not functional but I expect that could get all the info to create a functional one.

With that quirk it's expected to get a full of logs like:
```
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Received ZCL frame: b'\t\x97\x02\x00\xdf\x06\x01\x00\x01\x00'
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=151, command_id=2, *is_reply=True)
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Decoded ZCL frame: TuyaWindowCoverManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Received command 0x02 (TSN 151): set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] No datapoint handler for TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))
2022-06-25 15:19:25 WARNING (MainThread) [zigpy.zcl] [0xC958:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=223, dp=6, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=151, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-06-25 15:19:25 DEBUG (MainThread) [zigpy.zcl] [0xC958:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)
```

This kind of log (the `No 'handle_set_data_response' tuya handler found` lines) will tell us:
* `dp=6` the MCU datapoint that report a value
* `dp_type=<TuyaDPType.BOOL: 1>` the datapoint type (necessary for the new quirk)
* `payload=<Bool.false: 0>` reported value

With that info someone with the device must match the DP with the magnitude (Is the `dp=4` the Power value or the Voltage? And the `dp=11`?). With all that info it will be possible to create the new quirk.
Probably there will be a x10 or x100 factor in some magnitudes.

RealKanashii, 2022-09-23:
Testing...

RealKanashii, 2022-09-23:
Here the log...
```
2022-09-23 22:26:42.794 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=113, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'/\x04\x00\x00', *payload=1071)))
2022-09-23 22:26:42.859 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=114, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'/\x04\x00\x00', *payload=1071)))
2022-09-23 22:28:42.818 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=116, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'3\x04\x00\x00', *payload=1075)))
2022-09-23 22:28:42.885 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=117, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'3\x04\x00\x00', *payload=1075)))
2022-09-23 22:30:42.783 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=119, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'6\x04\x00\x00', *payload=1078)))
2022-09-23 22:30:42.840 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=120, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'6\x04\x00\x00', *payload=1078)))
2022-09-23 22:32:42.809 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=122, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'9\x04\x00\x00', *payload=1081)))
2022-09-23 22:32:43.087 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=123, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b':\x04\x00\x00', *payload=1082)))
2022-09-23 22:34:42.771 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=125, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'=\x04\x00\x00', *payload=1085)))
2022-09-23 22:34:42.837 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=126, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'=\x04\x00\x00', *payload=1085)))
2022-09-23 22:36:42.791 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=128, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'@\x04\x00\x00', *payload=1088)))
2022-09-23 22:36:42.848 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=129, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'@\x04\x00\x00', *payload=1088)))
2022-09-23 22:38:42.762 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=131, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'D\x04\x00\x00', *payload=1092)))
2022-09-23 22:38:42.820 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=132, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'D\x04\x00\x00', *payload=1092)))
2022-09-23 22:40:42.781 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=134, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'G\x04\x00\x00', *payload=1095)))
2022-09-23 22:40:42.850 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=135, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'G\x04\x00\x00', *payload=1095)))
2022-09-23 22:42:42.800 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=137, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'J\x04\x00\x00', *payload=1098)))
2022-09-23 22:42:42.870 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=138, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'J\x04\x00\x00', *payload=1098)))
2022-09-23 22:44:42.768 WARNING (MainThread) [zigpy.zcl] [0x8FFA:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=5, tsn=140, dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'N\x04\x00\x00', *payload=1102)))
[... cut ...]
2022-09-23 22:48:15.588 WARNING (MainThread) [zigpy.zcl] [0x7B96:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=15, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-23 22:48:15.669 WARNING (MainThread) [zigpy.zcl] [0x7B96:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=16, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
```

javicalle, 2022-09-23:
Device is reporting only DPs 1 and 16.
DP16 is some incremental value. Consumed power, uptime, ...
DP1 is a boolean. Usually it is a sensor or a switch. Is the current switch in HA working? if its not working try to replace in `dp_to_attribute` and the `data_point_handlers` the DP 11 for the 16 and try again. Or even better, switch off and on the device several times and look at the logs, you must be able to match the logs with the events.

Also you can try to unplug and plug again the device. Usually when the device is turned on it reports the status of its attributes.

RealKanashii, 2022-09-23:
DP16 -> kWh  ( acomulative kWh ) <-  For Energy panel I suppose.
DP1 -> switch ( on/off)

RealKanashii, 2022-09-23:
Some guy integrated it to HA via zigbee2mqtt I guess:
[image]

RealKanashii, 2022-09-23:
I changed the 11 for 1 ( as the incremental value is DP1 no DP16) now I only see the set_data_response from DP16:

```
Kanashii@casa:/usr/share/hassio/homeassistant $ grep "'handle_set_data_response" home-assistant.log
2022-09-24 01:30:08.620 WARNING (MainThread) [zigpy.zcl] [0x25C0:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=14, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:30:08.762 WARNING (MainThread) [zigpy.zcl] [0x25C0:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=15, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:30:08.846 WARNING (MainThread) [zigpy.zcl] [0x25C0:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=16, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:30:57.365 WARNING (MainThread) [zigpy.zcl] [0x6EE2:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=14, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:30:57.439 WARNING (MainThread) [zigpy.zcl] [0x6EE2:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=15, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:30:57.517 WARNING (MainThread) [zigpy.zcl] [0x6EE2:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=16, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:32:11.653 WARNING (MainThread) [zigpy.zcl] [0xF124:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=14, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:32:11.739 WARNING (MainThread) [zigpy.zcl] [0xF124:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=15, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
2022-09-24 01:32:11.833 WARNING (MainThread) [zigpy.zcl] [0xF124:1:0xef00] No 'handle_set_data_response' tuya handler found for set_data_response(data=TuyaCommand(status=0, tsn=16, dp=16, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>)))
```

RealKanashii, 2022-09-23:
Using this code I got no handle_set_data_response messages....
```
 dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # 1: DPToAttributeMapping(
        #     ContactSwitchCluster.ep_attribute,
        #     "zone_status",
        #     dp_type=TuyaDPType.BOOL,
        #     converter=lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
        # ),
        # 2: DPToAttributeMapping(
        #     TuyaPowerConfigurationCluster.ep_attribute,
        #     "battery_percentage_remaining",  # 0x0021
        #     dp_type=TuyaDPType.VALUE,
        # ),
        16: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.VALUE,
        )
    }

 data_point_handlers = {
        # 1: "_dp_2_attr_update",
        # 2: "_dp_2_attr_update",
        1: "_dp_2_attr_update",
        16: "_dp_2_attr_update",
    }
```

javicalle, 2022-09-24:
> I think you saw it inverted.... the incremental is DP1 and the boolean DP16. Aren't they?

Yes, they are 🤦🏻‍♂️

It can't be 2 DP mapped to the same cluster (of the same endpoint).

You have 3 devices reporting values isn't it?
* 0x25C0
* 0x6EE2
* 0xF124

Put a full trace from one device reporting. There must be something else.

TommisGR, 2022-09-24:
After trying the above with zha failing to load I put but the 1st changed `ts0601_din_power.py` deleted cache and restarted now i see this
[image]

and greping the device id
`grep "0xb6e9" home-assistant.log`
```
2022-09-24 08:16:35.492 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
2022-09-24 08:16:35.493 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 5 value: TS0601
2022-09-24 08:16:35.508 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 780
2022-09-24 08:16:37.392 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
2022-09-24 08:16:37.392 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 5 value: TS0601
2022-09-24 08:16:37.394 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 768 value: 0
2022-09-24 08:16:37.394 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1539 value: 1000
2022-09-24 08:16:37.395 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1538 value: 1
2022-09-24 08:16:37.395 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 1 value: 70
2022-09-24 08:16:37.396 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65506 value: 54
2022-09-24 08:16:37.397 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65508 value: 0
2022-09-24 08:16:37.397 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65503 value: ���*i���*i���*i���*i
2022-09-24 08:16:37.407 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 513 value: 4422
2022-09-24 08:16:37.410 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 0 value: 44.22
2022-09-24 08:16:37.411 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 780
2022-09-24 08:16:37.412 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1291 value: 78.0
2022-09-24 08:18:49.163 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [8, 247, 0, 51, 145, 0, 11, 204] for attribute 0x0006 (command 0x0002)
2022-09-24 08:18:49.347 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 17, 116] for attribute 0x0201 (command 0x0002)
```

TommisGR, 2022-09-24:
If I could get the current power and a way to reset to total I would call it a win :)

TommisGR, 2022-09-24:
Additionaly from the instruction it appear to be reporting the following
[image]

RealKanashii, 2022-09-24:
> After trying the above with zha failing to load I put but the 1st changed `ts0601_din_power.py` deleted cache and restarted now i see this [image]
> 
> 1st & 2nd value is constant 3rd is increasing seem to be the total energy consumption since install and greping the device id `grep "0xb6e9" home-assistant.log`
> 
> ```
> 2022-09-24 08:16:35.492 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
> 2022-09-24 08:16:35.493 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 5 value: TS0601
> 2022-09-24 08:16:35.508 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 780
> 2022-09-24 08:16:37.392 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
> 2022-09-24 08:16:37.392 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 5 value: TS0601
> 2022-09-24 08:16:37.394 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 768 value: 0
> 2022-09-24 08:16:37.394 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1539 value: 1000
> 2022-09-24 08:16:37.395 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1538 value: 1
> 2022-09-24 08:16:37.395 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 1 value: 70
> 2022-09-24 08:16:37.396 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65506 value: 54
> 2022-09-24 08:16:37.397 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65508 value: 0
> 2022-09-24 08:16:37.397 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65503 value: ���*i���*i���*i���*i
> 2022-09-24 08:16:37.407 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 513 value: 4422
> 2022-09-24 08:16:37.410 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 0 value: 44.22
> 2022-09-24 08:16:37.411 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 780
> 2022-09-24 08:16:37.412 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1291 value: 78.0
> 2022-09-24 08:18:49.163 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [8, 247, 0, 51, 145, 0, 11, 204] for attribute 0x0006 (command 0x0002)
> 2022-09-24 08:18:49.347 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 17, 116] for attribute 0x0201 (command 0x0002)
> ```

Can you post the full quirk code please?  And maybe we can ask for a PR to update de actual one or create a new one to share with everyone, @javicalle , what do you think?

TommisGR, 2022-09-24:
my running quirk
```
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0201 #0x0211
TUYA_CURRENT_ATTR = 0x0003      #0x0212
TUYA_POWER_ATTR = 0x0004        #0x0213
TUYA_VOLTAGE_ATTR = 0x0011      #0x0214
TUYA_DIN_SWITCH_ATTR = 0x000b   #0x0101

SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value / 10)
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value / 10)
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
            )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

    def voltage_reported(self, value):
        """Voltage reported."""
[... 130 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=61
                # input_clusters=[]
                # output_clusters=[21]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

TommisGR, 2022-09-24:
the changed part is the blind changes made to energy attributes

RealKanashii, 2022-09-24:
mmm I can see what you see. My data still unknown ( W, A, etc). Only switch is working.

RealKanashii, 2022-09-24:
wait, have I to add all the related  tuya quirk inside the quirk path?

TommisGR, 2022-09-24:
I am on  2022.9.6
on quirks folder there is only the 1 name _of_quirk.py
before restarting HASS  delete the folder named `__pycache__` from inside the quirk folder
all the it imports are already in the HASS installation

TommisGR, 2022-09-24:
My understanding of python is basic so basicaly breaking things till I find something someone with better understanding can use.
my reconf is the same 
[image]
the total need to pass a few minutes to show up and ur meter to be connected to a load

RealKanashii, 2022-09-24:
Nope
[image]

I cleared cache, deleted device, restarted and rejoined but still no appearing attributes. it's weird.

javicalle, 2022-09-24:
I believe that this entity is the `0x0510: power_factor` attribute from the `ElectricalMeasurement` cluster:
[image]

I have several devices that show it but do not report it and it also shows as unavailable.

@TommisGR, youe device seems to be reporting someting else in DP 6 and 201. You can read the values with someting like:

```python

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
        0x0006: ("dp_6", t.uint16_t , True),
        0x0201: ("dp_513", t.uint8_t, True),
    }
```

This way I hope that the logs will show the value of the magnitude in decimal, and in this way I can infer to which magnitude it corresponds and thus be able to assign it to the corresponding attribute.
Also you can read the values from the 'manage cluster' view in your device. Select the `TuyaManufClusterDinPower`, select the attribute and push the "get zigbee attribute" button.

TommisGR, 2022-09-24:
Will try it but as on now current valye of energy is reported in current atttr bu no entity is available

javicalle, 2022-09-24:
Just to be sure, what unit of measurement do you expect to have the 'current consumption'?
You already have the `W` and `kWh`.

And there must be a correction factor to that 726 value? Maybe 72,6? Or 7.260?

javicalle, 2022-09-24:
Just for better understanding of what is doing the quirk, ket me explain what is internally doing (in a schematic way).

That device (as many of Tuya ones) is reporting values not in the standard Zigbee clusters, but in a manufacturer specific Tuya cluster (the `0xef00`).
All these reports comes to the `TuyaManufClusterDinPower` and (if they exist) are read as attributes of the cluster (the `attributes` list in the class).

What the quirk tries to do is translate these attribute values to 'normal' Zigbee clusters and attributes. That part occurs in the `_update_attribute` method.
When an attribute is updated (because device reports it) that method make a call to update something else. For example:
```
        elif attrid == TUYA_CURRENT_ATTR: 
            self.endpoint.electrical_measurement.current_reported(value)
```
When the cluster updates the `TUYA_CURRENT_ATTR` it will call to the `current_reported` method from the `electrical_measurement` cluster. And where is defined that? Here:
```
class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    .../...
    CURRENT_ID = 0x0508

    .../...
    def current_reported(self, value):
        """Ampers reported."""
        self._update_attribute(self.CURRENT_ID, value)
```

So now, the attribute `CURRENT_ID` from the `ElectricalMeasurement` cluster has taken the reported value.

And why is all this necessary?
Well, because HA only reads the standard attributes of the clusters it 'knows'.

So for HA the device stream will always read from the `` cluster `` attribute and not from anywhere else.

And so the quirk needs to do the translation between what the device reports and what HA expects.

RealKanashii, 2022-09-24:
I'm starting to get it. I don't program since my university days... long time ago.  First I will download VScode again, put the code and try to understand all you told us. 

Basically the quirks are a parser or translator catching info from a not standard cluster ( and it's attributes) to a standard zigbee one. So we need to catch the original cluster and attributes, define them ( type and format) and the prepare a function or class to move this attributes  inside to the zigbee cluster.

javicalle, 2022-09-24:
The tricky part is knowing what values the device reports correspond to.
Some may be more or less easy to know, but others it is necessary to see the values that the device is reporting and interpret what magnitudes they correspond to. For this, the manufacturer's information can give clues to what values the device reports.

The logs allow us to see all the reports of the device and its analysis (depending on its real time use) is what will allow us to deduce what each thing is.

TommisGR, 2022-09-25:
Some success at least in the entities I was interested.
quirk I am using:

```quirk
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0201 #total energy /100  0x0211
TUYA_CURRENT_ATTR = 0x0011      #0x0212
TUYA_POWER_ATTR = 0x0006        #0x0213
TUYA_VOLTAGE_ATTR = 0x0004      #0x0214
TUYA_DIN_SWITCH_ATTR = 0x000b   #0x0101

SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),      #t.int16s
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),         #t.uint16_t
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value)   #  / 10
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value  / 10 )
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
            )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050b             #0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603

    _CONSTANT_ATTRIBUTES = {AC_CURRENT_MULTIPLIER: 1, AC_CURRENT_DIVISOR: 1000}

[... 168 lines cut ...]
2022-09-25 09:43:30.669 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1538 value: 1
2022-09-25 09:43:30.669 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 1 value: 70
2022-09-25 09:43:30.669 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65506 value: 54
2022-09-25 09:43:30.670 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65508 value: 0
2022-09-25 09:43:30.670 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65503 value: 9��*i;��*i<��*i<��*i
2022-09-25 09:43:30.673 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 513 value: 5569
2022-09-25 09:43:30.673 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 0 value: 55.69
2022-09-25 09:43:30.673 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 1178
2022-09-25 09:43:30.673 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1291 value: 265
2022-09-25 09:43:30.674 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 6 value: 265
2022-09-25 09:43:30.674 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1288 value: 526
2022-09-25 09:43:30.674 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1285 value: 117
2022-09-25 09:43:30.674 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0004] Attribute id: 65533 value: 2
2022-09-25 09:43:30.675 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0004] Attribute id: 0 value: 0
2022-09-25 09:44:46.634 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 44, 0, 6, 35, 0, 1, 10] for attribute 0x0006 (command 0x0002)
2022-09-25 09:44:46.780 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 21, 195] for attribute 0x0201 (command 0x0002)
2022-09-25 09:46:46.611 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 47, 0, 6, 41, 0, 1, 14] for attribute 0x0006 (command 0x0002)
2022-09-25 09:46:46.796 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 21, 196] for attribute 0x0201 (command 0x0002)
2022-09-25 09:48:46.618 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 33, 0, 6, 74, 0, 1, 20] for attribute 0x0006 (command 0x0002)
2022-09-25 09:48:46.769 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 21, 197] for attribute 0x0201 (command 0x0002)
2022-09-25 09:50:46.639 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 32, 0, 6, 48, 0, 1, 12] for attribute 0x0006 (command 0x0002)
2022-09-25 09:50:46.783 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 21, 198] for attribute 0x0201 (command 0x0002)
2022-09-25 09:52:46.604 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [9, 32, 0, 5, 255, 0, 1, 4] for attribute 0x0006 (command 0x0002)
2022-09-25 09:52:46.747 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 21, 199] for attribute 0x0201 (command 0x0002)
```

javicalle, 2022-09-25:
There are 2 approach for this:

1. From HA try to update the `ac_voltage_multiplier` from the `TuyaPowerMeasurement` cluster with a value of `2` (probably the actual value is None). Not sure if that will do the job or can have a side efect in other values.

2. Update the quirk with the x2 factor:
```python
    def voltage_reported(self, value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value*2)
```

Protip: you can prettify the code with:
````
```python
Your code here
```
````

TommisGR, 2022-09-25:
I tried both at the class dinpower and elctricmeasurements but no change.
As is now it works for me the current - voltage and power factor are not reported.

Additionally this also has a freeze date and time that can be set in tuya app so it reset the sum of kwh per month.
couldn't figure out where those are or what format should be used.

Thanks u for ur guidance in making this work.

final quirk used:
```python
 """Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0201 #total energy /100  0x0211
TUYA_CURRENT_ATTR = 0x0011      #0x0212
TUYA_POWER_ATTR = 0x0006        #0x0213
TUYA_VOLTAGE_ATTR = 0x0004      #0x0214
TUYA_DIN_SWITCH_ATTR = 0x000b   #0x0101

SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),      #t.int16s
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),         #t.uint16_t
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value)   #  / 10
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value*2 / 10)
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
            )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050b             #0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508
[... 138 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=61
                # input_clusters=[]
                # output_clusters=[21]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

TommisGR, 2022-09-25:
On reboot I get the following error
[image]

```log
grep "0xb6e9" home-assistant.log 
2022-09-25 19:31:58.962 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
2022-09-25 19:31:58.963 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 5 value: TS0601
2022-09-25 19:31:58.964 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 1178
2022-09-25 19:32:00.323 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 4 value: _TZE200_lsanae15
2022-09-25 19:32:00.325 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 5 value: TS0601
2022-09-25 19:32:00.326 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 768 value: 0
2022-09-25 19:32:00.326 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1539 value: 1000
2022-09-25 19:32:00.327 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1538 value: 1
2022-09-25 19:32:00.327 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 1 value: 70
2022-09-25 19:32:00.327 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65506 value: 54
2022-09-25 19:32:00.328 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65508 value: 0
2022-09-25 19:32:00.328 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0000] Attribute id: 65503 value: Hc�*iKc�*iLc�*iLc�*i�c�*i�c�*i�c�*i�c�*i8d�*i;d�*i<d�*i<d�*i
2022-09-25 19:32:00.330 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 513 value: 6566
2022-09-25 19:32:00.331 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0702] Attribute id: 0 value: 65.66
2022-09-25 19:32:00.331 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 4 value: 1178
2022-09-25 19:32:00.332 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1291 value: 370
2022-09-25 19:32:00.332 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0xef00] Attribute id: 6 value: 370
2022-09-25 19:32:00.332 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1288 value: 526
2022-09-25 19:32:00.333 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0b04] Attribute id: 1285 value: 117
2022-09-25 19:32:00.333 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0004] Attribute id: 65533 value: 2
2022-09-25 19:32:00.333 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0004] Attribute id: 0 value: 0
2022-09-25 19:32:00.334 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0005] Attribute id: 0 value: 0
2022-09-25 19:32:00.334 DEBUG (MainThread) [zigpy.appdb] [0xb6e9:1:0x0005] Attribute id: 3 value: 0
2022-09-25 19:32:45.740 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [8, 206, 0, 9, 12, 0, 1, 170] for attribute 0x0006 (command 0x0002)
2022-09-25 19:32:45.932 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 25, 168] for attribute 0x0201 (command 0x0002)
2022-09-25 19:34:45.637 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [8, 197, 0, 8, 236, 0, 1, 162] for attribute 0x0006 (command 0x0002)
2022-09-25 19:34:45.781 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 25, 170] for attribute 0x0201 (command 0x0002)
2022-09-25 19:36:45.660 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [8, 194, 0, 8, 75, 0, 1, 133] for attribute 0x0006 (command 0x0002)
2022-09-25 19:36:45.800 DEBUG (MainThread) [zhaquirks.tuya] [0xb6e9:1:0xef00] Received value [0, 0, 25, 171] for attribute 0x0201 (command 0x0002)
```

RealKanashii, 2022-09-25:
> ```python
>  """Tuya Din Power Meter."""
> from zigpy.profiles import zha
> import zigpy.types as t
> from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
> from [...]

your last quirk doesn't work. I'm getting the same error.

RealKanashii, 2022-09-25:
The quirk before works once I restart the full raspberry. Something was broken without restarting.

TommisGR, 2022-10-10:
https://github.com/Koenkk/zigbee2mqtt/issues/13097
looks like same device has some info on additional DP

What follows ids from that post

DP 17 Alarm set ( if Alarm 2 not 'null')
Alarm 1 - OverVoltage, 261V, Control On
Alarm 2 - OverCurrent, 40A, Control On
DP 11 Frozen set - day if month 22, hour of day - 15
```
2022-07-23T11:30:25 0xa4c1383a1774b16b (seq:  256, value: #0) [DP   6 Phase A] => raw: [   8 194   0   0   0   0   0   0 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq:  512, value: #0) [DP   1 Forward energy total  ] => int: 59
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq:  768, value: #0) [DP   3 month energy] => raw: [  22   7  22   7   0   0   0   0 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1024, value: #0) [DP   4 daily energy] => raw: [   7  23   7  23   0   0   0   0 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1280, value: #0) [DP  11 Frozen set] => raw: [  22  15 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1536, value: #0) [DP  17 Alarm set] => raw: [   3   1   1   5   1   1   0  40 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1792, value: #0) [DP  16 switch   ] => bool: True
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1792, value: #1) [DP  18 Meter ID ] => string: 22062315
2022-07-26T21:12:35 0xa4c1383a1774b16b (seq: 6400, value: #0) [DP  10 Fault ] => bitmap: 00000000 00000000 000000
```

RealKanashii, 2022-10-10:
Maybe we can update the quirk to get all the values...

TommisGR, 2022-10-10:
i wouldnt know where to start maybe if someone held our hands for a bit :) like  javicalle did.

RealKanashii, 2022-10-10:
I made a mod to AC_VOLTAGE_MULTIPLIER to get the right value...  I definied the constant and then assigned a value of 2.

Now I see the correct Voltage in HA. I know is not much but I'm making slowly progress understanding how all lthis works.

@TommisGR  if you want to update your quirk, multiply the "value" twice didn't work for me. I have to touch the constant voltage modifier:

```
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0201 #total energy /100  0x0211
TUYA_CURRENT_ATTR = 0x0011      #0x0212
TUYA_POWER_ATTR = 0x0006        #0x0213
TUYA_VOLTAGE_ATTR = 0x0004      #0x0214
TUYA_DIN_SWITCH_ATTR = 0x000b   #0x0101

SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),      #t.int16s
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),         #t.uint16_t
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value)   #  / 10
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value / 10 )
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
            )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050b             #0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603
[... 136 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=61
                # input_clusters=[]
                # output_clusters=[21]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

doreks, 2022-10-19:
Hello, 
thank you all for the support and support the integration of this device into zha.
in my case I have copied the last of the codes and it seems that the Switch button : on/off does not work

here log received after press on/off switch:

`2022-10-19 14:28:48.673 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140260156221392] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 90, in async_turn_on
    result = await self._on_off_channel.turn_on()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 342, in turn_on
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
2022-10-19 14:28:50.607 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140260156221392] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 97, in async_turn_off
    result = await self._on_off_channel.turn_off()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 350, in turn_off
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
`

bsj17, 2022-12-28:
Did someone made progress on this quirk? Got the device today. Tried above,  active powers seems only thing to be reading right. in my case.
Switch control still does not work.

Thanks!

RealKanashii, 2022-12-31:
> 

The switch works for me.

doreks, 2023-01-01:
How did you fix it?

RealKanashii, 2023-01-02:
I just posted the quirk in the correct place and rejoined. Sometimes if you put the quirk after join some functions are invisible.  Unlink, put the quirk, restart the HA and the force the join again. 
You can see the switch control there:
[image]

[101 more replies]

closed 2024-10-12 by github-actions[bot]