zigpy/zha-device-handlers #1769: [Device Support Request] BTicino F20T60A DIN Power Meter

webartoli, 2022-09-18:
**Is your feature request related to a problem? Please describe.**
Connecting BTicino F20T60A DIN Power Meter, some sensors are detected but values remains on 0.

**Describe the solution you'd like**
Power Sensor detect power consumption.

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0b04",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0005",
        "0x0006",
        "0x0019",
        "0xfc01"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " DIN power consumption module",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "5.15.61-v7",
[... 724 lines cut ...]
[48374:1:0x0003]: 'async_initialize' stage succeeded
[48374:1:0x0b04]: 'async_initialize' stage succeeded
[48374:1:0x0000]: 'async_initialize' stage succeeded
[48374:1:0x0006]: 'async_initialize' stage succeeded
[48374:1:0x0006]: 'async_initialize' stage succeeded
[48374:1:0x0019]: 'async_initialize' stage succeeded
[48374:1:0x0005]: 'async_initialize' stage succeeded
[48374]( DIN power consumption module): power source: Mains
[48374]( DIN power consumption module): completed initialization
[0x67CE](LLC011): last_seen is 680080.3258867264 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0x67CE](LLC011): Update device availability -  device available: False - new availability: False - changed: False
[0xEB0F](LWB006): last_seen is 674760.206833601 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0xEB0F](LWB006): Update device availability -  device available: False - new availability: False - changed: False
[48374]( DIN power consumption module): Device seen - marking the device available and resetting counter
[48374]( DIN power consumption module): Update device availability -  device available: True - new availability: True - changed: False
[0xA18D](LLC011): last_seen is None, marking the device unavailable
[0xA18D](LLC011): Update device availability -  device available: False - new availability: False - changed: False

```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

Valcob, 2022-10-03:
weird it kinda works for me though still some info is missing
[image]
I would suggest you to check the sensor connection and the right installation, meaning, follow the arrow on the sensor and make sure it's on the hot wire usually the brown one or black one

closed 2023-04-08 by github-actions[bot]

korvarn, 2024-07-11:
@webartoli my apologies for commenting this old issue, wondering if you were able to get a W reading eventually. Thank you

webartoli, 2024-07-17:
Yup, it works like a charm.
[image]