zigpy/zha-device-handlers #1773: [Device Support Request] Tint E14 RGB CCT

nicufarmache, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
My Tint E14 light doesn't correctly report the color capabilities (I can only change the color temperatire in HA)
[https://zigbee.blakadder.com/Muller_Licht_404019.html](https://zigbee.blakadder.com/Muller_Licht_404019.html)

**Describe the solution you'd like**
Add a quirk for it

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4635, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0x100f"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "MLI",
  "model": "tint-ExtendedColor",
  "class": "zhaquirks.mli.tintE14rgbcct.TintRGBCCTLight"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.9.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.61-v8+",
    "run_as_root": true
  },
[... 481 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```

```

</details>

**Additional context**
I already created a quirk for it and I'm currently using it successfully but I don't know how I should summit it here. I attached it to this issue
[tintE14rgbcct.py.zip](https://github.com/zigpy/zha-device-handlers/files/9606423/tintE14rgbcct.py.zip)

nicufarmache, 2022-09-22:
```python
"""Tint E14 RGB CCT."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class TintRGBCCTColorCluster(CustomCluster, Color):
    """Tint RGB+CCT Lighting custom cluster."""

    # Set correct capabilities to ct, xy, hs
    # Tint bulbs do not correctly report this attribute
    _CONSTANT_ATTRIBUTES = {0x400A: 0b11111}

class TintRGBCCTLight(CustomDevice):
    """Tint E15 RGB+CCT Lighting device."""

    signature = {
        MODELS_INFO: [("MLI", "tint-ExtendedColor")],
        ENDPOINTS: {
            1: {
                #   "profile_id": 260,
                #   "device_type": "0x010d",
                #   "in_clusters": [
                #     "0x0000",
                #     "0x0003",
                #     "0x0004",
                #     "0x0005",
                #     "0x0006",
                #     "0x0008",
                #     "0x0300",
                #     "0x1000",
                #     "0x100f"
                #   ],
                #   "out_clusters": [
                #     "0x0019"
                #   ]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    4111,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                #   "profile_id": 41440,
                #   "device_type": "0x0061",
                #   "in_clusters": [],
                #   "out_clusters": [
                #     "0x0021"
                #   ]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    TintRGBCCTColorCluster,
                    LightLink.cluster_id,
                    4111,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```

nicufarmache, 2022-10-24:
Is it ok if I create a PR with this file?

TheJulianJES, 2023-01-04:
Yes, please do so!

TheJulianJES, 2023-03-03:
Fixed with https://github.com/zigpy/zha-device-handlers/pull/2071

closed 2023-03-03 by TheJulianJES

rbierings, 2025-04-11:
Hi,

I have a Muller Tint outdoor chain of RGBCCT LED bulbs, which is Zigbee compatible. 

(https://hey-tint.de/collections/outdoor/products/tint-lichterkette-stella-1 ; sorry they only have German info here it seems?)

It used to pair and work with ZHA 
(although I was only able to control RGB color, CCT and on or off, but not the preprogrammed color effects  that can be operated manually from the device. )

**Problem:** 
somewhere last summer all of a sudden dropped out of the network. When  trying to repair it to ZHA, the device is found, can be operated for approximately 1 second and then becomes unresponsive. 
Looking in Signature tab under Zigbee Device management i noticed the following at the very end:

manufacturer": "MLI",
  "model": "tint-ExtendedColor",
  "class": "zhaquirks.mli.tintE14rgbcct.TintRGBCCTLight"

The tintE14RGBCCT is a single LED light bulb from the same manufacturer .  
Does this mean an incorrect quirk is being applied to my LED bulb cord ? How do I resolve this?