zigpy/zha-device-handlers #1774: [Device Support Request]

TommisGR, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
Recognize but no event are registered

**Describe the solution you'd like**
Event to register on button presses (single - double and hold)

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_wkai4ga5",
  "model": "TS0044",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
[... 453 lines cut ...]
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
[image]
Model: ESW-0ZAA-EU

TommisGR, 2022-09-20:
I used custom quirk "ts0044.py" and it appears to be working

javicalle, 2022-09-20:
Wich quirk have you used?
The `TuyaSmartRemote0044TO` seems to match your device signature.

TommisGR, 2022-09-20:
I used the ts0044.py as is from latest "zha-device-handlers-0.0.79.zip"

javicalle, 2022-09-20:
Ahhhh, so you are not in the 2022.09.X HA version, right?

TommisGR, 2022-09-21:
I am on 2022.9.5 but until i used the quirk buttons was not usable.

TommisGR, 2022-09-21:
Recheck and by removing custom quirk the " zhaquirks.tuya.ts0044.TuyaSmartRemote0044TO"
is activated so I must have 1st used the custom quirk then upgraded to 2022.9
so to summarize is works

closed 2022-09-21 by TommisGR