zigpy/zha-device-handlers #1775: [Device Support Request]

alexandre72, 2022-09-20:
Seems new  TS002 are sell, with power monitoring:
TS0002 - _TZ3000_bvrlqyj7

Seen as  ""zigpy.device.Device" so no recognized like other TS002. It don't use quirk :-( when enabling light 1 or light 2 they're always enabled together.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_bvrlqyj7",
  "model": "TS0002",
  "class": "zigpy.device.Device"
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-09-20:
So here we have a new `TS0002` flavor...

Can you try with that quirk?
```python
class Switch_2G_Metering(EnchantedDevice, CustomDevice):
    """Tuya 2 gang switch module with power metering."""

    signature = {
        MODEL: "TS0002",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 0x0702, 0x0b04, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=256
            # device_version=1
            # input_clusters=[4, 5, 6]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

You will need to add the new imports:
```python
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
```

alexandre72, 2022-09-20:
thanks you! good and fast job! all seems ok(I can switch L1 or L2), I can't test now power measure but entities are there

[image]

javicalle, 2022-09-20:
Probably you will need to remove the device from HA and pair it again.

alexandre72, 2022-09-22:
Ok today I tested with a load, power metering don't works(I don't need it, but maybe future other do). All evailable sensors: Electricalmeasurement, Electricalmeasurementrmscurrent... stay to 0

javicalle, 2022-09-22:
If you can enable the debug logs, attach any relevant info from your device (filter with the device Nwk: 0xeed5):
* https://www.home-assistant.io/integrations/zha/#debug-logging

Lobo2022, 2022-11-07:
Hello 
I have several  TS0002 - _TZ3000_bvrlqyj7 switches and I would be interested to have power metering working as well.
I tried the quirk above and the result is the same described by Alexandre72
I have enabled logs and the output (filtered by one of the switches) is attached below.

[Device Signature.txt](https://github.com/zigpy/zha-device-handlers/files/9953646/Device.Signature.txt)
[TS0002_0x26C5.log](https://github.com/zigpy/zha-device-handlers/files/9953647/TS0002_0x26C5.log)

Thank you in advance for help

javicalle, 2022-11-28:
I have checked the logs and I can't see any power/electtrical report from the device.
All the clusters requests are HA pollings and the device reports a 0 value for everything, so my guess will be that device don't support metering of any kind.
I have not seen that it is supported in Z2M either.

Lobo2022, 2022-11-28:
> I have checked the logs and I can't see any power/electrical report from the device. All the clusters requests are HA pollings and the device reports a 0 value for everything, so my guess will be that device don't support metering of any kind. I have not seen that it is supported in Z2M either.

Ok.
I've checked some sites where it is announced that this device supports power meter. For example this one:
https://zigbee.blakadder.com/Tuya_SML-02Z.html
So this information is not correct.
Thank you for your clarification

closed 2023-07-10 by github-actions[bot]

Jebolicious, 2024-09-30:
Hiya, I just wired up a TS0002 with a "new" variant:  _TZ3000_zmy4lslw and the switches work flawlessly as well as their states, but with this one I also get 0 on all power monitoring sensor values.

ZHA still mentions that these TS0002 have power meter support: https://zigbee.blakadder.com/Tuya_SML-02Z.html

Is there any new information about this being true?

I bought these thinking that they had the feature.

Kind regards,

Jebo