zigpy/zha-device-handlers #1776: [Device Support Request] zigbee wall thermostat not working. _TZE200_u9bfwha0

JonasEnstrom80, 2022-09-20:
I bought a zigbee wall thermostat for my bathroom underfloor heating. It joins my zigbee network but is not recognized as a climate entity in home assistant.

Model number: BHT-003GBLZB
https://www.aliexpress.com/item/1005004107863070.html?spm=a2g0o.cart.0.0.224738daF5KtbI&mp=1

Is it possible to get this thermostat to work with zha in home assistant?

[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_u9bfwha0",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Stockholm",
    "os_name": "Linux",
    "os_version": "5.15.67",
    "supervisor": "2022.09.1",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
[... 406 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

JonasEnstrom80, 2022-09-20:
Added my device signature "_TZE200_u9bfwha0"  to zhaquirks/tuya/ts0601_electric_heating.py

tested as a custom quirk and now it's working.

javicalle, 2022-09-21:
Do you mind to create the PR to add your device to the mainstream?

Just edit from the web the file content with the detected changes and follow the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_electric_heating.py

This way other users will be able to use the devices in ZHA thanks to you!

JonasEnstrom80, 2022-09-21:
I have created PR#1778, hope that's enough to get this working for everyone. 

Thank you for this incredible piece of software!

closed 2022-09-26 by dmulcahey

MattWestb, 2023-01-21:
@JonasEnstrom80 Hej Engström.
Can you testing the lats quirk from https://github.com/zigpy/zha-device-handlers/issues/1109#issuecomment-1397637752 then its looks that all black ones is working better with it but you have one white with little different design.
If its working we can updating all devices in this class to the updated quirk class but if its not working for your device we must leaving it in the old class for not braking it.

Thanks for contributing to our system!!

PR #1778 (2022-09-26): Add support for BHT-003GBLZB
Add signature ("_TZE200_u9bfwha0", "TS0601"),  for support of BHT-003GBLZB Electric floor heating termostat

Fixes: #1776