zigpy/zha-device-handlers #1780: [Device Support Request]  _TZ3000_x3ewpzyr doesn't show instant electrical measurement

davefx, 2022-09-22:
**Is your feature request related to a problem? Please describe.**
I've purchased and installed a Tuya Zibgee switch with power measurement support.

I can switch it on and off correctly, and I get correct reading for Electricalmeasurementrmsvoltage and Smartenergysummation attributes.
However, Electricalmeasurement attribute is always read as 0, and I get inconsistent and out-of-bounds values (like values about 40 A) for the Electricalmeasurementrmscurrent attribute.

**Describe the solution you'd like**
Both Electricalmeasurement and Electricalmeasurementrmscurrent attributes should return correct values.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_x3ewpzyr",
  "model": "TS0001",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
[... 627 lines cut ...]
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
```

</details>

**Additional context**
[image]

javicalle, 2022-09-22:
And here comes the new `TS0001` flavor...

Can you try with that quirk?
```python
class Switch_1G_Metering(EnchantedDevice, CustomDevice):
    """Tuya 1 gang switch module with power metering."""

    signature = {
        MODEL: "TS0001",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 0x0702, 0x0b04, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

You will need to add the new imports:
```python
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
```

Once the quirk is available, you have to remove the device from HA and pair it again.

Related to: #1775

blakadder, 2022-10-31:
is there a link to the device?

redQs, 2022-11-27:
how do i add the imports?

javicalle, 2022-11-28:
Copy in your local quirk folder the current file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts000x.py

Add the imports statements at the beginning of the file.
And add the new quirk at the bottom of file.

Save changes, restart HA and repair the device.

redQs, 2022-11-28:
thanks i got it working but its not the right quirk for my device.. :(
i opend an https://github.com/zigpy/zha-device-handlers/issues/1968

CharlyNaseweis, 2022-12-06:
> Copy in your local quirk folder the current file:
> 
> * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts000x.py
> 
> Add the imports statements at the beginning of the file. And add the new quirk at the bottom of file.
> 
> Save changes, restart HA and repair the device.

Hi, 

thank you for the answer. I have the same problem like redQs. 

I created a folder in config/custom_zha_quirks/

In there is a file named "ts000x.py" where I copied all the content from the provided link. 

Since I'm doing this the first time, I have a few questions:

Add the imports statements at the beginning of the file:

- what do I have to enter here? I simply don't have any idea? In which lines which information?

And add the new quirk at the bottom of file.

- I think, this will get clear, when I now which information where to enter.

-> Is there a kind of howto available somewhere?

Thank you, 
c

javicalle, 2022-12-07:
All the info is in my previous comment:
* https://github.com/zigpy/zha-device-handlers/issues/1780#issuecomment-1254814541

CharlyNaseweis, 2022-12-10:
Yes, thank you....but I'm really sorry. I still don't get it. My last programming was BASIC in school and 8088 Assembler, which is also some time ago. 

Would be really nice, if you can explain, what to enter where and where I can find this information in HA. I'm using the Zigbee integration from HA. 

From my understanding, I have to enter "cluster_id" or everything which is in black colour? And, after I found, entered the information, copy everything and put it at the end of the script????

Thank you, I really appreciate if you can help me...

thx
c

closed 2023-07-10 by github-actions[bot]