zigpy/zha-device-handlers #1782: [Device Support Request] Moes UFO-R11

geirskogul, 2022-09-24:
This device isn't available to use in ZHA, only zigbee2mqtt and I'd rather not change providers when my network is already working beautifully.

Would it be possible to port this device from Z2MQTT to ZHA?

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xf000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe004",
        "0xed00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3290_ot6ewjvmejq5ekhl",
  "model": "TS1201",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.09.1",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "midea_ac_lan": {
      "version": "v0.3.10",
      "requirements": []
    },
    "smartir": {
      "version": "1.17.6",
[... 372 lines cut ...]
    "unit": "%",
    "description": "Remaining battery in %",
    "value_min": 0,
    "value_max": 100
  },
  {
    "type": "numeric",
    "name": "voltage",
    "property": "voltage",
    "access": 1,
    "unit": "mV",
    "description": "Voltage of the battery in millivolts"
  },
  {
    "type": "numeric",
    "name": "linkquality",
    "property": "linkquality",
    "access": 1,
    "unit": "lqi",
    "description": "Link quality (signal strength)",
    "value_min": 0,
    "value_max": 255
  }
]
```

geirskogul, 2022-10-31:
Gosh I would pay to get this working with zha

neutrinus, 2023-04-18:
Seems like there is a PR already: https://github.com/zigpy/zha-device-handlers/pull/2336

darckense, 2023-10-17:
I bought the same universal remote control yesterday, and I tried to make it work on Home Assistant using ZHA, and I confirm that it didn't work.
I'm pretty new to home assistant, and my instance is working on a Raspberry 3.
Below the version number :
Home Assistant 2023.10.3 Supervisor 2023.10.0 Operating System 11.0

Tell me if there is anything pertinent to report to help with this issue.

Thanks

marcboivin, 2024-04-25:
I'm bumping this. Code looks good (I know my word means nothing here) works on my end with a custom quirks. I really want this into mainstream. Thanks for the contrib.

PR #2336 (2024-08-27): Add support for Tuya TS1201 IR blaster
TS1201 is an IR blaster, this quirk is able to send and to receive raw IR codes using Zigbee commands.
Received IR code is available in a Zigbee attribute.
Device is also known as:
- Moes UFO-R11
- Aubess ZXZIR-02

Should fix #1687 fix #1782 and fix #1955
closed 2024-08-27 by TheJulianJES