zigpy/zha-device-handlers #1783: (Device support) - Moes ms-105z

BrojJedan, 2022-09-24:
Hi!
First off, great job with ZHA, made my home smart with ease!
But I have a problem with moes ms-105z dimmer. It turns the light on or off, but when I select dimming level, the lights just turn off. Home assistant shows the lights as on and at the correct dimming level, but the lights are off. When I turn off the lights in HA and turn them on, then they turn on at the correct level. When I select the level, the lights tur off again the same as before.
Kind regards,
Matej

javicalle, 2022-09-24:
Probably related to #1748 and #1754 but without the device signature and some logs of the device operation it's hard to say.

BrojJedan, 2022-09-27:
This is probably the problem with my device.
Any estimates on when this will be added to official release?

javicalle, 2022-09-27:
Probably will be included in the tomorrow's beta release.
The official HA release will come October's the 5

BrojJedan, 2022-09-30:
Hi, 
thank you for the reply. I have tested this. It is sadly not working: 
**Failed to call service light/turn_on. TuyaLevelControl.command() got an unexpected keyword argument 'level'**
I can turn the light on and off without problems. But cannot set the dimming level.

javicalle, 2022-09-30:
Can you enable the debug logs and attach the full error?

BrojJedan, 2022-09-30:
I am not an expert with home assistant, so I hope this is correct..
`
[139928923961232] TuyaLevelControl.command() got an unexpected keyword argument 'level'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 546, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 291, in async_turn_on
    result = await self._level_channel.move_to_level_with_on_off(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 78, in wrapper
    result = await command(*args, **kwds)
TypeError: TuyaLevelControl.command() got an unexpected keyword argument 'level'
`

javicalle, 2022-09-30:
@BrojJedan can you post the device's signature? Please, be sure to wrap the code between ` ``` `.
Also will be helpfull if you can enable the ZHA debug logs and attach it again:
* https://www.home-assistant.io/integrations/zha/#debug-logging

javicalle, 2022-09-30:
@dmulcahey Sorry to ping you directly, but not sure if something can be broken in the beta version and I'm not able to follow the code.

Apparently, the changes from [HA#77676](https://github.com/home-assistant/core/pull/77676/files) can be related to the error.
But I'm not able to track how can affect from here:
* https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L291-L294

through here:
* https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/core/channels/base.py#L72-L78

to here:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L1276-L1283

(or maybe here):
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/mcu/__init__.py#L451-L458

If you can guide me in the right direction I can check where can be the problem.

dmulcahey, 2022-09-30:
Hit me up on discord

BrojJedan, 2022-10-01:
Device signature:

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_la2c2uo9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}`

BrojJedan, 2022-10-01:
I don't get anything in system logs. I enabled the logs just like in the link you sent.

javicalle, 2022-10-01:
I think the issue is localized and we are working to fix it.

closed 2022-10-03 by dmulcahey

javicalle, 2022-10-04:
@BrojJedan can you try with the `2022.10.0b5` version and report?