zigpy/zha-device-handlers #1784: [Device Support Request] Support for tuya 2 gang wall plug _TZ3000_raviyuvk

peterkh, 2022-09-25:
**Is your feature request related to a problem? Please describe.**
This plug is able to be added into ZHA fine, but no quirk is associated with it. Power monitoring figures are out by a factor of 1000 for current and 100 for energy

**Describe the solution you'd like**
A quirk added so that energy monitoring values are correct.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_raviyuvk",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.9.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.10.0-18-amd64",
    "run_as_root": true
[... 592 lines cut ...]
        "out_clusters": {}
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
This is an Australian plug. 
[image]
screenshot is with very minimal load on plug. Current should be 0.01A and 0.79 for kWh

Thanks!

javicalle, 2022-09-25:
Can you just try to update the `ac_current_divisor (id: 0x0603)` attribute from the `ElectricalMeasurement (Endpoint id: 1, Id: 0x0b04, Type: in)` cluster?
This can be done from device info, manage clusters option.

I'm not sure but for the energy I suppose that will be the `divisor (id: 0x0302)` attribute from the `Metering (Endpoint id: 1, Id: 0x0702, Type: in)` cluster.

As it is a 2 plug device, I'm not sure if you must repeat the update for both endpoints or just for the endpoint 1.

peterkh, 2022-09-25:
I gave that a try, but looks like the update isn't working. If I do a set, followed by a get its still set at None and I can see the following line in the home assistant log:

```
homeassistant_1  | 2022-09-25 22:18:33.727 ERROR (MainThread) [zigpy.zcl] [0x3DA2:1:0x0b04] Failed to convert attribute 0x0603 from None (<class 'str'>) to type <class 'zigpy.types.basic.uint16_t'>: invalid literal for int() with base 10: 'None'
```

MattWestb, 2022-09-25:
The device is not loading the quirk that shall fixing the power measurement and need one new device class in the https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py quirk.

javicalle, 2022-09-25:
If you want to try, this is my proposed quirk:
```python
class Plug_TZ3000_2AC(CustomDevice):
    """TS0011F 2 outlet plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=["0x0000", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04"]
            # output_clusters=["0x000a", "0x0019"]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=81
            # device_version=1
            # input_clusters=["0x0004", "0x0005", "0x0006", "0x0702", "0x0b04"]
            # output_clusters=[25, 10]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

Just enable the local quirk configuration, copy the current `ts011f_plug.py` file inside the `custom_zha_quirks` folder and add the suggested class at the end of the file.
Restart HA and repair the device.

peterkh, 2022-09-27:
Just gave it a try and works a treat, thank you!

[image]
With real load on it.

TheJulianJES, 2023-01-04:
Has the quirk been merged into the repo?

PR #2760 (2023-11-23): Add Tuya 2 gang wall plug TS011F `_TZ3000_raviyuvk` variation
## Proposed change
Have been using a custom quirk for a while for my tuya plug given to me in issue #1784. Would like to contribute it back for others.

## Additional information
Fixes #1784 

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2023-11-23 by TheJulianJES