zigpy/zha-device-handlers #1785: _TZE200_fm5yck8a TS0601

nekdo4, 2022-09-25:
Hello,
I am experiencing problems regarding below mentioned item. 
[https://www.aliexpress.com/item/1005004242288793.html?spm=a2g0o.order_detail.0.0.3d66f19cdtUBsi](https://www.aliexpress.com/item/1005004242288793.html?spm=a2g0o.order_detail.0.0.3d66f19cdtUBsi)

I can add the device to ZHA but no entities are visible. I have tried custom quirks with no result. I am new it this area.

In case someone else has experienced a similar problem, I would like to know how to solve it.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_fm5yck8a",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.67",
    "supervisor": "2022.09.1",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_miot": {
      "version": "0.6.12",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.6",
        "micloud>=0.3"
[... 272 lines cut ...]
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```

```
</details>

[image]
[image]

javicalle, 2022-09-26:
This is a curious device.
I think the normal switches (4 left) should be controlled by the `TuyaQuadrupleSwitchTO` quirk.
You can enable the local quirks and copy the current `ts0601_switch.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py

Then update the `TuyaQuadrupleSwitchTO` class like this:
```python
class TuyaQuadrupleSwitchTO(TuyaSwitch):
    """Tuya quadruple channel switch time on out cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_aqnazj70", "TS0601"),
            ("_TZE200_1ozguk6x", "TS0601"),
            ("_TZE200_fm5yck8a", "TS0601"),
            ("_TZE200_k6jhsr0q", "TS0601"),
        ],
        ENDPOINTS: {
            .../...
```

Repair the device after enable the local quirk.

You will need to enable the debug logs to know what is reporting the other 2 switches (2 right):
* https://www.home-assistant.io/integrations/zha/#debug-logging

Once we get that info we can see what can be done.

nekdo4, 2022-09-26:
Thank you for your reply, I did as you advised but the left four normal switches  on the device do not react and the other two (right) can only turn off first two light switches in home assistant. I am sending diagnostic data and thank you again for your help  

<details>
<summary>Diagnostic Data</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.67",
    "supervisor": "2022.09.1",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_miot": {
      "version": "0.6.12",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.6",
        "micloud>=0.3"
      ]
    },
    "zha_toolkit": {
      "version": "v0.8.16",
      "requirements": []
    },
    "localtuya": {
      "version": "4.1.0",
      "requirements": []
    },
    "xiaomi_cloud_map_extractor": {
      "version": "v2.2.0",
      "requirements": [
        "pillow",
        "pybase64",
        "python-miio",
        "requests",
        "pycryptodome"
      ]
    },
    "hacs": {
      "version": "1.27.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "mikrotik_router": {
      "version": "0.0.0",
      "requirements": [
        "librouteros>=3.2.0",
        "mac-vendor-lookup>=0.1.12"
      ]
    },
    "sonoff": {
      "version": "3.2.0",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
[... 361 lines cut ...]
        "in_clusters": {
          "0x0006": {
            "endpoint_attribute": "on_off",
            "attributes": {},
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "on_off"
              },
              "0x4003": {
                "attribute_name": "start_up_on_off"
              }
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}

```

</details>

[image]

javicalle, 2022-09-26:
Look at the logs for any info from your device (check the Nwk value for your device in the "Zigbee info" option).

nekdo4, 2022-09-27:
Hello, I followed your advice but I do not know how to continue. I am sending you the log and very grateful for your help.

<details>
<summary>LOG</summary>
```
2022-09-27 21:16:29.316 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-09-27 21:16:29.316 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 5 value: TS0601
2022-09-27 21:16:29.929 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-09-27 21:16:29.929 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 5 value: TS0601
2022-09-27 21:16:29.929 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0xef00] Attribute id: 61184 value: 1.0.0
2022-09-27 21:16:29.929 DEBUG (MainThread) [zigpy.appdb] [0x1496:2:0x0006] Attribute id: 0 value: 0
2022-09-27 21:16:29.930 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0006] Attribute id: 0 value: 0
2022-09-27 21:16:41.763 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x1496](_TZE200_fm5yck8a TS0601) restored as 'available', last seen: 0:03:11 ago, consider_unavailable_time: 7200 seconds
2022-09-27 21:16:41.873 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1496](TS0601): started initialization
2022-09-27 21:16:41.873 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:ZDO](TS0601): 'async_initialize' stage succeeded
2022-09-27 21:16:42.294 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0006]: initializing channel: from_cache: True
2022-09-27 21:16:42.294 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-27 21:16:42.294 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-09-27 21:16:42.295 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-09-27 21:16:42.295 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0006]: Reading attributes in chunks: ['on_off']
2022-09-27 21:16:42.295 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0006]: finished channel initialization
2022-09-27 21:16:42.296 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0000]: initializing channel: from_cache: True
2022-09-27 21:16:42.296 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0000]: finished channel initialization
2022-09-27 21:16:42.297 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0019]: initializing channel: from_cache: True
2022-09-27 21:16:42.297 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:1:0x0019]: finished channel initialization
2022-09-27 21:16:42.306 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:2:0x0006]: initializing channel: from_cache: True
2022-09-27 21:16:42.307 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:2:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-27 21:16:42.307 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:2:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-09-27 21:16:42.307 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:2:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-09-27 21:16:42.308 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:2:0x0006]: Reading attributes in chunks: ['on_off']
2022-09-27 21:16:42.308 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:2:0x0006]: finished channel initialization
2022-09-27 21:16:42.308 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:3:0x0006]: initializing channel: from_cache: True
2022-09-27 21:16:42.309 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:3:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-27 21:16:42.309 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:3:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-09-27 21:16:42.309 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:3:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-09-27 21:16:42.310 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:3:0x0006]: Reading attributes in chunks: ['on_off']
2022-09-27 21:16:42.310 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:3:0x0006]: finished channel initialization
2022-09-27 21:16:42.310 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:4:0x0006]: initializing channel: from_cache: True
2022-09-27 21:16:42.311 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:4:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-27 21:16:42.311 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1496:4:0x0006]: Reading attributes in chunks: ['start_up_on_off']
[... cut ...] ion=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=41, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-09-27 21:18:22.053 DEBUG (MainThread) [zigpy.zcl] [0x1496:1:0xef00] Decoded ZCL frame: MoesSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=127, dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>)))
```
</details>

javicalle, 2022-09-27:
Keep the breaklines:

````
<details>
<summary>LOG</summary>

```
2022-09-27 21:16:29.316 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a 2022-09-27 21:16:29.316 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 5 value: TS0601 2022-09-27 21:16:29.929 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a 2022-09-27 21:16:29.929 DEBUG (MainThread) [zigpy.appdb] [0x1496:1:0x0000] Attribute id: 5 value: TS0601 2022-09-27 21:16:29.929
```

</details>
````

javicalle, 2022-09-27:
Quick and dirty alternative:
1. restore the file version (or remove the `_TZE200_fm5yck8a` line)
2. add at the end of the `` file this classes:
```python
class AtlanticSwitchManufCluster(MoesSwitchManufCluster):
    """On/Off Tuya cluster with extra device attributes."""

    dp_to_attribute: Dict[
        int, DPToAttributeMapping
    ] = MoesSwitchManufCluster.dp_to_attribute.copy()
    dp_to_attribute.update(
        {
            19: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=19,
            ),
        }
    )
    dp_to_attribute.update(
        {
            20: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=20,
            ),
        }
    )
    dp_to_attribute.update(
        {
            21: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=21,
            ),
        }
    )
    dp_to_attribute.update(
        {
            22: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=22,
            ),
        }
    )

    data_point_handlers = MoesSwitchManufCluster.data_point_handlers.copy()
    data_point_handlers.update({19: "_dp_2_attr_update"})
    data_point_handlers.update({20: "_dp_2_attr_update"})
    data_point_handlers.update({21: "_dp_2_attr_update"})
    data_point_handlers.update({22: "_dp_2_attr_update"})

class AtlanticSwitch(TuyaSwitch):
    """Tuya quadruple channel switch time on out cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fm5yck8a", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AtlanticSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            19: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            20: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            22: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```
3. delete any `__pycache__` folder
4. restart HA
5. repair the device

It must show 6 lights. 
Lights 3-6 would reflect the 4 left gangs.
Lights 1-2 will react somehow to the 2 right gangs. Is not intended to be implemented this way, just for testing pusposes.
Backlight could be controlled with the `backlight_mode` attribute from cluster `AtlanticSwitchManufCluster`. Maybe also the `power_on_state`.

Logs of any incorrect behavior are welcomed and necessary to analyze the behavior (and possible solution).

nekdo4, 2022-09-28:
Thank you for your advice, here is the result.
There was immediately an error Dict not defined so I added `from typing import Dict` to the beginning of the file, and after that the error was  DPToAttributeMapping not defined, so I added  `from zhaquirks.tuya.mcu import` 

```
DPToAttributeMapping,
EnchantedDevice,
TuyaDPType,
TuyaMCUCluster,
TuyaPowerConfigurationCluster,
```
I do not know if this is everything needed, I copied it from a different file. After these modifications 4 switches function without problems, however 2 right switches again can only turn off the lights but not on. As they are designed for scenes would it be possible to set them for events instead of entities?  And I would also like to ask how could `backlight_mode` be used.
Thank you in advance

<details>
<summary>LOG</summary>

```
2022-09-28 12:15:36.611 DEBUG (MainThread) [zigpy.appdb] [0x327a:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-09-28 12:15:36.611 DEBUG (MainThread) [zigpy.appdb] [0x327a:1:0x0000] Attribute id: 5 value: TS0601
2022-09-28 12:15:38.444 DEBUG (MainThread) [zigpy.appdb] [0x327a:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-09-28 12:15:38.444 DEBUG (MainThread) [zigpy.appdb] [0x327a:1:0x0000] Attribute id: 5 value: TS0601
2022-09-28 12:15:38.444 DEBUG (MainThread) [zigpy.appdb] [0x327a:19:0x0006] Attribute id: 0 value: 1
2022-09-28 12:15:38.444 DEBUG (MainThread) [zigpy.appdb] [0x327a:20:0x0006] Attribute id: 0 value: 1
2022-09-28 12:15:38.445 DEBUG (MainThread) [zigpy.appdb] [0x327a:21:0x0006] Attribute id: 0 value: 1
2022-09-28 12:15:38.445 DEBUG (MainThread) [zigpy.appdb] [0x327a:22:0x0006] Attribute id: 0 value: 1
2022-09-28 12:15:38.445 DEBUG (MainThread) [zigpy.appdb] [0x327a:1:0xef00] Attribute id: 61184 value: 1.0.0
2022-09-28 12:15:38.445 DEBUG (MainThread) [zigpy.appdb] [0x327a:2:0x0006] Attribute id: 0 value: 0
2022-09-28 12:15:38.445 DEBUG (MainThread) [zigpy.appdb] [0x327a:1:0x0006] Attribute id: 0 value: 0
2022-09-28 12:15:48.502 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x327A](_TZE200_fm5yck8a TS0601) restored as 'available', last seen: 0:44:39 ago, consider_unavailable_time: 7200 seconds
2022-09-28 12:15:48.544 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x327A](TS0601): started initialization
2022-09-28 12:15:48.545 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:ZDO](TS0601): 'async_initialize' stage succeeded
2022-09-28 12:15:48.663 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0006]: initializing channel: from_cache: True
2022-09-28 12:15:48.663 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-28 12:15:48.663 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-09-28 12:15:48.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-09-28 12:15:48.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0006]: Reading attributes in chunks: ['on_off']
2022-09-28 12:15:48.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0006]: finished channel initialization
2022-09-28 12:15:48.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0000]: initializing channel: from_cache: True
2022-09-28 12:15:48.664 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0000]: finished channel initialization
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0019]: initializing channel: from_cache: True
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:1:0x0019]: finished channel initialization
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:2:0x0006]: initializing channel: from_cache: True
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:2:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:2:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:2:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-09-28 12:15:48.665 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:2:0x0006]: Reading attributes in chunks: ['on_off']
2022-09-28 12:15:48.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:2:0x0006]: finished channel initialization
2022-09-28 12:15:48.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:19:0x0006]: initializing channel: from_cache: True
2022-09-28 12:15:48.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:19:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-09-28 12:15:48.666 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x327A:19:0x0006]: Reading attributes in chunks: ['start_up_on_off']
[... cut ...]
2022-09-28 12:55:28.551 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x327A](TS0601): Update device availability - device available: True - new availability: True - changed: False
2022-09-28 12:56:38.551 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x327A](TS0601): Device seen - marking the device available and resetting counter
2022-09-28 12:56:38.551 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x327A](TS0601): Update device availability - device available: True - new availability: True - changed: False
```

</details>

javicalle, 2022-09-29:
> As they are designed for scenes would it be possible to set them for events instead of entities?

Yes, that is the intention. I'll come back with some update.

> And I would also like to ask how could `backlight_mode` be used.

That must be done from the 'Manage clusters' option from the device view.
From the `AtlanticSwitchManufCluster`, select the `backlight_mode` attribute. You can 'get zigbee attribute' to see the current value (if any), and then try to set a new value:
[image]

You can try with 0-2 values or even greather ones, it depends on device. For example there are:
```python
class MoesBacklight(t.enum8):
    """MOES switch backlight mode enum."""

    off = 0x00
    light_when_on = 0x01
    light_when_off = 0x02
    freeze = 0x03
```

nekdo4, 2022-09-29:
Good evening,
thank you for your explanation, however backlight is still not functioning.. I cannot change it even in the original aplication tuya (I still have tuya gateway so I could test it). So it might be possible that with this switch backlight cannot be set. If you find it interesting or helpful I am sending the log again.

<details>
<summary>LOG</summary>

```
2022-09-29 20:28:12.542 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0xb4e7, 80:4b:50:ff:fe:44:69:9a, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x246c]
2022-09-29 20:28:12.543 INFO (MainThread) [zigpy.application] New device 0xb4e7 (80:4b:50:ff:fe:44:69:9a) joined the network
2022-09-29 20:28:12.544 DEBUG (MainThread) [zigpy.device] [0xb4e7] Scheduling initialization
2022-09-29 20:28:12.552 INFO (MainThread) [zigpy.device] [0xb4e7] Requesting 'Node Descriptor'
2022-09-29 20:28:12.556 DEBUG (MainThread) [zigpy.device] [0xb4e7] Extending timeout for 0x2c request
2022-09-29 20:28:12.558 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 80:4b:50:ff:fe:44:69:9a/0xb4e7
2022-09-29 20:28:12.619 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=114), 144, -64, 0xb4e7, 255, 255, b'\x85\xe7\xb4\x9aiD\xfe\xffPK\x80\x8e']
2022-09-29 20:28:12.620 INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0xb4e7, 80:4b:50:ff:fe:44:69:9a
2022-09-29 20:28:12.621 INFO (MainThread) [zigpy.application] Device 0xb4e7 (80:4b:50:ff:fe:44:69:9a) joined the network
2022-09-29 20:28:12.622 DEBUG (MainThread) [zigpy.device] [0xb4e7] Scheduling initialization
2022-09-29 20:28:12.622 DEBUG (MainThread) [zigpy.device] [0xb4e7] Canceling old initialize call
2022-09-29 20:28:12.632 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4E7 ieee=80:4b:50:ff:fe:44:69:9a is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x85\xe7\xb4\x9aiD\xfe\xffPK\x80\x8e'
2022-09-29 20:28:12.633 DEBUG (MainThread) [zigpy.zdo] [0xb4e7:zdo] ZDO request ZDOCmd.Device_annce: [0xB4E7, 80:4b:50:ff:fe:44:69:9a, 142]
2022-09-29 20:28:12.649 INFO (MainThread) [zigpy.device] [0xb4e7] Requesting 'Node Descriptor'
2022-09-29 20:28:12.650 DEBUG (MainThread) [zigpy.device] [0xb4e7] Extending timeout for 0x2e request
2022-09-29 20:28:12.659 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 80:4b:50:ff:fe:44:69:9a/0xb4e7
2022-09-29 20:28:12.796 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xB4E7, EmberApsFrame(profileId=0, clusterId=2, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=46), 47, b'.\xe7\xb4')
2022-09-29 20:28:13.163 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32770, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=115), 192, -52, 0xb4e7, 255, 255, b'.\x00\xe7\xb4\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00']
2022-09-29 20:28:13.164 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4E7 ieee=80:4b:50:ff:fe:44:69:9a is_initialized=False> from ep 0 to ep 0, cluster 32770: b'.\x00\xe7\xb4\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
2022-09-29 20:28:13.172 INFO (MainThread) [zigpy.device] [0xb4e7] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
2022-09-29 20:28:13.175 INFO (MainThread) [zigpy.device] [0xb4e7] Discovering endpoints
2022-09-29 20:28:13.178 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xB4E7, EmberApsFrame(profileId=0, clusterId=5, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=48), 49, b'0\xe7\xb4')
2022-09-29 20:28:13.340 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32773, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=116), 228, -43, 0xb4e7, 255, 255, b'0\x00\xe7\xb4\x01\x01']
[... cut ...]
2022-09-29 20:37:26.678 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=241, dp=19, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>)))
2022-09-29 20:37:40.793 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB4E7](TS0601): Device seen - marking the device available and resetting counter
2022-09-29 20:37:40.793 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB4E7](TS0601): Update device availability - device available: True - new availability: True - changed: False
```

</details>

javicalle, 2022-09-29:
From your logs I can see that HA tries to write the attribute and the device respond with a success:
```
2022-09-29 20:31:09.653 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] write_attributes --> record: Attribute(attrid=0x8001, value=TypeValue(type=MoesBacklight, value=<MoesBacklight.freeze: 3>))

2022-09-29 20:31:09.659 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4098, tsn=136, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)

2022-09-29 20:31:09.661 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=135, dp=15, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x03', *payload=<enum8.undefined_0x03: 3>)))

2022-09-29 20:31:09.821 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received ZCL frame: b'\x0c\x02\x10\x88\x0b\x00\x00'

2022-09-29 20:31:09.829 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
```

So, probably device don't support the function. If Tuya app can't set it either will be very probable.
One last test that you can do will be (keeping the debug logs activated) power off and power on the device and listen to the logs. Usually device will report all its attributes and maybe there are some information there.

nekdo4, 2022-10-03:
Good evening,
so I tried to do pretty much everything which came to my mind, the log is too long  so I had to convert it to a text file.  I do not know if it is helpful, but I hope there are all options. If there is something I forgot to test or something I should change, I will do it again. 
Thank you for your help

[LOG_TZE200_fm5yck8a.txt](https://github.com/zigpy/zha-device-handlers/files/9700505/LOG_TZE200_fm5yck8a.txt)

javicalle, 2022-10-08:
I do not see any reports from the device apart from those already known (1, 2, 19, 20, 21, 22).
Assuming that it is not possible to control the state of the `backlight` my quirk proposal would be:
```python
class AtlanticSwitchManufCluster(TuyaOnOffManufCluster):
    """On/Off Tuya cluster with extra device attributes."""

    dp_to_attribute: Dict[
        int, DPToAttributeMapping
    ] = TuyaOnOffManufCluster.dp_to_attribute.copy()
    dp_to_attribute.update(
        {
            19: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=19,
            ),
        }
    )
    dp_to_attribute.update(
        {
            20: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=20,
            ),
        }
    )
    dp_to_attribute.update(
        {
            21: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=21,
            ),
        }
    )
    dp_to_attribute.update(
        {
            22: DPToAttributeMapping(
                TuyaOnOff.ep_attribute,
                "on_off",
                dp_type=TuyaDPType.BOOL,
                endpoint_id=22,
            ),
        }
    )

    data_point_handlers = TuyaOnOffManufCluster.data_point_handlers.copy()
    data_point_handlers.update({19: "_dp_2_attr_update"})
    data_point_handlers.update({20: "_dp_2_attr_update"})
    data_point_handlers.update({21: "_dp_2_attr_update"})
    data_point_handlers.update({22: "_dp_2_attr_update"})

class AtlanticSwitch(TuyaSwitch):
    """Tuya quadruple channel switch time on out cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fm5yck8a", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AtlanticSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster, Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster],
            },
            19: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            20: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            22: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }
```

With this quirk only 4 buttons will apear in HA.
The other 2 buttons have to be handled through events.
`short`, `long` and `double` events have been implemented, but it is not certain that they will work. Try different keystrokes and look at the logs to see what information they give us. 
It is also possible to listen to `zha_event` events from the developer view. You may also be able to obtain information about the events generated by the device there.

nekdo4, 2022-10-12:
Good evening,
so I changed it, filled in the imports and I would say in the log that the "scenes" is reacting to the switches, however in reality it does nothing as for the automatisation. Furthermore there is new switch, which does not react and if I try to turn it on, it automatically turns off. 
Thank you very much for your help. Unfortunately I am not able to understand very well what is going on in the log.

<details>
<summary>LOG</summary>

```
2022-10-12 21:05:14.753 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=231), 216, -46, 0xb4e7, 255, 255, b'\t_\x01\x00\t\x01\x04\x00\x01\x00']
2022-10-12 21:05:14.754 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=231, profile_id=260, cluster_id=61184, data=Serialized[b'\t_\x01\x00\t\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
2022-10-12 21:05:14.755 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received ZCL frame: b'\t_\x01\x00\t\x01\x04\x00\x01\x00'
2022-10-12 21:05:14.756 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=95, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:05:14.757 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-12 21:05:14.758 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received command 0x01 (TSN 95): get_data(data=TuyaCommand(status=0, tsn=9, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-10-12 21:05:14.775 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=95, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-12 21:05:14.776 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-10-12 21:05:14.777 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), dst_ep=1, source_route=None, extended_timeout=False, tsn=95, profile_id=260, cluster_id=61184, data=Serialized[b'\x18_\x0b\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-12 21:05:14.779 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb4e7, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=95), 19, b'\x18_\x0b\x01\x00')
2022-10-12 21:05:14.922 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=232), 216, -46, 0xb4e7, 255, 255, b'\t_\x01\x00\t\x01\x04\x00\x01\x00']
2022-10-12 21:05:14.923 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4E7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=232, profile_id=260, cluster_id=61184, data=Serialized[b'\t_\x01\x00\t\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
2022-10-12 21:05:14.924 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Received ZCL frame: b'\t_\x01\x00\t\x01\x04\x00\x01\x00'
2022-10-12 21:05:14.925 DEBUG (MainThread) [zigpy.zcl] [0xB4E7:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=95, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...] x0b\x01\x86'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-12 21:07:15.633 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xb4e7, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=104), 54, b'\x18h\x0b\x01\x86')

```

</details>

[image]

javicalle, 2022-10-16:
> there is new switch, which does not react and if I try to turn it on, it automatically turns off.

My fault. Fixed in the code.
Just remove the `TuyaOnOff,` in the replacement part that is not needed.

Also, the escene keys DP1 and DP2 are getting an ENUM type (that don't match the actual type) and don't like it.
Before trying to fix anything, can you try making different keystrokes (short/long/double) from the physical device and try to identify the values in the logs?
I hope you will find logs with this content but with different values in `raw`:
```
TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
```
I am interested in relating the keystrokes with that values. You can also try with diferents keystrokes to see if there are more values (triple? quadruple? release event?)

nekdo4, 2022-11-13:
Hello,
I am sorry for the late reply, but I had a lot of work with testing and not enough time for anything else.
So:

Scene 1 switch
1-click
TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2-click
TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
LongClick 
TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))

Scene 2 switch
1-click
TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2-click
TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
LongClick 
TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))

I do not see any difference regardless of the way I click, when 2-click it just repeats the same, as if I clicked twice. But I think this is normal because in the tuya aplication you cannot use anything else than 1-click, everything else does the same.

As usual thank you for your advice and here is the whole log:

<details>
<summary>LOG</summary>

```
2022-11-13 23:40:59.211 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4996](TS0601): power source: Mains
2022-11-13 23:40:59.211 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4996](TS0601): completed initialization
2022-11-13 23:41:42.280 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x4996, Relays=[])
2022-11-13 23:41:42.332 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x4996, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=76, SecurityUse=<Bool.false: 0>, TimeStamp=141219, TSN=0, Data=b'\x09\x4B\x01\x00\x18\x01\x04\x00\x01\x00', MacSrcAddr=0x4996, MsgResultRadius=29)
2022-11-13 23:41:42.336 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4996), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tK\x01\x00\x18\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=76, rssi=None)
2022-11-13 23:41:42.337 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Received ZCL frame: b'\tK\x01\x00\x18\x01\x04\x00\x01\x00'
2022-11-13 23:41:42.337 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=75, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-13 23:41:42.337 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=24, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-11-13 23:41:42.337 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Received command 0x01 (TSN 75): get_data(data=TuyaCommand(status=0, tsn=24, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-11-13 23:41:42.338 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] No datapoint handler for TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2022-11-13 23:41:42.338 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=75, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-11-13 23:41:42.338 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-11-13 23:41:42.338 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4996), dst_ep=1, source_route=None, extended_timeout=False, tsn=75, profile_id=260, cluster_id=61184, data=Serialized[b'\x18K\x0b\x01\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-13 23:41:42.339 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4996), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=75, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x4B\x0B\x01\x86')
[... cut ...] data=Serialized[b'\x18U\x0b\x01\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-13 23:54:31.928 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4996), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=85, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x55\x0B\x01\x86')
```

</details>

javicalle, 2022-11-29:
Sorry for the late reply.

In the logs I can see that quirk complins about DP1 and DP2:
```
2022-11-13 23:49:44.370 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Received command 0x01 (TSN 81): get_data(data=TuyaCommand(status=0, tsn=24, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-11-13 23:49:44.370 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] No datapoint handler for TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
.../...
2022-11-13 23:51:17.215 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=24, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-11-13 23:51:17.216 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] Received command 0x01 (TSN 82): get_data(data=TuyaCommand(status=0, tsn=24, datapoints=[TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-11-13 23:51:17.216 DEBUG (MainThread) [zigpy.zcl] [0x4996:1:0xef00] No datapoint handler for TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
```

There is also a DP25 with a value 395. No idea what it can be.

Can you post your current quirk?
Please, can you also refresh me the current state? What is working and what not?

nekdo4, 2022-11-30:
Hello,
so DP=25 I do not know, what this is, it appears always after the first testing of switch 1 - click, long click, double click and then dp=25, not before this regardless of whatever I do. It is now on a separate testing set where I have only coordinator and the switch and nothing else. So I really have no idea what this could be.
As of this time, all 4 light switches are functioning, but the switches themself do nothing except showing something in the log, it is not reacting in the automatization.

here is the current log:

<details>
<summary>Quirk</summary>

```
"""Tuya DP based switches."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    CLUSTER_ID,
    COMMAND,
    COMMAND_MOVE,
    COMMAND_MOVE_SATURATION,
    COMMAND_OFF,
    COMMAND_ON,
    COMMAND_STEP,
    COMMAND_STOP,
    COMMAND_STOP_MOVE_STEP,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PARAMS,
    PROFILE_ID,
    RIGHT,
    ROTATED,
    ROTATED_FAST,
    ROTATED_SLOW,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)

from zhaquirks.tuya import (
    TuyaSwitch,
    TuyaSmartRemoteOnOffCluster,
    TuyaZBE000Cluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
    MoesSwitchManufCluster,
    TuyaOnOff,
[... 838 lines cut ...]
            22: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }
```

</details>

As always thank you a lot for your help.
Regards

javicalle, 2022-11-30:
Just to be sure:
* can you control from HA the 4 buttons on the left? 
* is HA updating the status for the 4 buttons on the left when you operate them from the physical device?

The 2 buttons in the right just trigger commands. And I believe that they are not sending events to HA, right?

Replace the `data_point_handlers` in the `class AtlanticSwitchManufCluster(TuyaOnOffManufCluster):` like that:
```python
    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        19: "_dp_2_attr_update",
        20: "_dp_2_attr_update",
        21: "_dp_2_attr_update",
        22: "_dp_2_attr_update",
    }
```

Save changes, delete any `__pycache__` folder, restart HA and repair your device.

javicalle, 2022-11-30:
Please also can try to modify `class AtlanticSwitch(TuyaSwitch):` the ENDPOINT 2 in the `replacement` part like that:
```python
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster],
                OUTPUT_CLUSTERS: [],
            },
```

Attach logs from operation.

nekdo4, 2022-11-30:
So I tested it and cannot see a visible change, see the log below:

<details>
<summary>LOG</summary>

```
2022-11-30 23:51:52.899 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-11-30 23:51:52.899 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:1:0x0000] Attribute id: 5 value: TS0601
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:1:0x0000] Attribute id: 5 value: TS0601
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:19:0x0006] Attribute id: 0 value: 1
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:20:0x0006] Attribute id: 0 value: 1
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:21:0x0006] Attribute id: 0 value: 1
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:22:0x0006] Attribute id: 0 value: 1
2022-11-30 23:51:52.903 DEBUG (MainThread) [zigpy.appdb] [0xc5ed:1:0xef00] Attribute id: 61184 value: 1.0.0
2022-11-30 23:51:55.545 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0xC5ED](_TZE200_fm5yck8a TS0601) restored as 'available', last seen: 0:02:10 ago, consider_unavailable_time: 7200 seconds
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC5ED](TS0601): started initialization
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:ZDO](TS0601): 'async_initialize' stage succeeded
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0x0000]: initializing channel: from_cache: True
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0x0000]: finished channel initialization
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0xef00]: initializing channel: from_cache: True
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0xef00]: finished channel initialization
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0x0006]: initializing channel: from_cache: True
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0x0006]: finished channel initialization
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0x0019]: initializing channel: from_cache: True
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:1:0x0019]: finished channel initialization
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:19:0x0006]: initializing channel: from_cache: True
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:19:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:19:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-11-30 23:51:55.549 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:19:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:19:0x0006]: Reading attributes in chunks: ['on_off']
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:19:0x0006]: finished channel initialization
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:20:0x0006]: initializing channel: from_cache: True
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:20:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:20:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:20:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:20:0x0006]: Reading attributes in chunks: ['on_off']
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:20:0x0006]: finished channel initialization
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:21:0x0006]: initializing channel: from_cache: True
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:21:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:21:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:21:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-11-30 23:51:55.550 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC5ED:21:0x0006]: Reading attributes in chunks: ['on_off']
[... cut ...] ata=Serialized[b'\x18,\x0b\x01\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-30 23:52:39.306 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xC5ED), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=44, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x2C\x0B\x01\x86')

```

</details>

javicalle, 2022-12-01:
I caught it...

This version (just the changed classes) would trigger the scene buttons events:
<details>
<summary>ts0601_switch.py</summary>

```python
class AtlanticSwitchManufCluster(TuyaOnOffManufCluster):
    """On/Off Tuya cluster with extra device attributes."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaSmartRemoteOnOffCluster.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            TuyaSmartRemoteOnOffCluster.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=2,
        ),
        19: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=19,
        ),
        20: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=20,
        ),
        21: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=21,
        ),
        22: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=22,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        19: "_dp_2_attr_update",
        20: "_dp_2_attr_update",
        21: "_dp_2_attr_update",
        22: "_dp_2_attr_update",
    }

class AtlanticSwitch(TuyaSwitch):
    """Tuya quadruple channel switch time on out cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fm5yck8a", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AtlanticSwitchManufCluster,
                ],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster, Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [TuyaSmartRemoteOnOffCluster],
            },
            19: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            20: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            22: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }
```

</details>

Can you confirm if the 'normal' switches (the 4 on the left) are working as expected?

nekdo4, 2022-12-01:
Thank you, but nothing has changed., 4 switches for lights are working OK and both switches for the automatisation of the scenes do not run.

<details>
<summary>LOG</summary>

```
2022-12-01 21:08:40.255 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-12-01 21:08:40.255 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:1:0x0000] Attribute id: 5 value: TS0601
2022-12-01 21:08:40.259 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:1:0x0000] Attribute id: 4 value: _TZE200_fm5yck8a
2022-12-01 21:08:40.260 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:1:0x0000] Attribute id: 5 value: TS0601
2022-12-01 21:08:40.260 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:19:0x0006] Attribute id: 0 value: 0
2022-12-01 21:08:40.260 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:20:0x0006] Attribute id: 0 value: 0
2022-12-01 21:08:40.260 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:21:0x0006] Attribute id: 0 value: 0
2022-12-01 21:08:40.260 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:22:0x0006] Attribute id: 0 value: 0
2022-12-01 21:08:40.260 DEBUG (MainThread) [zigpy.appdb] [0x6a9b:1:0xef00] Attribute id: 61184 value: 1.0.0
2022-12-01 21:08:44.288 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x6A9B](_TZE200_fm5yck8a TS0601) restored as 'available', last seen: 0:00:25 ago, consider_unavailable_time: 7200 seconds
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x6A9B](TS0601): started initialization
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:ZDO](TS0601): 'async_initialize' stage succeeded
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0x0000]: initializing channel: from_cache: True
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0x0000]: finished channel initialization
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0xef00]: initializing channel: from_cache: True
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0xef00]: finished channel initialization
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0x0006]: initializing channel: from_cache: True
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0x0006]: finished channel initialization
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0x0019]: initializing channel: from_cache: True
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:1:0x0019]: finished channel initialization
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:2:0x0006]: initializing channel: from_cache: True
2022-12-01 21:08:44.290 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:2:0x0006]: finished channel initialization
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:19:0x0006]: initializing channel: from_cache: True
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:19:0x0006]: initializing cached channel attributes: ['start_up_on_off']
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:19:0x0006]: Reading attributes in chunks: ['start_up_on_off']
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:19:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[True]
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:19:0x0006]: Reading attributes in chunks: ['on_off']
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:19:0x0006]: finished channel initialization
2022-12-01 21:08:44.291 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6A9B:20:0x0006]: initializing channel: from_cache: True
20
```

</details>

javicalle, 2022-12-02:
😓 This is turning out to be more difficult than expected.

Next proposal:
<details open>
<summary>ts0601_switch.py</summary>

```python
class OnOffEventsCluster(EventableCluster, TuyaOnOff):
    """OnOffEventsCluster: fire events corresponding to press type."""

class AtlanticSwitchManufCluster(TuyaOnOffManufCluster):
    """On/Off Tuya cluster with extra device attributes."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            OnOffEventsCluster.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        2: DPToAttributeMapping(
            OnOffEventsCluster.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=2,
        ),
        19: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=19,
        ),
        20: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=20,
        ),
        21: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=21,
        ),
        22: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
            endpoint_id=22,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        19: "_dp_2_attr_update",
        20: "_dp_2_attr_update",
        21: "_dp_2_attr_update",
        22: "_dp_2_attr_update",
    }

class AtlanticSwitch(TuyaSwitch):
    """Tuya quadruple channel switch time on out cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fm5yck8a", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AtlanticSwitchManufCluster,
                ],
                OUTPUT_CLUSTERS: [OnOffEventsCluster, Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [OnOffEventsCluster],
            },
            19: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            20: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            22: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }
```

</details>

You need to add a new import statement in the beggining of file:
```python
from zhaquirks import EventableCluster
```

There is a new `OnOffEventsCluster` to fire the button events. Not sure if `device_automation_triggers` must be adjusted.
The new class replaces the `TuyaSmartRemoteOnOffCluster` class.

Save changes, delete `__pycache__` folder, restart HA and pair the device again.

nekdo4, 2022-12-03:
Hello,
thank you a lot, it seems I have bought something extra special  :D Well, scenes are still not functioning, log seems to be a bit longer. Below is a part describing one click on one and other switch. Unfortunately it does not run the automatization.

<details>
<summary>LOG</summary>

```
2022-12-03 20:25:39.029 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x469D, Relays=[])
2022-12-03 20:25:39.077 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x469D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=102, SecurityUse=<Bool.false: 0>, TimeStamp=85548, TSN=0, Data=b'\x09\x0A\x01\x00\x33\x01\x04\x00\x01\x00', MacSrcAddr=0x469D, MsgResultRadius=29)
2022-12-03 20:25:39.078 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x469D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\n\x01\x003\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=102, rssi=None)
2022-12-03 20:25:39.078 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Received ZCL frame: b'\t\n\x01\x003\x01\x04\x00\x01\x00'
2022-12-03 20:25:39.078 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=10, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-03 20:25:39.079 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=51, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-12-03 20:25:39.079 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Received command 0x01 (TSN 10): get_data(data=TuyaCommand(status=0, tsn=51, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-12-03 20:25:39.079 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] No datapoint handler for TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2022-12-03 20:25:39.079 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=10, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-03 20:25:39.080 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.UNSUPPORTED_ATTRIBUTE: 134>)
2022-12-03 20:25:39.080 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x469D), dst_ep=1, source_route=None, extended_timeout=False, tsn=10, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\n\x0b\x01\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-03 20:25:39.080 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x469D), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=10, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x0A\x0B\x01\x86')
2022-12-03 20:25:58.334 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x469D, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=94, SecurityUse=<Bool.false: 0>, TimeStamp=145730, TSN=0, Data=b'\x09\x0B\x01\x00\x33\x02\x04\x00\x01\x00', MacSrcAddr=0x469D, MsgResultRadius=29)
2022-12-03 20:25:58.334 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x469D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x0b\x01\x003\x02\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=94, rssi=None)
2022-12-03 20:25:58.335 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Received ZCL frame: b'\t\x0b\x01\x003\x02\x04\x00\x01\x00'
2022-12-03 20:25:58.335 DEBUG (MainThread) [zigpy.zcl] [0x469D:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=11, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...] =Serialized[b'\x18\x0b\x0b\x01\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-03 20:25:58.336 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x469D), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=11, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x0B\x0B\x01\x86')

```

</details>

javicalle, 2022-12-03:
The device still complains about the DP1 and DP2...

Can you double check that `__pycache__` folder has been deleted and the quirk is the one from https://github.com/zigpy/zha-device-handlers/issues/1785#issuecomment-1335450597?

If checks are fine, add the `OnOffEventsCluster` also in the `INPUT_CLUSTERS` part (in any order) and check again.

nekdo4, 2022-12-06:
Ahoj,
pokud jsem tedy neudelal chybu tak vse bylo spravne, po pridani OnOffEventsCluster do INPUT_CLUSTERS se chovani zmenilo za prve se objevili prepinace ktere na nic nereaguji, a eventy tez nepusti automatizace pro spusteni sceny, 
[image]
nicmene v prave casti kde jsou zaznamy zarizeni to zacalo neco psat viz obrazek. Pro jistotu pripojuji jak log tak i quirk. V quirku jsem nechal pouze tento vypinac a vsechny importy co tam byly celou dobu, mozna je chyba v tom

<details>
<summary>ts0601_switch.py
[image]
</summary>

```
"""Tuya DP based switches."""
from typing import Dict

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    CLUSTER_ID,
    COMMAND,
    COMMAND_MOVE,
    COMMAND_MOVE_SATURATION,
    COMMAND_OFF,
    COMMAND_ON,
    COMMAND_STEP,
    COMMAND_STOP,
    COMMAND_STOP_MOVE_STEP,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PARAMS,
    PROFILE_ID,
    RIGHT,
    ROTATED,
    ROTATED_FAST,
    ROTATED_SLOW,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)

from zhaquirks import EventableCluster

from zhaquirks.tuya import (
    TuyaSwitch,
    TuyaSmartRemoteOnOffCluster,
    TuyaZBE000Cluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaDPType,
    TuyaMCUCluster,
[... 510 lines cut ...]
2022-12-06 22:57:36.257 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x52CD), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=15, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x0F\x0B\x11\x00')
2022-12-06 22:58:13.579 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x52CD, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=115, SecurityUse=<Bool.false: 0>, TimeStamp=577408, TSN=0, Data=b'\x09\x10\x01\x00\x0E\x01\x04\x00\x01\x00', MacSrcAddr=0x52CD, MsgResultRadius=29)
2022-12-06 22:58:13.580 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x52CD), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x10\x01\x00\x0e\x01\x04\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=115, rssi=None)
2022-12-06 22:58:13.581 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] Received ZCL frame: b'\t\x10\x01\x00\x0e\x01\x04\x00\x01\x00'
2022-12-06 22:58:13.581 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=16, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-06 22:58:13.582 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] Decoded ZCL frame: AtlanticSwitchManufCluster:get_data(data=TuyaCommand(status=0, tsn=14, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-12-06 22:58:13.583 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] Received command 0x01 (TSN 16): get_data(data=TuyaCommand(status=0, tsn=14, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
2022-12-06 22:58:13.585 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=16, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-06 22:58:13.585 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] Sending reply: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
2022-12-06 22:58:13.586 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x52CD), dst_ep=1, source_route=None, extended_timeout=False, tsn=16, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\x10\x0b\x01\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-06 22:58:13.586 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x52CD), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=16, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x10\x0B\x01\x00')
[... cut ...] =Serialized[b'\x18\x11\x0b\x01\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-12-06 22:58:16.275 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x52CD), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=17, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x11\x0B\x01\x00')

```

</details>

javicalle, 2022-12-07:
No the imports would not be a problem.

And now we have new neighbours in the block...
```
2022-12-06 22:57:14.295 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] No datapoint handler for TuyaDatapointData(dp=25, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
2022-12-06 22:57:14.335 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] No datapoint handler for TuyaDatapointData(dp=26, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
2022-12-06 22:57:14.376 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] No datapoint handler for TuyaDatapointData(dp=27, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
2022-12-06 22:57:14.418 DEBUG (MainThread) [zigpy.zcl] [0x52CD:1:0xef00] No datapoint handler for TuyaDatapointData(dp=28, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
```

DP25-DP28 with value=0
They can be a dim level, some kind of metering or nothing at all. 
But they came in a sequence after all the DP19-DP22 buttons press. Any idea what they can be?

It seems that quirk is reading the scene buttons. Aren't they updating in HA if you press the phisycal device buttons? Can you attach any logs from the scene physical buttons?

Can you also check in the developers view if the device trigger any event, listening to the `zha_event`?
The `OnOffEventsCluster` would at least trigger the events.

The truth is that I'm a bit lost here. I would try to change the device type for DP1 and DP2 as:
```python
            1: {
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    AtlanticSwitchManufCluster,
                ],
                OUTPUT_CLUSTERS: [OnOffEventsCluster, Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [OnOffEventsCluster],
            },
```

The logs from the physical device would be helpfull here.

nekdo4, 2022-12-13:
Hello,
so DP25-DP28  does not know what it is, it sometimes appears but not every time.

In developer tools - events -  during zha_event and during * there is no reaction to scenes switch, and the light switch during * does this:

<details>
<summary>Events *</summary>

```

event_type: state_changed
data:
  entity_id: light.tze200_fm5yck8a_ts0601_light_2
  old_state:
    entity_id: light.tze200_fm5yck8a_ts0601_light_2
    state: "on"
    attributes:
      supported_color_modes:
        - onoff
      color_mode: onoff
      off_with_transition: false
      off_brightness: null
      friendly_name: _TZE200_fm5yck8a TS0601 Light
      supported_features: 0
    last_changed: "2022-12-13T20:15:25.870934+00:00"
    last_updated: "2022-12-13T20:15:25.870934+00:00"
    context:
      id: 01GM6J1DNE6GT9ZVBZGA3ZMCPM
      parent_id: null
      user_id: null
  new_state:
    entity_id: light.tze200_fm5yck8a_ts0601_light_2
    state: "off"
    attributes:
      supported_color_modes:
        - onoff
      off_with_transition: false
      off_brightness: null
      friendly_name: _TZE200_fm5yck8a TS0601 Light
      supported_features: 0
    last_changed: "2022-12-13T20:15:26.597627+00:00"
    last_updated: "2022-12-13T20:15:26.597627+00:00"
    context:
      id: 01GM6J1EC52DZ9B9SJZRCAEH4W
      parent_id: null
      user_id: null
origin: LOCAL
time_fired: "2022-12-13T20:15:26.597627+00:00"
context:
  id: 01GM6J1EC52DZ9B9SJZRCAEH4W
  parent_id: null
  user_id: null
```

</details>

Even on the picture below you can see only reaction to light.

I am sending the complete log below.

[log.txt](https://github.com/zigpy/zha-device-handlers/files/10221792/log.txt)

[image]

closed 2023-07-10 by github-actions[bot]

SolusCado, 2024-12-16:
I don't know if anyone on this thread is still struggling to get this to work, but I have a very similar device:
https://www.aliexpress.com/item/1005004888723698.html?spm=a2g0o.order_list.order_list_main.23.59491802DVpbxu

Which was having the exact same issues outlined in this thread. I finally have everything working perfectly for my device, so I thought I'd share my quirk here:

`"""Tuya 4+2 Button Switch+Scene Selector"""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, OnOff, Ota, Time, Scenes, Identify

from zhaquirks.const import (
    BUTTON_1,
    BUTTON_2,
    COMMAND,
    ENDPOINT_ID,
    SHORT_PRESS,
)

from zhaquirks.tuya import (
    TuyaSwitch,
    TuyaSmartRemoteOnOffCluster,
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)

class Coswall(TuyaSwitch):
    """Custom quirk for _TZ3002_sal078g8 TS0726."""

    signature = {
        "models_info": [("_TZ3002_sal078g8", "TS0726")],
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0004,
                "input_clusters": [
                    Basic.cluster_id,  # Basic
                    Identify.cluster_id,  # Identify
                    Groups.cluster_id,  # Groups
                    Scenes.cluster_id,  # Scenes
                    OnOff.cluster_id,  # OnOff
                    TuyaZBE000Cluster.cluster_id,  # Manufacturer Specific
                    TuyaZBExternalSwitchTypeCluster.cluster_id,  # Manufacturer Specific
                ],
                "output_clusters": [
                    Time.cluster_id,  # Time
                    Ota.cluster_id,  # OTA
                ],
            },
            2: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0004,
                "input_clusters": [
                    Groups.cluster_id,  # Groups
                    Scenes.cluster_id,  # Scenes
                    OnOff.cluster_id,  # OnOff
                    TuyaZBExternalSwitchTypeCluster.cluster_id,  # Manufacturer Specific
                ],
                "output_clusters": [],
            },
            3: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0004,
                "input_clusters": [
                    Groups.cluster_id,  # Groups
                    Scenes.cluster_id,  # Scenes
                    OnOff.cluster_id,  # OnOff
                    TuyaZBExternalSwitchTypeCluster.cluster_id,  # Manufacturer Specific
                ],
                "output_clusters": [],
            },
[... 104 lines cut ...]
                    TuyaSmartRemoteOnOffCluster,
                ],
                "output_clusters": [],
            },
            242: {
                "input_clusters": [],
                "output_clusters": [
                    GreenPowerProxy,
                ],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 5, COMMAND: SHORT_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 6, COMMAND: SHORT_PRESS},
    }`

Some key points: you may need to do some testing to figure out which ENDPOINT_IDs are pointing to your Scene Selector buttons, and which ones are pointing to your Switches. In my case, it is 5 and 6. It is the  TuyaSmartRemoteOnOffCluster tagged onto the end of the input_clusters that ultimately makes the device_automation_triggers work, so you need both to get that Scene Selector reporting ZHA events. Your automations will then look like this:

[image]

(My Switches are labeled Top Right, Top Middle, Bottom Right, and Bottom Middle - and my Scene Selectors are the "First Button Pressed" and "Second Button Pressed" triggers. And don't let those labels confuse you; I have my switch installed upside down - your Scene Selector Buttons will probably be on the right side of the switch.)

One final note: The device_type values in the quirk signature don't seem to correspond to the correct zha.DeviceType enum. When I tried using zha.DeviceType.ON_OFF_SWITCH (0x0000), the signature doesn't match the device - but the value that DOES match (0x0004) appears to correspond to zha.DeviceType.ON_OFF_LIGHT - but applying that seems to end up with a signature mismatch as well... So I'm just hardcoding the values returned by the default signature of the device.