zigpy/zha-device-handlers #1786: Tesla Smart Thermostatic Valve TV100 [Device Support Request]

dohnalekp, 2022-09-25:
Hi, i buy two Tesla Smart Thermostatic Valve TV100. Adding to home assistant is OK, but no entities show.
A have the same working device. 

**Working device zigbee signature:**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000d",
        "0x000f",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0006",
        "0x000d"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0007",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZE200_husqqvux",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.ZonnsmartTV01_ZG"
}

**New device zigbee signature:**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_lnbfnyxd",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

Old model name: TSL-TVR-TV01ZG
New model name: TSL-TVR-TV01ZG

Old model working fine, but when I adding new model to HA, entities not show.

Can you help me?

javicalle, 2022-09-26:
The new manufacturer code needs to be added to the current quirk.

Enable the local quirk folder and copy inside the current `ts0601_trv.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py

Then update the `ZonnsmartTV01_ZG` class like this (just the `MODELS_INFO` part):
```python

class ZonnsmartTV01_ZG(TuyaThermostat):
    """ZONNSMART TV01-ZG Thermostatic radiator valve."""

    def __init__(self, *args, **kwargs):
        """Init device."""
        self.boost_bus = Bus()
        self.child_lock_bus = Bus()
        self.online_mode_bus = Bus()
        self.temperature_calibration_bus = Bus()
        self.window_detection_bus = Bus()
        self.window_temperature_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=0 input_clusters=[0, 4, 5, 61184]
        #  output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_e9ba97vf", "TS0601"),
            ("_TZE200_husqqvux", "TS0601"),
            ("_TZE200_kly8gjlz", "TS0601"),
            ("_TZE200_hue3yfsn", "TS0601"),  # TV02-ZG
            ("_TZE200_lnbfnyxd", "TS0601"),
        ],
        ENDPOINTS: {
            .../...
```

Repair the device after enable the local quirk.

dohnalekp, 2022-09-27:
Than you @javicalle for help. Its working fine now.

PR #1792 (2022-09-27): Add `_TZE200_lnbfnyxd` TRV
Adds the `manufacturer_code` for the ' Tesla Smart Thermostatic Valve TV100'
I've also sorted the list alphabetically.

Fixes: #1786
closed 2022-09-27 by dmulcahey