zigpy/zha-device-handlers #1787: ZHA Blitzwolf SHP 15 energy monitor outlet, on/off switch wont register in Home Assistant

yfands, 2022-09-25:
Hello,

Purchaised two Blitzwolf SHP 15, all seems to be ok.
But if I turn the device on/off (on the device itself) Home Assistant wont register it. Other way around, switching in Home assistant, does turn the device on/off.

The Blitzwolf SHP 15 shows up as:
model": “TS011F
device_type”: "0x010a

```
“node_descriptor”: “NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)”,
“endpoints”: {
“1”: {
“profile_id”: 260,
“device_type”: “0x010a”,
“in_clusters”: [
“0x0000”,
“0x0003”,
“0x0004”,
“0x0005”,
“0x0006”,
“0x0702”,
“0x0b04”,
“0xe000”,
“0xe001”
],
“out_clusters”: [
“0x000a”,
“0x0019”
]
}
},
“manufacturer”: “_TZ3000_u5u4cakc”,
“model”: “TS011F”,
“class”: “zhaquirks.tuya.ts011f_plug.Plug”
}
```

all my other devices Blitzwolf SHP 13 are working correct and show up as model: TS0121

Can somebody verify my findings? (and a possible solution) thanks in advance.
Regards Frank

javicalle, 2022-09-26:
It seems to me a pairing issue. Have you tried to repair the device?

It the problem persist, enable the debug logging and attach the relevant info about your device:
* https://www.home-assistant.io/integrations/zha/#debug-logging

yfands, 2022-09-26:
Hello Javicalle,

Thanks for the reply,
As you requested, I paired both devices again (after removing them) but both came up again as model": TS011F
Home Assistant not reacting on the physical On/Off button.

[image]

Device:
Blitzwolf SHP 15, outlet energy monitor   (equal to Blitzwolf SHP 13)

Signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_u5u4cakc",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```
Diagnostics:
[TZ3000_u5u4cakc TS011F.json.txt](https://github.com/zigpy/zha-device-handlers/files/9649521/TZ3000_u5u4cakc.TS011F.json.txt)

Debug:
from the moment the device was added.
[home-assistant - kopie.log](https://github.com/zigpy/zha-device-handlers/files/9649532/home-assistant.-.kopie.log)

Thanks In Advance..
Regards Frank

javicalle, 2022-09-26:
We will need the logs from the on/off action. From HA and from the physical device.

javicalle, 2022-09-26:
Look at the logs for any info from your device (check the Nwk value for your device in the "Zigbee info" option).

yfands, 2022-09-26:
the device Nwk = 0x72b3

I created two files, the upper debugged the action on -> off  3 times from HA point of few
the latter from the device it self.

[home-assistant - switched from HA.log](https://github.com/zigpy/zha-device-handlers/files/9649711/home-assistant.-.switched.from.HA.log)

[home-assistant - switched from device.log](https://github.com/zigpy/zha-device-handlers/files/9649718/home-assistant.-.switched.from.device.log)

and again thank you for looking into this...

yfands, 2022-10-08:
Hello, anything you could find in the log files that gives you any idea why pressing the physical on/off switch wont register in HA ?

Charisketi, 2022-10-12:
I have 5 of these switches. I can confirm this specific issue. The switch stays open in the home assistant when the physical button is pressed to close it. As a result of this, opens again after a while by itself.

Charisketi, 2022-10-12:
> Device isn't reporting the switch status change.
> The same has been reported in Z2M:
>
>    - Koenkk/zigbee2mqtt#9493 (comment)
>
> The device would update its status after a while when HA polled the device
> status (not sure which time interval for switches). @Charisketi
> <https://github.com/Charisketi> is this what are you seing in your
> devices (that HA status is updated after a time)? or the device is turning
> on after a while?

As i mention earlier, device state will not change with physical control.
If for example the plug has state "on" on home assistant and switch it off
from plug physical button, the switch state on home assistant doesn't
change even if the switch is physically turned off. The result of this
situation is that after a while the switch turn on by itself.

javicalle, 2022-10-12:
> If for example the plug has state "on" on home assistant and switch it off from plug physical button, the switch state on home assistant doesn't change even if the switch is physically turned off.

That's the know behavior already reported and seems to be the device faulty.

> The result of this situation is that after a while the switch turn on by itself.

That part don't make sense to me. I don't know any mechanism that would turn on the plug just because have an `on` state in HA. Are you sure that don't have any automation or that the plug is  not used in any other integration like a thermostat or similar?
If you can't find a reason for the behavior, can you enable the debug logs and attach the relevant part?

Charisketi, 2022-10-12:
I just checked again. You have right, it was an automation that turn the
switch on again. I'm sorry.
As for the device fault, is there any solution to fix this issue;

javicalle, 2022-10-12:
It seems that diferents firmware versions of the device have diferents behaviors.
If I understood it right, maybe there is a solution for your devices.

Enable the local quirk in your instalation.
Copy in your local quirk folder the file `ts011f_plug.py`:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py
Edit the file and change the `Plug` class definition as (just replace the class inheritance):
```python
class Plug(EnchantedDevice):
    """Tuya plug with restore power state support."""
```
Restart HA and pair your device again. Keep the pairing logs and attach it if the problem persists.

yfands, 2022-10-13:
> It seems that diferents firmware versions of the device have diferents behaviors. If I understood it right, maybe there is a solution for your devices.
>
> Enable the local quirk in your instalation. [...]

Doesnt work for me, same issue.. regards Frank

javicalle, 2022-10-14:
I have reviewed a few of the Z2M issues about the that device and everything points me to an `EnchantedDevice` issue.

Have you tried to repair the device? I would try a couple of times.

yfands, 2022-10-16:
> I have reviewed a few of the Z2M issues about the that device and everything points me to an `EnchantedDevice` issue.
> 
> Have you tried to repair the device? I would try a couple of times.

Hello Javicalle, "Try a couple of times" was the right advice..  3 times I had to remove and re-pair my "test" device. finally it worked, my second Blitzwolf AHP 15 was no problem on first try remove/re-pair did the trick, All is in working order now. 

oh, indeed you have to edit the file ts011f_plug.py and change the class

Thanks a million.. Regards Frank

PR #1833 (2022-10-17): Make `TS011F` plugs inherit `EnchantedDevice`
There are reports that some Blitzwolf SHP 15 plugs (depending on its version) don't report the On/Off update from the physical device. The `herdsman-converters` lib have 3 implementations for the `TS011F` plugs, 2 of them 'implements' the EnchantedDevice approach:
* https://github.com/Koenkk/zigbee-herdsman-converters/blob/96563772e92eb459bf52c10e241c3a725968bcab/devices/tuya.js#L2185

My assumption is that the use of the `EnchantedDevice` cannot be harmful. Also I have not located implementations that could be affected if a device starts to behave [... cut]
closed 2022-10-17 by dmulcahey