zigpy/zha-device-handlers #1788: [Device Support Request] _TZE200_2cs6g9i7 - Brennenstuhl Zigbee Connect 01

ich777, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
I bought a [Brennenstuhl Heizkörperthermostat Zigbee Connect 01](https://www.conrad.at/de/p/brennenstuhl-heizkoerperthermostat-zigbee-connect-01-2523459.html) Valve.
This valve is able to be added into ZHA fine, but no quirk is associated with it. It only lists `Lqi` and `Rssi` but nothing else, the valve looks like [Alecto Thermostatic Radiator Valve](https://zigbee.blakadder.com/Alecto_SMART-HEAT10.html) and [essentials Smart Home Heizkörperthermostat Premium](https://zigbee.blakadder.com/essentials_120112.html) and others.

**Describe the solution you'd like**
Add a quirk to make the device working in HA

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_2cs6g9i7",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2022.9.6",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.9.2",
    "docker": false,
    "arch": "x86_64",
    "timezone": "Europe/Vienna",
    "os_name": "Linux",
    "os_version": "5.19.9-Unraid",
    "run_as_root": false
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.33.1",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.79",
      "zigpy-deconz==0.18.1",
      "zigpy==0.50.3",
      "zigpy-xbee==0.15.0",
      "zigpy-zigate==0.9.2",
      "zigpy-znp==0.8.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      }
    ],
    "dependencies": [
      "file_upload"
    ],
    "after_dependencies": [
      "onboarding",
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "config": {
      "zigpy_config": {
        "ota": {
          "ikea_provider": true,
          "ledvance_provider": true,
          "salus_provider": true,
          "otau_directory": "/home/homeassistant/.homeassistant/zigpy_ota"
        },
        "database_path": "/home/homeassistant/.homeassistant/zigbee.db",
        "device": {
          "path": "socket://10.0.0.16:6638",
          "baudrate": 115200,
          "flow_control": null
        }
      },
      "device_config": {},
      "enable_quirks": true
    },
    "config_entry": {
      "entry_id": "af8d1df79d806a286af68b93d3d3abab",
      "version": 3,
      "domain": "zha",
      "title": "socket://10.0.0.16:6638",
      "data": {
        "device": {
          "path": "socket://10.0.0.16:6638",
          "baudrate": 115200,
          "flow_control": null
        },
        "radio_type": "znp"
      },
      "options": {
        "custom_configuration": {
          "zha_options": {
            "enable_identify_on_join": true,
            "consider_unavailable_battery": 21600,
            "consider_unavailable_mains": 7200
          },
          "zha_alarm_options": {
            "alarm_arm_requires_code": false,
            "alarm_failed_tries": 3,
            "alarm_master_code": "**REDACTED**"
          }
        }
      },
      "pref_disable_new_entities": false,
      "pref_disable_polling": false,
      "source": "user",
      "unique_id": null,
      "disabled_by": null
    },
    "application_state": {
      "node_info": {
        "nwk": 0,
        "ieee": "**REDACTED**",
        "logical_type": 0
      },
      "network_info": {
        "extended_pan_id": "**REDACTED**",
        "pan_id": 13143,
        "nwk_update_id": 0,
        "nwk_manager_id": 0,
        "channel": 15,
        "channel_mask": 32768,
        "security_level": 5,
        "network_key": "**REDACTED**",
        "tc_link_key": {
          "key": [
            90,
            105,
            103,
            66,
            101,
            101,
            65,
            108,
            108,
            105,
            97,
            110,
            99,
            101,
            48,
            57
          ],
          "tx_counter": 0,
          "rx_counter": 0,
          "seq": 0,
          "partner_ieee": "**REDACTED**"
        },
        "key_table": [],
        "children": [],
        "nwk_addresses": {},
        "stack_specific": {
          "zstack": {
            "tclk_seed": "d9c338ae43eabab317501603fc28a684"
          }
        },
        "metadata": {
          "zstack": {
            "TransportRev": 2,
            "ProductId": 1,
            "MajorRel": 2,
            "MinorRel": 7,
            "MaintRel": 1,
            "CodeRevision": 20220219,
            "BootloaderBuildType": 0,
            "BootloaderRevision": null
          }
        },
        "source": "zigpy-znp@0.8.2"
      },
      "counters": {},
      "broadcast_counters": {},
      "device_counters": {},
      "group_counters": {}
    },
    "versions": {
      "bellows": "0.33.1",
      "zigpy": "0.50.3",
      "zigpy_deconz": "0.18.1",
      "zigpy_xbee": "0.15.0",
      "zigpy_znp": "0.8.2",
      "zigpy_zigate": "0.9.2",
      "zhaquirks": "0.0.79"
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
none
```

</details>

**Additional context**
Maybe someone could help with that...

Thanks!

MattWestb, 2022-09-26:
Servos. 
First configure local quirks in (Z)HA.
For getting  "basic" functionality add your device ID at one new line after the other in this code part (no extra taps and spaces):
https://github.com/zigpy/zha-device-handlers/blob/0ad9fa2f6fd2cd53c9c010553d46dbc8092c1a48/zhaquirks/tuya/ts0601_trv.py#L1421-L1430 in one copy if the quirk linked and saving it in the new configured quirk folder and restarting HA (done need deleting the device it shall working OK).

If that is working OK we is having one quirk not being merged in ZHA that have more functionality and deleting not working functions but is needing more work and we is testing it later.

ich777, 2022-09-26:
@MattWestb Hallo.

Thank you! Working perfectly fine:
[image]

Do I have to make a PR for that?

Danke nochmal!

MattWestb, 2022-09-26:
Yes pleas its not one must but good for all users getting there device working in ZHA!!

If you like getting more functions you can testing this quirk (i have adding your device ID in it) but deleting the "old" one so not having 2 different with the same device ID and restart HA.
[ts0601_trv_siterwell.zip](https://github.com/zigpy/zha-device-handlers/files/9649067/ts0601_trv_siterwell.zip)

For getting the presence working OK its needed patching HA docker container file with this adding the line 710 and to the end to the `climate.py` in the HAs ZHA folder (its tricky) (its one old file so i think its not working copy it then its not compatible with the last changes so copy the code to your current file but do one backup of the original if not working OK).
[climate.zip](https://github.com/zigpy/zha-device-handlers/files/9649095/climate.zip)

ich777, 2022-09-26:
@MattWestb no, this doesn't work, looking like this after I've changed climate.yaml and the other file (I think you've missed to put in the device name in both of the files or copied the wrong ones but I've fixed that):
[image]

I will create a PR for the device that it will at least work.

closed 2022-09-26 by ich777

MattWestb, 2022-09-27:
Thanks for the PR !!

Its very likely i was to tired bu you have getting the climate.py working but the quirk is not being loaded OK. Look in the log if its complaining or making one double check that  the device ID is added in the right place with the other TRVs of the same type.

cemizm, 2022-11-21:
> Yes pleas its not one must but good for all users getting there device working in ZHA!!
>
> If you like getting more functions you can testing this quirk (i have adding your device ID in it) but [...]

Hi @MattWestb, i added the siterwell custom quirk and seems to add the entities but they are not really working:

[image]

Chaning the temperature calibration error results in the error: 

Logger: homeassistant.components.websocket_api.http.connection
Source: components/zha/core/channels/general.py:111
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 19:55:28 (10 occurrences)
Last logged: 20:04:44

```
[1559320872] 'present_value'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 109, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/usr/src/homeassistant/homeassistant/components/zha/number.py", line 361, in async_set_native_value
    if await self._analog_output_channel.async_set_present_value(num_value):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 111, in async_set_present_value
    res = await self.cluster.write_attributes({"present_value": value})
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 96, in write_attributes
    attrid = self.attributes[attrid].id
KeyError: 'present_value'
```

I have also the Saswell custom quirks enabled for my other thermostats. could they interfer with each other?

MattWestb, 2022-11-21:
Different quirks is not fighting in the code only one can being loaded for one device.
In the log its complaining about presets, have yoi isntalling the presets in ZHA with the [climate.zip](https://github.com/zigpy/zha-device-handlers/files/9649095/climate.zip) ?

cemizm, 2022-11-21:
No the only other warning i get is a
```
2022-11-21 20:15:18.165 WARNING (MainThread) [zigpy.util] Error calling listener <bound method SiterwellThermostat.hass_climate_state_change of <Siterwell.SiterwellThermostat object at 0x59c01d90>> with args (514, 50): TypeError("int() argument must be a string, a bytes-like object or a real number, not 'NoneType'")
```

i also tested the switches witch seems control the child lock, window detection and valve initial calibration.. i did not install the climate.zip since my home assistant instance is running inside a docker container... is it possible to use without the presets?

MattWestb, 2022-11-21:
I was using the quirk and the climate.py but im to lazy installing in the container after every HA upgrade and its working OK.
I have not testing changing temperature calibration without patched presets.

You can easy copy to the container but must have access to the OS CLI.
Also i think its bets adding the Siterwell presets in one current clinate.py is its changes made that is not in the one i have posting.

cemizm, 2022-11-21:
yeah it is easy to copy but only available until i update the container :D maybe  i will give it a try later, because i would really love to calibrate the temperature with an external temperature sensor automatically... do you have / are there any plans to integrate the presets into an upcoming version of HA?

MattWestb, 2022-11-21:
We have making PR for extended functions for all tuya TRVs but we was not able doing the test code and was not getting enough help doing it so its not being merged :-((

The temperature calibrating is adjusting the heat / idle for the ZHA graph so its looking nice then the device dont sending valve state. I have all my on -0.8°C for getting realistic.

If you like doing calibration from one external temperature sensor use it and adding the offset (+ / - ) for patching the setpoint temperature in the climate component then its the only way adjusting how mush the TRV is heating (its using PID algorithms).
Switching it on and off is bad for PID devices then they is loosing the internal calibration.