zigpy/zha-device-handlers #1790: [Device Support Request] - TS0601 - _TZE200_wbhaespm

flaktivated, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
When I attempt to Pair the Tuya Zigbee Metering Circuit Breaker to Home Assistant via ZHA it is unable to identify the device and cannot use its features.

**Describe the solution you'd like**
I would appreciate if I could get some assistance to get the new DIN 2-Pole Circuit Breaker with Power Meter onboarded to be supported in ZHA.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_wbhaespm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2022.9.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "4.4.59+",
    "supervisor": "2022.09.1",
    "host_os": null,
    "docker_version": "20.10.3",
    "chassis": null,
    "run_as_root": true
  },
  "custom_components": {
[... 427 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            }
        }
    }
```

flaktivated, 2022-09-26:
<details>
<summary>Additional logs</summary>

```
New device 0x9f05 (a4:c1:38:7f:77:6b:4b:c7) joined the network
[0x9f05] Scheduling initialization
Tries remaining: 3
[0x9f05] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9f05] Extending timeout for 0x07 request
Device 0x9f05 (a4:c1:38:7f:77:6b:4b:c7) joined the network
[0x9f05] Scheduling initialization
[0x9f05] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\x05\x9f\xc7Kkw\x7f8\xc1\xa4\x8e'
[0x9f05:zdo] ZDO request ZDOCmd.Device_annce: [0x9F05, a4:c1:38:7f:77:6b:4b:c7, 142]
Tries remaining: 3
[0x9f05] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9f05] Extending timeout for 0x09 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\t\x00\x05\x9f\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x9f05] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x9f05] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x0b\x00\x05\x9f\x02\x01\xf2'
[0x9f05] Discovered endpoints: [1, 242]
[0x9f05] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x9f05:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\r\x00\x05\x9f\x14\x01\x04\x01Q\x00\x01\x04\x04\x00\x05\x00\x00\xef\x00\x00\x02\x19\x00\n\x00'
[0x9f05:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
Unknown cluster 0xEF00
[0x9f05:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x0f\x00\x05\x9f\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x9f05:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x9F05:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=17, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x9F05:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x9F05:1:0x0000] Received ZCL frame: b'\x18\x11\x01\x04\x00\x00B\x10_TZE200_wbhaespm\x05\x00\x00B\x06TS0601'
[0x9F05:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=17, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x9F05:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_wbhaespm')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0x9f05] Read model 'TS0601' and manufacturer '_TZE200_wbhaespm' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x9f05] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_wbhaespm' nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_wbhaespm' nwk=0x9F05 ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=True>
Checking quirks for _TZE200_wbhaespm TS0601 (a4:c1:38:7f:77:6b:4b:c7)
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
Fail because endpoint list mismatch: {1, 2, 242} {1, 242}
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 242}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
[... cut ...]
[0x9F05:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=158, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x9F05:1:0xef00] Unknown cluster command 2 b'\x00f\x13\x03\x00\n0000000001'
[0x9F05:1:0xef00] Received command 0x02 (TSN 158): b'\x00f\x13\x03\x00\n0000000001'
[0x9F05:1:0xef00] No explicit handler for cluster command 0x02: b'\x00f\x13\x03\x00\n0000000001'
```

</details>

javicalle, 2022-09-26:
The quirk looks good to me but for some reason is not loading.
I would check the usual suspects:
* local quirk:
  * https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274
* delete the local cache, any `__pycache__` folder in your local quirk
* repair the device. If not works, remove from HA and pair it again

flaktivated, 2022-09-26:
I installed it previously but was getting errors in ZHA, so assumed that the quirk was wrong. Below are my steps to get the quirk loaded in HA:

1. Removed device from HA
2. Deleted __pycache__ folder
3. Placed the above code in the zha_quirks folder I created and mapped in my container
4. Restarted HA
5. Confirmed ZHA initialised without errors
6. Added Device via ZHA UI
[image]
7. Paired the device
[image]
8. When attempting to interact get error messages:
[image]
[image]
[image]
[image]

<details>
<summary>Additional logs - Zigbee Radio</summary>

```
New device 0x69ac (a4:c1:38:7f:77:6b:4b:c7) joined the network
[0x69ac] Scheduling initialization
Tries remaining: 3
[0x69ac] Requesting 'Node Descriptor'
Tries remaining: 2
[0x69ac] Extending timeout for 0x07 request
Device 0x69ac (a4:c1:38:7f:77:6b:4b:c7) joined the network
[0x69ac] Scheduling initialization
[0x69ac] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xaci\xc7Kkw\x7f8\xc1\xa4\x8e'
[0x69ac:zdo] ZDO request ZDOCmd.Device_annce: [0x69AC, a4:c1:38:7f:77:6b:4b:c7, 142]
Tries remaining: 3
[0x69ac] Requesting 'Node Descriptor'
Tries remaining: 2
[0x69ac] Extending timeout for 0x09 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\t\x00\xaci\x01@\x8eA\x11BB\x00\x00*B\x00\x00'
[0x69ac] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x69ac] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x0b\x00\xaci\x02\x01\xf2'
[0x69ac] Discovered endpoints: [1, 242]
[0x69ac] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=242 in=[] out=[] status=<Status.NEW: 0>>]
[0x69ac:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\r\x00\xaci\x14\x01\x04\x01Q\x00\x01\x04\x04\x00\x05\x00\x00\xef\x00\x00\x02\x19\x00\n\x00'
[0x69ac:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
Unknown cluster 0xEF00
[0x69ac:242] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x0f\x00\xaci\n\xf2\xe0\xa1a\x00\x00\x00\x01!\x00'
[0x69ac:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
[0x69AC:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=17, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x69AC:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x69AC:1:0x0000] Received ZCL frame: b'\x18\x11\x01\x04\x00\x00B\x10_TZE200_wbhaespm\x05\x00\x00B\x06TS0601'
[0x69AC:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=17, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x69AC:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_wbhaespm')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0x69ac] Read model 'TS0601' and manufacturer '_TZE200_wbhaespm' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x69ac] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_wbhaespm' nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_wbhaespm' nwk=0x69AC ieee=a4:c1:38:7f:77:6b:4b:c7 is_initialized=True>
[... cut ...]
[0x69AC:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=223, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x69AC:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
```

</details>

flaktivated, 2022-10-01:
After fiddling around with log levels and manually toggling the switch of the device have identified that the following messages are trigged with each time the switch is toggled:

`2022-10-01 21:40:55.180 DEBUG (MainThread) [zigpy.zcl] [0x256A:1:0xef00] Received ZCL frame: b'\t\x95\x02\x00B\x10\x01\x00\x01\x01'
2022-10-01 21:40:55.180 DEBUG (MainThread) [zigpy.zcl] [0x256A:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=149, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-01 21:40:55.181 DEBUG (MainThread) [zigpy.zcl] [0x256A:1:0xef00] Decoded ZCL frame: TuyaManufClusterDinPower:set_data_response(param=Command(status=0, tsn=66, command_id=272, function=0, data=[1, 1]))
2022-10-01 21:40:55.181 DEBUG (MainThread) [zigpy.zcl] [0x256A:1:0xef00] Received command 0x02 (TSN 149): set_data_response(param=Command(status=0, tsn=66, command_id=272, function=0, data=[1, 1]))
2022-10-01 21:40:55.181 DEBUG (MainThread) [zigpy.zcl] [0x256A:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=149, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-01 21:40:55.181 DEBUG (MainThread) [zigpy.zcl] [0x256A:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2022-10-01 21:40:55.181 DEBUG (MainThread) [zhaquirks.tuya] [0x256a:1:0xef00] Received value [1] for attribute 0x0110 (command 0x0002)`

javicalle, 2022-10-01:
Another test that you can try is the `EnchantedDevice`.

You have to modify the quirk as:

1. add the new class
```python
.../...
from zhaquirks.tuya.mcu import EnchantedDevice
```

add to the device:
```python
.../...

class TuyaPowerMeter(EnchantedDevice, TuyaSwitch):
    """Tuya power meter device."""

```

Delete `__pycache__` folder, restart HA and repair device.

flaktivated, 2022-10-02:
To do: 
- [ ] Check and attach errors from logs during paring for record keeping
- [ ] Filter logs collected for distinct list of messages received from the device.
- [ ] Try to use the EnchantedDevice import

javicalle, 2022-10-02:
Few comments from a first quick review:

I don't usually give importance to `FOREIGN KEY constraint failed` errors. I think that are related with the register update (that already have values from previous pairing).

All these `command_id` are reports from device that must update some device attributes. I'll take a closer look.

I can see the device 'cast spell', so it has been aplied but no seems to fix the device.

That `UNSUP_MANUF_CLUSTER_COMMAND` can explain the `OnOff` malfunction, but not all the other metering problems. Can you check if the errors are related to the `OnOff` action from HA?
If yes, try importing the `TuyaOnOffNM` and use it in the replacement part:
```python

from zhaquirks.tuya.mcu import TuyaOnOffNM

.../...

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },

```

javicalle, 2022-10-02:
Values from diferents commands:

| command_id  | Data type  | value |
| ------------- | ------------- | ------------- |
| 513  | value | 18 |
| 272  | boolean | 0 |
| 525 | value | 0 |
| 1289 | bitmap | ¿? |
| 787 | string | 0000000001 |

These commands must match with the cluster attributes:
``` python
    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }
```

So, my guess would be that the boolean will be the switch:
```python
TUYA_DIN_SWITCH_ATTR = 0x0110  #272
```

But I have no idea what the other reports mean (especially the string and the bitmap one).

Maybe with a known power load and looking at the report values can be guessed.
Bu for sure, the reports command_id (in decimal value) must match with all these `TUYA_WHATEVER_ATTR` attribute_id from the cluster (in hexadecimal value).

flaktivated, 2022-10-02:
Thanks for the guidance above. Will focus on just getting the OnOff toggle to work first. Have made some progress on with the advice above and feel like we're close.

Your guess ```TUYA_DIN_SWITCH_ATTR = 0x0110``` above is correct I now can see the status of the switch update immediately after manually toggled on the device. I just cannot trigger the switch from HA.

Looking at the debug logs below, is it possible that the toggle switch function is on another command?
```
:38:56.980 DEBUG (MainThread) [zigpy.zcl] [0x1216:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=58, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2022-10-03 10:38:56.980 DEBUG (MainThread) [zigpy.zcl] [0x1216:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=257, function=0, data=[1, 0]))
2022-10-03 10:38:56.981 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1216:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: Status.SUCCESS
2022-10-03 10:38:56.981 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x1216, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=58), 59, b'\x05A\x11:\x00\x00\x00\x01\x01\x00\x01\x00')
2022-10-03 10:38:56.982 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'34d021a9602a15a44b904b25455493099d4e2791d6c262caecfc6389fc7f3ea7eacd08197e'
2022-10-03 10:38:56.984 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139846300536256] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 96, in async_turn_off
    result = await self._on_off_channel.turn_off()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 350, in turn_off
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
```

javicalle, 2022-10-07:
I have just realized that this part is not needed (already defined in `TuyaSwitc` class):
```python
    def __init__(self, *args, **kwargs):
        """Init device."""
        self.switch_bus = Bus()
        super().__init__(*args, **kwargs)
```

Also replace this part (the `attrid`  --> `1`):
```python
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(SWITCH_EVENT, 1, value)
```

The event listener expects the channel (or endpoint) here:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L497-L507

flaktivated, 2022-10-08:
The above changes have made some progress. The switch is now togglable in the UI without error and get the following error in the logs
```
[0xE3D3:1:0x0006] Sending Tuya Cluster Command... Cluster Command is 1, Arguments are ()
[0xE3D3:1:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Default_Response(command_id=1, status=<Status.SUCCESS: 0>)
[0xE3D3:1:0x0006] no such attribute: 0
[0xE3D3:1:0xef00] Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 39, in _convert_type
    return self.type(value)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 62, in __new__
    n = super().__new__(cls, *args, **kwargs)
TypeError: __int__ returned non-int (type NotImplementedType)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 39, in _convert_type
    return self.type(value)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 115, in __new__
    setattr(instance, name, field._convert_type(value))
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 41, in _convert_type
    raise ValueError(
ValueError: Failed to convert status=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True) from type <class 'zhaquirks.tuya.mcu.TuyaClusterData'> to <class 'zigpy.types.basic.uint8_t'>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 310, in request
    hdr, request = self._create_request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 270, in _create_request
    request = schema(*args, **kwargs)  # type:ignore[operator]
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 115, in __new__
    setattr(instance, name, field._convert_type(value))
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 41, in _convert_type
    raise ValueError(
ValueError: Failed to convert param=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=1, expect_reply=True) from type <class 'zhaquirks.tuya.mcu.TuyaClusterData'> to <class 'zhaquirks.tuya.TuyaManufCluster.Command'>
```

javicalle, 2022-10-08:
It seems that you have a wrong class anywhere. I would bet that your quirk is importing a `zhaquirks.tuya.mcu` class, can be? Maybe some tests with the `TuyaOnOff` from mcu?

flaktivated, 2022-10-23:
I was able to resolve the class issue by stripping away the conflicting changes and had some success with updating the attribute directly via the following screen:

[image]

While I was reading through the class `TuyaOnOff` I noticed the following line:
`            cmd_payload.command_id = TUYA_CMD_BASE + self.endpoint.endpoint_id`

How would I go about setting the value of `self.endpoint.endpoint_id`?

dmulcahey, 2022-10-23:
It’s in your screenshot. Look at the description of the selected cluster.

flaktivated, 2022-10-23:
Thanks for that @dmulcahey switch toggle now works in HA UI. But every toggle I get the following error message in the UI:
[image]

have you seen this before?

HA logs:
```
2022-10-24 00:16:49.759 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140121999293264] 'Status' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 202, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/switch.py", line 97, in async_turn_off
    result = await self._on_off_channel.turn_off()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/general.py", line 350, in turn_off
    if isinstance(result, Exception) or result[1] is not Status.SUCCESS:
TypeError: 'Status' object is not subscriptable
```

javicalle, 2022-10-23:
I believe that there is something wrong here:
https://github.com/zigpy/zha-device-handlers/blob/f2822031a464bfdf766dfccebf667a502622ebbb/zhaquirks/tuya/__init__.py#L532-L534

@flaktivated, do you have the `tuya/__init__.py` file in your local quirk folder?

flaktivated, 2022-10-23:
> 

I don't have that in my local quirk folder - I placed it in there - but it didn't seem to pick it up (wasn't in the `pycache` folder).

javicalle, 2022-10-24:
I believe that the problem has been addressed in the mainstream but you have to wait to the next version or try to implement the changes in your environment (but not easy).
If the rest is working as expected, can you share your version? Or event better create a PR?

flaktivated, 2022-11-01:
Will raise a PR for this device, will need to clean up code. In the interim I got another DIN switch that worked in Z2M and worked on integrating that with ZHA - https://github.com/zigpy/zha-device-handlers/issues/1877

flaktivated, 2022-11-01:
Rejecting support for this device. After getting the energy monitoring configured in HA it does not accurately report energy.
From logs you can see the values go to 0 and then back to measuring.
```
2022-11-02 09:39:43.474 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 92, 0, 0, 125, 0, 0, 30] for attribute 0x0006 (command 0x0002)
2022-11-02 09:39:53.484 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 93, 0, 0, 0, 0, 0, 0] for attribute 0x0006 (command 0x0002)
2022-11-02 09:40:03.499 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 92, 0, 0, 142, 0, 0, 34] for attribute 0x0006 (command 0x0002)
2022-11-02 09:40:13.519 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 92, 0, 1, 24, 0, 0, 67] for attribute 0x0006 (command 0x0002)
2022-11-02 09:40:23.535 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 96, 0, 0, 0, 0, 0, 0] for attribute 0x0006 (command 0x0002)
2022-11-02 09:40:33.554 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 96, 0, 0, 129, 0, 0, 31] for attribute 0x0006 (command 0x0002)
2022-11-02 09:40:53.589 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 87, 0, 0, 0, 0, 0, 0] for attribute 0x0006 (command 0x0002)
2022-11-02 09:41:13.623 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 83, 0, 0, 231, 0, 0, 55] for attribute 0x0006 (command 0x0002)
2022-11-02 09:41:23.640 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 89, 0, 0, 0, 0, 0, 0] for attribute 0x0006 (command 0x0002)
2022-11-02 09:42:43.782 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 66, 0, 0, 172, 0, 0, 41] for attribute 0x0006 (command 0x0002)
2022-11-02 09:42:53.802 DEBUG (MainThread) [zhaquirks.tuya] [0xe1fa:1:0xef00] Received value [9, 75, 0, 0, 0, 0, 0, 0] for attribute 0x0006 (command 0x0002)
```
Sending this device back for a refund. Massive thanks @javicalle for all the learnings shared on this issue!

closed 2022-11-01 by flaktivated

javicalle, 2022-11-02:
Thanks to you for your work 👍🏻