zigpy/zha-device-handlers #1794: [Device Support Request] Aqara 2 Button (WXKG17LM)

DeadlySin2, 2022-09-29:
I have Aqara E1 switch (WXKG17LM) it can be added to Home Assistant by ZHA integration but i can’t use it because ZHA is not seeing buttons only battery and identity button. 

I am using Convee 2 stick with the latest firmware.

Can you please help with adding it to Home Assistant.

<details>

<summary>{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": [
        "0x0006"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": [
        "0x0006"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.acn004",
  "class": "zigpy.device.Device"
}</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
[... 414 lines cut ...]
    }
  }
}</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</detail
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9670751/home-assistant.log)
s>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-10-02:
That would be my proposal:

<details>
<summary>remote_e1.py</summary>

```python
"""Aqara E1-series wireless remote."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    PowerConfiguration,
)

from zhaquirks.const import (
    ALT_DOUBLE_PRESS,
    ALT_SHORT_PRESS,
    ARGS,
    BUTTON,
    COMMAND,
    COMMAND_OFF,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    RIGHT,
    SHORT_PRESS,
    TRIPLE_PRESS,
)
from zhaquirks.xiaomi import (
    LUMI,
    BasicCluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import (
    COMMAND_1_DOUBLE,
    COMMAND_1_HOLD,
    COMMAND_1_SINGLE,
    COMMAND_1_TRIPLE,
    COMMAND_2_DOUBLE,
    COMMAND_2_HOLD,
    COMMAND_2_SINGLE,
    COMMAND_2_TRIPLE,
    COMMAND_3_DOUBLE,
    COMMAND_3_HOLD,
    COMMAND_3_SINGLE,
    COMMAND_3_TRIPLE,
    MultistateInputCluster,
)
from zhaquirks.xiaomi.aqara.remote_h1 import (
    AqaraRemoteManuSpecificCluster,
    PowerConfigurationClusterH1Remote,
)

BOTH_BUTTONS = "both_buttons"

class RemoteE1DoubleRocker1(XiaomiCustomDevice):
    """Aqara E1 Wireless Remote Double Rocker."""

    manufacturer_id_override = 0x115F
[... 101 lines cut ...]
        # when operation_mode == command
        (ALT_SHORT_PRESS, BUTTON): {COMMAND: COMMAND_TOGGLE, ENDPOINT_ID: 1, ARGS: []},
        (ALT_DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_OFF, ENDPOINT_ID: 1, ARGS: []},
    }
```

</details>

Most of the code adapted from `remote_h1.py`.

Device is supposed to have two `click_mode`s:
```
            # click_mode:
            # 1 means single click mode, which is low latency (50ms) but only sends
            #   single click events.
            # 2 means multiple click mode, which has a slightly higher latency but
            #   supports single/double/triple click and long press.
            # default value after factory reset: 1.
```

According to 'herdsman-converters' this line will be required:
```python
manufacturer_id_override = 0x115F
```
Please, comment the line to confirm if it's needed to work.

DeadlySin2, 2022-10-03:
Thanks for your answer. I can try to check if the code upwords is working, but i need an instraction how to integrate it. Or i need to just remove and repair aqara switch?

DeadlySin2, 2022-10-03:
I have made changes in configurqation.yaml. Made a new folder named "custom_zha_quirks"  and copied "remote_e1.py" there. 

Now i can see the battery status and can see buttons then i'am creating new automatisations. Bu

t:

1. There are no buttons in the card of this device
2. I can see that some events were fired:
LUMI lumi.remote.acn004 Attribute Updated event was fired
11:28:02 
LUMI lumi.remote.acn004 Attribute Updated event was fired
11:28:02 
LUMI lumi.remote.acn004 Remote Button Short Press - Left event was fired
3. I tried to subcribe to "zha_events" but i can't see any of them
4. I have made some automatisations with the buttons by they are not working

 
[zha-a675266e45f90e8007b17de3da803ba4-LUMI lumi.remote.acn004-94e33ad35168ff8fcb3480b59c3baf84.json.txt](https://github.com/zigpy/zha-device-handlers/files/9695934/zha-a675266e45f90e8007b17de3da803ba4-LUMI.lumi.remote.acn004-94e33ad35168ff8fcb3480b59c3baf84.json.txt)

[Zigbee signature.txt](https://github.com/zigpy/zha-device-handlers/files/9695943/Zigbee.signature.txt)

javicalle, 2022-10-06:
> 1. There are no buttons in the card of this device

I'd say that's normal behavior if it's a remote: there's no switch and the keystrokes are handled via events.

> 2\. only single button press can fire events if a try to make long button press or double button press i recieve "single button press" message in event section

That's exactly what is explained here:

> Device is supposed to have two `click_mode`s:
>
> ```
>             # click_mode:
>             # 1 means single click mode, which is low latency (50ms) but only sends
>             #   single click [...]

Try changing the value of the `click_mode` attribute to `2`.
You can do this from the device view --> manage cluster --> select the `AqaraRemoteManuSpecificCluster` cluster --> select the `click_mode` attribute

First get the current value, and then try to update the value to `2`.

DeadlySin2, 2022-10-08:
Thanks. I tried to set the new value for click_mode but i failed even to recieve the state value. 
  
[image]

javicalle, 2022-10-08:
Any errors from the log?

DeadlySin2, 2022-10-09:
Yes, there are some. I have attachet log files 
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/9741592/home-assistant.log)
[home-assistant1.log](https://github.com/zigpy/zha-device-handlers/files/9741594/home-assistant1.log)

javicalle, 2022-10-09:
Users are reporting problems for this device in the setting of attributes:
> What I had to do when getting or setting attributes is first click the switch 5 times fast, then click either the set or get attribute button. This was the only way my switch would accept this. Don't know for sure if this helps, but it didn't work for me if I first clicked set or get, then woke up the switch by clicking 5 times fast. Also it only accepted the get and set for a really short time after I clicked the 5 times, so I would click 5 times fast immediately followed by clicking the get or set on my computer.

Refs:
* https://github.com/zigpy/zha-device-handlers/issues/940#issuecomment-1172558963

DeadlySin2, 2022-10-09:
Thanks. Everything working now as expected. Thanks for your support.

javicalle, 2022-10-09:
Just to be sure. What was the problem? The setting value procedure?
Are all the events working? From both buttons? And the both button event too?

Thanks in advanced.

DeadlySin2, 2022-10-10:
Setting new value procedure is possible after clicking the button 5 times. Them the value can be read and written. 
All events are working with write and left buttons, i was able to get Short press button event, but wasn’t able to bet long press and double press long button events.

javicalle, 2022-10-10:
Have you tried with the `operation_mode=1` (event)? 
If not working, maybe listening to events can give us some info.

DeadlySin2, 2022-10-11:
Hello. I tried to set operation_mode to 1, but it doesn’t work. After i’m setting up a value 1 and again reading the value i get operation_mode=o.
Both buttons are only working in a short press mode, if i pressing them for a long time i receive long button press from right or left button
[image]

DeadlySin2, 2022-10-11:
[image]

javicalle, 2022-10-11:
Can you try to listen to `zha_event`s?
That can be done from the developers view panel, events tab. Activate the `zha_event` listening and try the buttons presses.

DeadlySin2, 2022-10-12:
i have attached log from zha_events, but their is no double and long clicks on 3 button
[zha event.txt](https://github.com/zigpy/zha-device-handlers/files/9763802/zha.event.txt)

javicalle, 2022-10-12:
It seems nothing more can be done here. Args from command are the same, so it is not possible to distinguish.

closed 2022-10-22 by dmulcahey

johanstekelenburg, 2023-10-14:
Hello everyone,

I have two Aqara E1 switch double rocker (WXKG17LM) and one Aqara E1 switch single rocker (WXKG16LM). They can be paired with ZHA. For every switch, I have the same problem. When I press a rocker, no event is fired to the device's log. I have checked zha_events via dev tools, where no events are registered when any rocker is pressed. One exception is when the button is pressed 5 times fast. Then the device log shows: "[device name] Attribute Updated event was fired". I have followed the instructions mentioned above and have reset and added the devices multiple times. I have put the quirk code in the same .py file as the other quirk that I already have (/config/custom_zha_quirks/lidl_button.py). I have a working custom quirk for my Lidl buttons that I configured in the past. Therefore I also already had the same lines added to my configuration.yaml that were required. Putting the code in a separate .py file was not successful either. Using only the e1.py quirk while deleting the Lidl quirk did not change anything. I was able to read each click_mode, which was 1. I was not able to change it to 2. I have tried to get the events fired for multiple days by trying multiple things, but up to this moment, no success. Some idea's that I have left:
- Update the devices filmware OTA via Zigbee2MQTT. I have no experience how to do this. Update devices via the Aquara app is no option for me since I do not have an Aquara supported hub
- Update my Sonoff zigbee-stick 3.0 usb plus-p filmware. I am not sure if this will solve the problem and to me it feels like a risky task to execute

I hope you have any ideas on how to solve this problem. I have put quite some hours into this and can not figure it out. Excuse me for the format of this comment, it is my first post. Thank you in advance!

PR #1820 (2022-10-22): New remote_e1 quirks file
Similar to the existing `remote_h1.py`.

Fix: #1809 
Fix: #1794