zigpy/zha-device-handlers #1799: [Device Support Request] [Device UPDATE] Woox TS0202 motion sensor

fantnhu, 2022-10-01:
**Is your feature request related to a problem? Please describe.**
I have several Woox TS0202 motion sensors. I recently bought new ones.
The new ones don't work well, it doesn't detect movement and a door opening sensor also appeared on this device.
I noticed that the old ones (which work well) are all by the manufacturer: _TYZB01_2b8f6cio
The manufacturer of the new ones is different: _TZ3000_sr0vaafi
However, the device type is the same: TS0202

**Describe the solution you'd like**
The old and new signatures are different. Please confirm it

<details>

**This is his old signature that works:**
<summary>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_2b8f6cio",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
</summary>

**And this is the new one, it doesn't work:**
<summary>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_sr0vaafi",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
</summary>
```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>
[... 449 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
The new and the old device look the same, there is no difference on the box
[image]
[image]

javicalle, 2022-10-02:
Please, edit the first post and format the device's signatures. If you don't get it, just put all the 'code' between ` ``` ` marks.

Said that, that seems to me a pretty standar zigbee device. Are you sure that is not a pairing problem?
Battery powered devices often need to be kept 'awake' during pairing. To do this, it is necessary that during the time that the pairing of the device lasts, you briefly press the pairing button.

fantnhu, 2022-10-03:
> Please, edit the first post and format the device's signatures. If you don't get it, just put all the 'code' between `` ``` `` marks.
>
> Said that, that seems to me a pretty standar zigbee device. Are [...]

Hi! Thanks for the reply. I did, ``` marked it (hope it's good now).
I have tried to re-pair the devices many times, but those with a different manufacturer do not work.
The opening entity always appears and the motion detection does not work well either.
The movement: 'Ias_zone' this works for the old one.
For the new one: 'Iaszone'
I don't know if that matters. Thanks for the help!

javicalle, 2022-10-07:
it's still hard to read. Please, edit again and remove any `<detail>` and/or `<summary>` tag from the post (keep the ` ``` `).
It's strange that `Ias_zone` vs `Iaszone`, but I don't think that may be related.

Can you enable the debug logs and attach the relevant info:
* https://www.home-assistant.io/integrations/zha#debug-logging

fballiano, 2023-01-18:
@fantnhu did you solve it? I also have the same TZ3000 and I don't even have the binary sensor exposed in ZHA :-(

closed 2023-07-24 by github-actions[bot]