zigpy/zha-device-handlers #1801: [BUG] AQARA Roller Shutter ZHAQuirk not recognised

JoostGadellaa, 2022-10-02:
Copied from https://github.com/home-assistant/core/issues/70837 since I have the same problem and it seems like it wasn't reported here.

**Describe the bug**
The device pairs, but the zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ is not applied

**To Reproduce**
Pair the device

**Expected behavior**
The device should be matched to the quick 

**Screenshots**
[image]
[image]
[image]

`2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.curtain.acn002 (54:ef:44:10:00:32:90:d1)
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ'>
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint

....
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.curtain.acn002 (54:ef:44:10:00:33:ac:14)
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ'>
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint

....

2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.curtain.acn002 (54:ef:44:10:00:3e:35:9d)
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ'>
2022-04-27 10:21:49 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint

...
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.curtain.acn002 (54:ef:44:10:00:32:90:d1)
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ'>
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint

...

2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.curtain.acn002 (54:ef:44:10:00:33:ac:14)
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ'>
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint

...

2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.curtain.acn002 (54:ef:44:10:00:3e:35:9d)
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ'>
2022-04-27 10:29:07 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint`

**Additional context**
Issues with this device are often caused by outdated ConBeeII firmware, but mine is up to date with https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26780700.bin.GCF

javicalle, 2022-10-02:
Enable the local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Copy inside the `custom_zha_quirks` the current `roller_curtain_e1.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/roller_curtain_e1.py

Add at the end of file this quirk:
```python

class RollerE1AQ_2(RollerE1AQ):
    """Aqara Roller Shade Driver E1 (version 2) device."""

    signature = {
        MODELS_INFO: [(LUMI, "lumi.curtain.acn002")],
        ENDPOINTS: {
            # <SizePrefixedSimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=1
            # input_clusters=[0, 2, 3, 4, 5, 6, 9, 13, 19, 258]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Alarms.cluster_id,
                    AnalogOutput.cluster_id,
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Groups.cluster_id,
                    Identify.cluster_id,
                    MultistateOutput.cluster_id,
                    OnOff.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    Time.cluster_id,
                ],
            },
            # <SizePrefixedSimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0,
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }

```

Restart HA and pair the device again.

JoostGadellaa, 2022-10-02:
That fixed the problem!

closed 2022-10-03 by dmulcahey

javicalle, 2022-10-03:
The quirk will be available in the next HA version.
Remove your local quirk once updated.

aperezva, 2023-07-14:
Hi

i have 6 aqara roller shutter and sometimes more of them don’t react to my cover open o closed instructions but reports correctly temperature and battery. Also if I add again in the network without remove them start to works for a few hours. It’s not a coverage problem because I have a enough lqi. 

Any idea how to fix this problem ?

aperezva, 2023-07-14:
Logger: homeassistant.components.websocket_api.http.connection
Source: components/zha/cover.py:162 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: 01:21:42 (14 occurrences) 
Last logged: 06:27:43

[140153997668176]
[140153078588432]
[140152972252240]
[140153228200336]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 310, in request
    return await req.result
           ^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 205, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1965, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2005, in _execute_service
    return await cast(
           ^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 848, in entity_service_call
    response_data = task.result()  # pop exception if have
                    ^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1192, in async_request_call
    return await coro
           ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 162, in async_stop_cover
    res = await self._cover_cluster_handler.stop()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 132, in retry
    return await func()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/xiaomi/aqara/roller_curtain_e1.py", line 133, in command
    (res,) = await self.endpoint.multistate_output.write_attributes(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 618, in write_attributes
    return await self.write_attributes_raw(attrs, manufacturer)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 624, in write_attributes_raw
    result = await self._write_attributes(attrs, manufacturer=manufacturer)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/quirks/__init__.py", line 321, in _write_attributes
    return await super()._write_attributes(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 375, in request
    return await self._endpoint.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request
    return await self.device.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 309, in request
    async with asyncio_timeout(timeout):
  File "/usr/local/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

PR #1803 (2022-10-03): New signature for Aqara roller shade driver E1
New device signature for the Aqara roller shade driver E1. 
The new quirk extends the existing `RollerE1AQ` and just overwrite the signature.

Fix: #1801