zigpy/zha-device-handlers #1805: [Device Support Request]

alexisiglesias1, 2022-10-04:
**Is your feature request related to a problem? Please describe.**
i bought a couple of iris v1 keyfob 903, i was thinking i could add them to zha, but they are not supported i did some research and notice on hubitat forums they managed included them, i can provide a test device to the developers since i alredy have a few. 

**Describe the solution you'd like**
i would like to use this device as an arrival sensor since the v2 version works like that but with less battery, thats why i realle would like this one work
<details>

<summary>Device signature</summary>
since zha only can see but not add the device this is the only iinfo i have : 
IEEE: 00:0d:6f:00:02:3a:d9:6b
NWK: 0x5a34

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```
Device 0x5a34 (00:0d:6f:00:02:3a:d9:6b) joined the network
Device 00:0d:6f:00:02:3a:d9:6b changed id (0xa28d => 0x5a34)
[0x5a34] Scheduling initialization
[0x5a34] Canceling old initialize call
Tries remaining: 3
[0x5a34] Requesting 'Node Descriptor'
Tries remaining: 2
[0x5a34] Extending timeout for 0xd8 request
</details>

<details>
## System Information

version | core-2022.9.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | America/Chicago
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.09.1
agent_version | 1.3.0
docker_version | 20.10.17
disk_total | 28.5 GB
disk_used | 6.1 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.1.74), Studio Code Server (5.4.0), File editor (5.4.1), Samba share (10.0.0), Plex Media Server (3.1.1), Portainer (2022.9.0), Terminal & SSH (9.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 28, 2022 at 2:11 PM
-- | --
current_recorder_run | September 30, 2022 at 10:23 AM
estimated_db_size | 53.86 MiB
database_engine | sqlite
database_version | 3.38.5

</details>
<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-10-04:
Without device signature or any other info like pairing logs there's no much that can be done.
That is all the pairin log that the device generate?

alexisiglesias1, 2022-10-04:
i was looking for more log this is the only one i could find 
2022-10-04 13:27:43.320 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x4A8C, SrcIEEE=00:0d:6f:00:02:3a:d9:6b, ParentNwk=0x0000)

alexisiglesias1, 2022-10-04:
this is a post where they describe the process to include them through Xbee 
http://www.desert-home.com/2015/05/hacking-into-iris-door-sensor-part-3.html 
this is far complex for my understanding, but i thin it has a lot of useful information

alexisiglesias1, 2022-10-04:
and this is the integration used by hubitat https://raw.githubusercontent.com/birdslikewires/hubitat/master/alertme/drivers/alertme_fob.groovy

javicalle, 2022-10-07:
And there is some info about getting the info requested:
* https://www.home-assistant.io/integrations/zha#reporting-issues
* https://github.com/zigpy/zha-device-handlers#contribution-guidelines

alexisiglesias1, 2022-10-07:
Thanks @javicalle i will try to get as much info as I can with the debug, im really hoping this can be included I think it's a good device

alexisiglesias1, 2022-10-07:
debug logs:
2022-10-07 18:35:14.383 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0xE272, SrcIEEE=00:0d:6f:00:02:3a:d9:6b, ParentNwk=0x0000)
2022-10-07 18:35:14.384 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0xE272, SrcIEEE=00:0d:6f:00:02:3a:d9:6b, ParentNwk=0x0000)
2022-10-07 18:35:14.385 INFO (MainThread) [zigpy.application] Device 0xe272 (00:0d:6f:00:02:3a:d9:6b) joined the network
2022-10-07 18:35:14.386 DEBUG (MainThread) [zigpy.application] Device 00:0d:6f:00:02:3a:d9:6b changed id (0xced4 => 0xe272)
2022-10-07 18:35:14.388 DEBUG (MainThread) [zigpy.device] [0xe272] Scheduling initialization
2022-10-07 18:35:14.390 DEBUG (MainThread) [zigpy.device] [0xe272] Canceling old initialize call
2022-10-07 18:35:14.394 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2022-10-07 18:35:14.395 INFO (MainThread) [zigpy.device] [0xe272] Requesting 'Node Descriptor'
2022-10-07 18:35:14.396 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2022-10-07 18:35:14.399 DEBUG (MainThread) [zigpy.device] [0xe272] Extending timeout for 0x28 request
2022-10-07 18:35:14.400 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE272), dst_ep=0, source_route=None, extended_timeout=True, tsn=40, profile_id=0, cluster_id=<ZDOCmd.Node_Desc_req: 0x0002>, data=Serialized[b'(r\xe2'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-10-07 18:35:14.403 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0xE272, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
2022-10-07 18:35:14.419 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.SUCCESS: 0>)
2022-10-07 18:35:14.422 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0xE272, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2022-10-07 18:35:14.435 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2022-10-07 18:35:14.436 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xE272), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=2, TSN=40, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x28\x72\xE2')
2022-10-07 18:35:14.451 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)

javicalle, 2022-10-08:
And the device signature?
Any info about the device? What do you expect to have or what is not working?
Are these logs related with the device interaction?

alexisiglesias1, 2022-10-08:
Yes these logs are about the device and I can't get the signature device, because I tried to pair to zha, which is able to find it, zha doesn't add it, that's why I was offering send a sample o the device to one of the developers because I don't have enough knowledge, I dm @dmulcahey but I think he hasn't read my mail, I really appreciate ir you are looking on this @javicalle

javicalle, 2022-10-08:
I see the problem.
Have you tried to keep awake the device during the pairing process. Usually **short pushing** the button or whatever you do to pair the device will work.
Attach all the pairing logs from device.

alexisiglesias1, 2022-10-11:
I tried pushing the buttons on different combinations but nothing still not able to add it
this is the complete log:
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=4, profile_id=0, cluster_id=32822, data=Serialized[b'\x04\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0x8F75](ZB-CL01): Device seen - marking the device available and resetting counter
[0x8F75](ZB-CL01): Update device availability -  device available: True - new availability: True - changed: False
[0x334F](lumi.sensor_motion.aq2): Device seen - marking the device available and resetting counter
[0x334F](lumi.sensor_motion.aq2): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9E1D), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1794, data=Serialized[b'\x18}\n\x00\x00%\x87\xa3\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=138, rssi=None)
[0x9E1D:1:0x0702] Received ZCL frame: b'\x18}\n\x00\x00%\x87\xa3\x00\x00\x00\x00'
[0x9E1D:1:0x0702] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=125, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x9E1D:1:0x0702] Decoded ZCL frame: Metering:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=41863))])
[0x9E1D:1:0x0702] Received command 0x0A (TSN 125): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=41863))])
[0x9E1D:1:0x0702] Attribute report received: current_summ_delivered=41863
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x44CC), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=32, data=Serialized[b'\x19o\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=138, rssi=None)
[0x44CC:2:0x0020] Received ZCL frame: b'\x19o\x00'
[0x44CC:2:0x0020] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=111, command_id=0, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x44CC:2:0x0020] Decoded ZCL frame: PollControl:checkin()
[0x44CC:2:0x0020] Received command 0x00 (TSN 111): checkin()
[0x44CC:2:0x0020] No explicit handler for cluster command 0x00: checkin()
[0x44CC:2:0x0020]: Received 111 tsn command 'checkin': checkin()
[0x44CC:2:0x0020] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=111, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x44CC:2:0x0020] Sending request: checkin_response(start_fast_polling=<Bool.true: 1>, fast_poll_timeout=8)
[0x44cc] Extending timeout for 0x6f request
[0xB963](lumi.remote.b1acn01): Device seen - marking the device available and resetting counter
[0xB963](lumi.remote.b1acn01): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x44CC), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=32, data=Serialized[b'\x18o\x0b\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=134, rssi=None)
[0x44CC:2:0x0020] Received ZCL frame: b'\x18o\x0b\x00\x00'
[0x44CC:2:0x0020] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=111, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x44CC:2:0x0020] Decoded ZCL frame: PollControl:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
[0x44CC:2:0x0020]: executed 'checkin_response' command with args: '(True, 8)' kwargs: '{'tsn': 111}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
[0x44CC:2:0x0020] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_resp [... cut ...] eneral=True, *is_reply=True), tsn=15, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x4213:1:0x0702] Decoded ZCL frame: Metering:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=37729))])
[0x4213:1:0x0702] Received command 0x0A (TSN 15): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=37729))])
[0x4213:1:0x0702] Attribute report received: current_summ_delivered=37729

alexisiglesias1, 2022-10-11:
New device 0x0b34 (00:0d:6f:00:02:3a:d9:6b) joined the network this is the device btw
the device is a key fob with two buttons (iris key fob 903). The v2 version is supported with zha with the same functions but less battery, which is this https://zigbee.blakadder.com/Centralite_3450.htm

mbaran5, 2022-10-18:
The KEY903 uses the same proprietary protocol as all of the other Iris V1 stuff. Looking at the main zigpy repo - I see a 2 year old issue where it wasn’t feasible to add support for the custom zigbee clusters those devices use. 

https://github.com/zigpy/zigpy/issues/132

I’d love to see a custom quirk for it, as I’ve got a boatload of DWS800 sensors that only work today on Hubitat.

alexisiglesias1, 2022-10-18:
this user made some advance https://github.com/zigpy/zigpy/issues/121 but definitely have not idea how to implement it.

alexisiglesias1, 2022-10-18:
This is the iris's source code the published back when they close the platform https://github.com/arcus-smart-home/arcusios

Hedda, 2022-10-19:
> i bought a couple of iris v1 keyfob 903

Is that specific device listed on Zigbee Alliance Certifies Products, Zigbee2MQTT supported devices, or Blakadder's databases?

https://csa-iot.org/csa-iot_products/

https://www.zigbee2mqtt.io/supported-devices/

https://zigbee.blakadder.com/

I only see that Zigbee2MQTT and Blakadder currently list Iris Smart Fob Model 3450-L

https://www.zigbee2mqtt.io/devices/3450-L.html

https://zigbee.blakadder.com/Iris_3450-L.html

Hubitat docs at least looks to have some basic info on how to activate join-mode on IRIS V1 Keyfob KEY903

</body>
</html>

https://docs.hubitat.com/index.php?title=Join_and_Reset_Instructions_v1#IRIS

<html>
<body>

Brand | Device | Product Number | Device Type | Protocol
-- | -- | -- | -- | --
IRIS | V1 Keyfob | KEY903 | Presence Sensor | Zigbee

<p style="margin: 0.5em 0px; line-height: inherit; color: rgb(34, 34, 34); font-family: Helvetica, Arial, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><b>Factory Reset</b></p><ol style="margin: 0.3em 0px 0px 3.2em; padding: 0px; list-style-image: none; color: rgb(34, 34, 34); font-family: Helvetica, Arial, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><li style="margin-bottom: 0.1em;">Insert the supplied access tool into the small opening at the top of the key fob. Gently pry open the case to reveal the battery compartment.</li><li style="margin-bottom: 0.1em;">Remove the battery for at least<span> </span><b>10</b><span> </span>seconds. Reinsert the battery.</li><li style="margin-bottom: 0.1em;">Within<span> </span><b>10</b><span> </span>seconds of reinserting the battery, press either button<span> </span><b>8</b><span> </span>times in a row, once per second. The lights on the key fob should begin to flash.<ul style="list-style-type: disc; margin: 0.3em 0px 0px 1.6em; padding: 0px; list-style-image: url(&quot;data:image/svg+xml,%3Csvg xmlns=%22http://www.w3.org/2000/svg%22 width=%225%22 height=%2213%22%3E %3Ccircle cx=%222.5%22 cy=%229.5%22 r=%222.5%22 fill=%22%2300528c%22/%3E %3C/svg%3E&quot;);"><li style="margin-bottom: 0.1em;"><b>NOTE:</b><span> </span><i>The LED can be dim, if you're in a well lighted area this light could be difficult to see</i>.</li></ul></li></ol><p style="margin: 0.5em 0px; line-height: inherit; color: rgb(34, 34, 34); font-family: Helvetica, Arial, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><b>Join the Zigbee Network</b></p><ol style="margin: 0.3em 0px 0px 3.2em; padding: 0px; list-style-image: none; color: rgb(34, 34, 34); font-family: Helvetica, Arial, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><li style="margin-bottom: 0.1em;">Perform a factory reset before attempting to pair.</li><li style="margin-bottom: 0.1em;">From the Hubitat Web Interface home page, select<span> </span><b>Zigbee</b><span> </span>on the<span> </span><b>Device Discovery</b><span> </span>page to start device discovery.<ul style="list-style-type: disc; margin: 0.3em 0px 0px 1.6em; padding: 0px; list-style-image: url(&quot;data:image/svg+xml,%3Csvg xmlns=%22http://www.w3.org/2000/svg%22 width=%225%22 height=%2213%22%3E %3Ccircle cx=%222.5%22 cy=%229.5%22 r=%222.5%22 fill=%22%2300528c%22/%3E %3C/svg%3E&quot;);"><li style="margin-bottom: 0.1em;">See<span> </span><a rel="nofollow" class="external text" href="https://docs.hubitat.com/Discovering%20your%20Devices" style="text-decoration: none; color: rgb(51, 102, 187); background: linear-gradient(transparent, transparent) right center no-repeat, url(&quot;data:image/svg+xml,%3Csvg xmlns=%22http://www.w3.org/2000/svg%22 width=%2212%22 height=%2212%22%3E %3Cpath fill=%22%23fff%22 stroke=%22%2336c%22 d=%22M1.5 4.518h5.982V10.5H1.5z%22/%3E %3Cpath fill=%22%2336c%22 d=%22M5.765 1H11v5.39L9.427 7.937l-1.31-1.31L5.393 9.35l-2.69-2.688 2.81-2.808L4.2 2.544z%22/%3E %3Cpath fill=%22%23fff%22 d=%22M9.995 2.004l.022 4.885L8.2 5.07 5.32 7.95 4.09 6.723l2.882-2.88-1.85-1.852z%22/%3E %3C/svg%3E&quot;);  [... cut ...] flash.<ul style="list-style-type: disc; margin: 0.3em 0px 0px 1.6em; padding: 0px; list-style-image: url(&quot;data:image/svg+xml,%3Csvg xmlns=%22http://www.w3.org/2000/svg%22 width=%225%22 height=%2213%22%3E %3Ccircle cx=%222.5%22 cy=%229.5%22 r=%222.5%22 fill=%22%2300528c%22/%3E %3C/svg%3E&quot;);"><li style="margin-bottom: 0.1em;"><b>NOTE:</b><span> </span><i>The LED can be dim, if you're in a well lighted area this light could be difficult to see</i>.</li></ul></li></ol>
</body>
</html>Brand

alexisiglesias1, 2022-10-19:
Thanks @Hedda, sadly the device seams that is not part of those database, I tried the method describe by Hubitat but I can't make it join zha. Zha is able to discover the device and tried to make it join but after a couple minutes just stop the searching.

closed 2023-07-10 by github-actions[bot]