zigpy/zha-device-handlers #1808: [Device Support Request] Clarify Tuya smart knob (TS004F) automation events

igorsantos07, 2022-10-05:
**Is your feature request related to a problem? Please describe.**
> I'm a complete noob to ZHA and Zigbee in general, so I'm sorry in advance for stupid stuff I write and I'm open to learning <3

The events available in the automation UI of HA are a bit confusing in regards to this knob. To my understanding, those are the events the device support (from the manual and online reading), and the command in Dev Tools > Events > Listen:
- 3-click: change between "scene" and "dimmer" modes (#1372)
  - `{ command: attribute_updated, args: { attribute_name: switch_mode, value: 0 } }` (dimmer mode)
  - `{ command: attribute_updated, args: { attribute_name: switch_mode, value: 1 } }` (scene mode)
- "scene" (cluster_id: 6) (see possible bug below?):
  - rotate left: `{ command: left }` then 2x `{ command: rotate_type, args: [1], { params: rotate_type: 1 } }`
  - rotate right: `{ command: right }` then 2x `{ command: rotate_type, args: [0], { params: rotate_type: 0 } }`
  - 1-click: `{ command: press_type, args: [0], params: { press_type: 0 } }`
  - 2-click: `{ command: press_type, args: [1], params: { press_type: 1 } }`
  - long-click, after release: `{ command: press_type, args: [2], params: { press_type: 2 } }`
- "dimmer" (cluster_id: 8):
  - 1-click: `{ command: toggle }`
  - 2-click: _does nothing_
  - long-click 3s (as in the manual): _does nothing_
  - long-click, instantly around 4s: `{ command: move_hue, args: [1,15], params: { move_mode: 1, rate: 15 }` with various arguments, right after `move_saturation` with similar args: 1, 200
  - long-click, after 4s and releasing: `{ command: stop_move_step }`
  - rotate left: `{ command: step, args: [same as params], params: { step_mode: 1, step_size, transition_time }`
  - rotate right: `{ command: step, args: [same as params], params: { step_mode: 0, step_size, transition_time }`
  - press+rotate left: `{ command: step_color_temp, args: [same as params], params: { step_mode: 3, step_size, transition_time, color_temp_min_mireds, color_temp_max_mireds }`
  - press+rotate right: `{ command: step_color_temp, args: [same as params], params: { step_mode: 1, step_size, transition_time, color_temp_min_mireds, color_temp_max_mireds }`

I see that transition_time means rotating...
1. slow
2. average
3. fast

I also see that step_size is 13 for 1 "click", 25 for 2, 37 for 3... As if one step was rounded up from 12.3333. It seems to have 20 steps if you do a complete rotation?

**_Some extra behavior I cannot understand (bugs?)_**
- while testing the actions above, after (apparently) some test to check the long-click scene time, it started firing `command: remote_button_short_press` and then three repeated events of whatever click I do (1/2/long). Not sure if that was the default behavior and it took me a while to notice, though? Nonetheless... that's weird?

**Describe the solution you'd like**

Things that could be improved:
- many triggers in the Automation screen are confusing, like:
  - `"button" button pressed` vs `"First button" button pressed`
  - `"button" button released after long press` vs `"First button" released after long press`
  - I guess "First button" is the scene button, given the plain "button" doesn't have double clicked?
  - some events use "click" and others use "press", this could be unified
  - left/right device_rotated_slow/device_rotated_fast: these mix snake_case and "humanized" text
  - Device rotated right/left: I guess these are for scene mode too, since they do not identify speed?
  - unclear what is "'button' button continuously pressed". I guess these are triggered as soon as move_hue/saturation are fired?
  - **bottom-line**: it would be waaaay clearer if those triggers were named after the mode they're in, but I wonder if they're that way because they reuse strings for other events?

Also:
- there are no events for press+rotate in dimmer mode
- also, no events for average-speed rotation, although I guess it's complicated to identify between 1/2/3, so only 1/3 got implemented?
- it would be great of there was a way to configure debounce, since sometimes you can fire some fast rotation and then finish with slow rotation, etc.

**Bonus questions**
- is there a way to use the step_size to decide what action to take, or what do do in a given action (e.g. change AC temperature by 0.5C for each 13 steps)?
- what is the "identifybutton" entity? That's also available as an automation trigger...

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
[... 402 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0300": {
            "endpoint_attribute": "light_color",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
Sorry for the long post, but I figured I could write down what I was trying to understand while trying to report the bug. I couldn't find many reliable info on all actions doable on those knobs, only a couple of posts reporting bugs, or the initial implementation happening at #969.

If really nothing can be done as to clarify all those details directly in the HA UI, is there an official place where we can write the relationships between its actions and whatever shows up as automation triggers? Something like it happens when you're writing a Lovelace card and there's a (?) button at the top-right corner, linking to official docs.

MattWestb, 2022-10-07:
Thanks for starting some things out that is not explained in the code or some other easy places then the dev thread that is large.

All the TS004F device default working mode is Dimmer mode and its sending light commands to one light group and we is using `button` for the DA (Device Automatons) then its the default mode.

For the event mode (aka tuya scene mode = not Zigbee scenes only sending events) we have using the `button1` for getting separating them from the command mode.

Scene mode is sending 0, 1 and 2 for clicks and rotating 0 and 1 (as rotate direction) for directions and no more is being sent from the device but we have problems that the device is sending multiple commands and we need sending default response in the wrong direction that i have making one request for but not response from the core devs https://github.com/zigpy/zha-device-handlers/issues/1448.
The event mode log press (>5 sec) is not one real long press its one long press release but then not getting one log press we is using it for that (right or wrong ?).

In command mode is device not sending anything for double click and long press 5 seconds so 4 second is one short press. Then the device is sending many different commands we is using one of them for the DA and you can using the other in automatons if you like but its useless doing DAs for then then its the same function in the end.
The rotate in command mode is very interesting and its possible doing automatons with "slow 3 steps right" but we have not doing it in DA then we have so many DA added for this device and the maintainer was not liking spamming ZHA with DA.
And we is only using "slow" and " medium" rotate "speed" as "rotate slow" and "rotate fast" (for making different form the event mode DA "rotate") and have finding that the fast rotate is not practical using in the system. If you like is the math for the steps in the dev thread https://github.com/zigpy/zha-device-handlers/issues/1322#issuecomment-1028452713.

Also press rotate is working but i was not practical in real and we was not implanting DA for it but you can doing one PR for adding them or making automatons from the events the device is sending.

The command mode long press and release is using color move command for the DA:
https://github.com/zigpy/zha-device-handlers/blob/f13ba18814a5666b30057093c958cdd9d21e404c/zhaquirks/tuya/ts004f.py#L134-L144

The denounce shall being implanted in Zpgpy and ther radio libs and not in the quirk and its needed for other devices that  is "spamming" the network with commands.

"identifybutton" is sending one identify command to the device and is very likely not working with sleeping end device like this if you is not waking it  up so it can recessive the command. If sending it to one light is blinking for some seconds so you can identifying it.

We have and must reusing the code for device automatons in HA and ZHA and trying making the best of it but its not so easy then tuya device is all things but not standard Zigbee devices and i think we have getting most out of the device without spamming the code base (some user like many DA other only using basic function) and you is finding all commands we have found in this thread https://github.com/zigpy/zha-device-handlers/issues/1322#.

If doing one PR and doing braking changes for DA is very likely not being merged then the maintainers done like user getting broken functions in there system.

In the end we is trying making good functions from the commands the device is sending and we cant changing how the device is working then its how tuya have making it in there firmware.

elupus, 2023-01-17:
So i was looking at this device a bit. I have a _TZ3000_ixla93vd

1. Can't we just ditch the slow/fast rotation modes. This knob have many other "speeds" of rotation. I think just left/right rotation is more logical. The device trigger event can still include additional information that can be used in an automation, if you wan't to get into specifics on speeds.

3. Would be nice if it could expose a diagnostic sensor for the mode of the knob. It seem to be part of TuyaZBOnOffAttributeCluster which does not seem to be added to _TZ3000_ixla93vd. So I don't fully understand if this is possible to read on this knob.

MattWestb, 2023-01-17:
1. In scene mode its only reporting right and left rotating. In command mode its sending 4 (or 5) different rotating speed in each direction and only the 2 slowest is useful. So we is having 1 for scene mode and 2 for command mode. Foe making different DA for then is scene mode only rotating and the command mode named fast and slow. If doing only 1 rotating DA then users cant see if its one rotating from scene or command mode so  its little tricky. If like using more speeds for command mode its only doing normal automatons and using the parameter you like but its not so easy for not so technical users so i still think its better leaving the rotate as it is.
2. That wold being great !! PR for implanting it in ZHA is more then welcome then its out of my skills. You can setting the attribute from manager zigbee device from the device card and waking the device up then sending the command to it. If caging the mode on the device is it sending the updated attribute and ZHA is saving the last state. reading it is also possible but then you need waking the device up and normally is the last saved state OK if not having large network problems.
``` 
event_type: zha_event
data:
  device_ieee: 00:3c:84:ff:fe:b1:5b:b3
  unique_id: 00:3c:84:ff:fe:b1:5b:b3:1:0x0006
  device_id: b83611f96747bead3cbb01e3fee2133e
  endpoint_id: 1
  cluster_id: 6
  command: attribute_updated
  args:
    attribute_id: 32772
    attribute_name: switch_mode
    value: 0
  params: {}
origin: LOCAL
time_fired: "2023-01-17T20:45:49.987861+00:00"
context:
  id: 01GQ0QP81347B8NK38J6AM67JW
  parent_id: null
  user_id: null
```

elupus, 2023-01-17:
1. All speeds are useful (steps in rthis case), i use them to control fan speed by dping step of percentage. But i cant use the device trigger since it filters away the speed information. Would be better to not filter on speed at all.
2. I dont seem to be getting that attribute updated event here.

MattWestb, 2023-01-17:
We is not filtering only using 2 of the speed in command mode then its make no sin making DA for all different speeds and steps. Use normal automatons and you can stepping and with all speeds you like its little work but its unlimited.

I think i have seen some device (firmware) is not sending changed attribute but i cant saying it always do or if its somthing else that is making it not sending it.
My  `_TZ3000_4fjiwweb` is doing it for the moment but i think it was some not PRed single knobs that is not doing it or was the user not doing it right.

closed 2023-07-23 by github-actions[bot]