zigpy/zha-device-handlers #1809: [Device Support Request] Aqara E1 Single Rocker Remote Switch - WXKG16LM

bartlomiejmaciejowski, 2022-10-05:
**Is your feature request related to a problem? Please describe.**
The problem is that the device does not trigger zha_event.

**Describe the solution you'd like**
I would like the device to be supported and working :)

<details>

<summary>v</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.acn003",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.9.7",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "run_as_root": true
  },
  "custom_components": {
    "philips_airpurifier_coap": {
      "version": "0.7.0",
      "requirements": [
        "aioairctrl @ git+https://github.com/betaboon/aioairctrl@v0.2.1"
      ]
    },
    "ble_monitor": {
      "version": "10.4.0",
      "requirements": [
        "pycryptodomex>=3.14.1",
[... 325 lines cut ...]
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
2022-10-05 07:25:51.678 DEBUG (MainThread) [zigpy.zcl] [0xF5EB:1:0x0012] Received ZCL frame: b'\x18)\nU\x00!\x01\x00'
2022-10-05 07:25:51.679 DEBUG (MainThread) [zigpy.zcl] [0xF5EB:1:0x0012] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=41, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-10-05 07:25:51.680 DEBUG (MainThread) [zigpy.zcl] [0xF5EB:1:0x0012] Decoded ZCL frame: MultistateInput:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=uint16_t, value=1))])
2022-10-05 07:25:51.680 DEBUG (MainThread) [zigpy.zcl] [0xF5EB:1:0x0012] Received command 0x0A (TSN 41): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=uint16_t, value=1))])
2022-10-05 07:25:51.680 DEBUG (MainThread) [zigpy.zcl] [0xF5EB:1:0x0012] Attribute report received: present_value=1

```

</details>

(the log entries above show when the button is pressed on the device)

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-10-06:
My quick & dirty proposal would be:

```python
"""Aqara E1-series wireless remote."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    PowerConfiguration,
)

from zhaquirks.const import (
    ALT_DOUBLE_PRESS,
    ALT_SHORT_PRESS,
    ARGS,
    BUTTON,
    COMMAND,
    COMMAND_OFF,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    RIGHT,
    SHORT_PRESS,
    TRIPLE_PRESS,
)
from zhaquirks.xiaomi import (
    LUMI,
    BasicCluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import (
    COMMAND_1_DOUBLE,
    COMMAND_1_HOLD,
    COMMAND_1_SINGLE,
    COMMAND_1_TRIPLE,
    COMMAND_2_DOUBLE,
    COMMAND_2_HOLD,
    COMMAND_2_SINGLE,
    COMMAND_2_TRIPLE,
    COMMAND_3_DOUBLE,
    COMMAND_3_HOLD,
    COMMAND_3_SINGLE,
    COMMAND_3_TRIPLE,
    MultistateInputCluster,
)
from zhaquirks.xiaomi.aqara.remote_h1 import (
    AqaraRemoteManuSpecificCluster,
    PowerConfigurationClusterH1Remote,
)

BOTH_BUTTONS = "both_buttons"

class RemoteE1SingleRocker1(XiaomiCustomDevice):
    """Aqara E1 Wireless Remote Double Rocker."""

    manufacturer_id_override = 0x115F

    signature = {
        MODELS_INFO: [(LUMI, "lumi.remote.acn003")],
        ENDPOINTS: {
            1: {
                # SizePrefixedSimpleDescriptor(
                #   endpoint=1, profile=260, device_type=0,
                #   input_clusters=["0x0000", "0x0001", "0x0003", "0x0012", "0xfcc0"],
                #   output_clusters=["0x0003", "0x0006", "0x0019"])
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    MultistateInput.cluster_id,
                    AqaraRemoteManuSpecificCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    BasicCluster,
                    Identify.cluster_id,
                    PowerConfigurationClusterH1Remote,
                    MultistateInputCluster,
                    AqaraRemoteManuSpecificCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                ],
            },
        }
    }
    device_automation_triggers = {
        # triggers when operation_mode == event
        # the button doesn't send an release event after hold
        (SHORT_PRESS, LEFT): {COMMAND: COMMAND_1_SINGLE},
        (DOUBLE_PRESS, LEFT): {COMMAND: COMMAND_1_DOUBLE},
        (TRIPLE_PRESS, LEFT): {COMMAND: COMMAND_1_TRIPLE},
        (LONG_PRESS, LEFT): {COMMAND: COMMAND_1_HOLD},
        # triggers when operation_mode == command
        (ALT_SHORT_PRESS, BUTTON): {COMMAND: COMMAND_TOGGLE, ENDPOINT_ID: 1, ARGS: []},
        (ALT_DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_OFF, ENDPOINT_ID: 1, ARGS: []},
    }
```

Most of the code adapted from `remote_h1.py`.

Device is supposed to have two `click_mode`s:
```
            # click_mode:
            # 1 means single click mode, which is low latency (50ms) but only sends
            #   single click events.
            # 2 means multiple click mode, which has a slightly higher latency but
            #   supports single/double/triple click and long press.
            # default value after factory reset: 1.
```

According to 'herdsman-converters' this line will be required:
```python
manufacturer_id_override = 0x115F
```
Please, comment the line to confirm if it's needed to work.

xardas-eu, 2022-10-07:
hi @javicalle thanks for the reply. I have actually made my own hack based on the H1 version. It looks very similiar to yours, I think maybe even identical.

Problem is that it kinda works. It only supports single click events and instead of "single", the event reads "1_single". I can work with the latter by adjusting my automations but it still would be good to have click_mode=2 working.

Do you have any ideas on how to force it into this mode? Thanks.

javicalle, 2022-10-07:
Yes I think that can be done this way:

From the device view --> manage cluster --> select the `AqaraRemoteManuSpecificCluster` cluster --> select the `click_mode` attribute

First get the current value, and then try to update the value to `2`.

Related to: #1794

bartlomiejmaciejowski, 2022-10-07:
Unfortunately this does not work. I don't have "AqaraRemoteManuSpecificCluster" :(

I only have "ManufacturerSpecificCluster" (twice, endpoints #1 and #2) but both have no attributes to select from.
[image]

could this attribute be set at the quirk level with some command/service call?
thanks!

javicalle, 2022-10-07:
If you don't have the `AqaraRemoteManuSpecificCluster` that means your device hasn't caught the quirk.
But also your device don't have 2 endpoints. At least not in my quirk.

The device should reflect your local quirk in the signature. Can you confirm it?

javicalle, 2022-10-07:
So, now your device shows only 1 enpoint, right?
Can you check if your quirk have the `manufacturer_id_override = 0x115F` line as mine?

Enable the debug logs and attach the relevant info:
* https://www.home-assistant.io/integrations/zha/#debug-logging

xardas-eu, 2022-10-07:
Yep only one endpoint there and I have the override in place as well (I just copied your quirk). I'll go thru the logs later but just so I provide the most useful information - what will be the relevant part? you want the "set zigbee attribute" logged, right?

thanks!

javicalle, 2022-10-07:
Well, any device related info can be usefull, specially the reports from device.
When you remove the power from device and power on again, the device usually send it's status report. This kind of info is valuable.

xardas-eu, 2022-10-07:
Gotcha. I'll post the logs here in the evening. What I suspect is the problem is that the attribute indices may actually be different from H1, e.g.  the click mode is probably not on 0x0125 but some other index. I read somewhere something about 0xff22 so I think I'll try that as well.
Thanks for all the help.

javicalle, 2022-10-07:
I think most of the implementation matches the Z2M.

```javascript
    {
        zigbeeModel: ['lumi.remote.acn003'],
        model: 'WXKG16LM',
        vendor: 'Xiaomi',
        description: 'Aqara wireless remote switch E1 (single rocker)',
        meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
        fromZigbee: [fz.xiaomi_multistate_action, fz.aqara_opple],
        toZigbee: [tz.xiaomi_switch_click_mode],
        exposes: [e.battery(), e.battery_voltage(), e.action(['single', 'double', 'hold']),
            exposes.enum('click_mode', ea.ALL, ['fast', 'multi'])
                .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                    'multi: supports more events like double and hold')],
        configure: async (device, coordinatorEndpoint, logger) => {
            await device.getEndpoint(1).write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
        },
    },
```

where `tz.xiaomi_switch_click_mode` is:
```javascript
    xiaomi_switch_click_mode: {
        key: ['click_mode'],
        convertSet: async (entity, key, value, meta) => {
            const lookupState = {'fast': 0x1, 'multi': 0x02};
            await entity.write('aqaraOpple', {0x0125: {value: lookupState[value], type: 0x20}}, manufacturerOptions.xiaomi);
            return {state: {click_mode: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('aqaraOpple', [0x125], manufacturerOptions.xiaomi);
        },
    },
```

and `manufacturerOptions.xiaomi` are:
```javascript
xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
```

javicalle, 2022-10-10:
It seems that there is a special procedure to setting attribute values:
> What I had to do when getting or setting attributes is first click the switch 5 times fast, then click either the set or get attribute button. This was the only way my switch would accept this. Don't know for sure if this helps, but it didn't work for me if I first clicked set or get, then woke up the switch by clicking 5 times fast. Also it only accepted the get and set for a really short time after I clicked the 5 times, so I would click 5 times fast immediately followed by clicking the get or set on my computer.

Refs:
* https://github.com/zigpy/zha-device-handlers/issues/940#issuecomment-1172558963

xardas-eu, 2022-10-11:
> It seems that there is a special procedure to setting attribute values:

@javicalle YOU ARE THE BEST. I didn't have time to keep investigating this so had to postpone going for the logs for a while but just saw your reply and out of curiosity, tried it.

After clicking the switch 5 times, I was able to read (and then subsequently write) the click_mode attribute and now everything works.

Thanks so much!
Bart

javicalle, 2022-10-11:
> the manufacturer code override was **not** needed for this to work

Thanks for checking it. 👍🏻

Have you need to change the `operation_mode=1` to have double press or hold events?

xardas-eu, 2022-10-11:
No, I just set `click_mode` to 2. Figured the `operation_mode` was only relevant to the switches with a built-in relay.

PR #1820 (2022-10-22): New remote_e1 quirks file
Similar to the existing `remote_h1.py`.

Fix: #1809 
Fix: #1794
closed 2022-10-22 by dmulcahey