zigpy/zha-device-handlers #1818: TZE200_bvu2wnxz TS0601

MS-Turbo, 2022-10-10:
**Is your feature request related to a problem? Please describe.**
I bought 5 ZigBee Thermostatic Radiator Valves but they are not recognized properly in ZHA.
Running ZHA with conbee II stick

**Describe the solution you'd like**
ZHA support for TZE200_bvu2wnxz TS0601

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_bvu2wnxz",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.10.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
    "os_version": "5.15.67",
    "supervisor": "2022.10.0",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "fullykiosk": {
      "version": "1.1.0",
      "requirements": [
        "python-fullykiosk==0.0.11"
      ]
    },
    "scheduler": {
[... 272 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Link to device:
https://www.alibaba.com/product-detail/Tuya-Version-Wi-Fi-ZIGBEE-Smart_1600398708828.html
https://www.aliexpress.com/i/1005003468882163.html

javicalle, 2022-10-12:
The signature can match to any of that quirks:
* SiterwellGS361_Type2
* MoesHY368_Type1
* _TZE200_b6wax7g0
* ZonnsmartTV01_ZG
* Thermostat_TZE200_c88teujp

MattWestb, 2022-10-12:
All is using the same tuya Zigbe-MCU modules (_TYST11 / _TZE200) and cant only being identified by the design if the manufacture not have doing one change of the MCU firmware.

From my first point of view is one clone of the: 
<html>
<body>

Saswell | SEA802 | _TZE200_c88teujp | TS0601 | ✅ Sas I 🥇 | 576
-- | -- | -- | -- | -- | --

</body>

Also Z2M was adding it in april as one Saswell and deleting it in august for adding it agen for 10 hours as one Zonnsmart.
<html>
<body>

Zonnsmart | TV01-ZG | _TZE200_e9ba97vf | TS0601 | ✅ ZS | 983
-- | -- | -- | -- | -- | --

</body>

Must testing witch is working OK but i think the last is the right  the the Sas is little old and not updated and the new have little updated hardware design.

MattWestb, 2022-10-12:
Also 2 different box names for it  around for it AVATTO TRV06 (Z2M) and MYUET ME167 (DE).

MS-Turbo, 2022-10-12:
Thanx for the update! Is there a step by step user guide on how to implement this an easy way?

javicalle, 2022-10-12:
There is a good guide to create your local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

You need to copy the `ts0601_trv.py` file to your local zha_quirk folder and edit the content:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py

At the `ZonnsmartTV01_ZG` edit the `MODELS_INFO` adding your device:
```python
        MODELS_INFO: [
            ("_TZE200_7yoranx2", "TS0601"),  # MOES TRV TV01 ZTRV-ZX-TV01-MS
            ("_TZE200_bvu2wnxz", "TS0601"),
            ("_TZE200_e9ba97vf", "TS0601"),
            ("_TZE200_hue3yfsn", "TS0601"),  # TV02-ZG
            ("_TZE200_husqqvux", "TS0601"),
            ("_TZE200_kly8gjlz", "TS0601"),
            ("_TZE200_lnbfnyxd", "TS0601"),
            ("_TZE200_mudxchsu", "TS0601"),
        ],
```

Restart HA, and pair your device again.

MS-Turbo, 2022-10-12:
Thanx :)

MS-Turbo, 2022-10-12:
Followed the guides and added my device ("_TZE200_bvu2wnxz", "TS0601"), under ZonnsmartTV01_ZG and models_info.
I dont think my TVR is using the same as class as the ZonnsmartTV01_ZG (se picture)
Any other suggested devices I can try adding my model under MODELS_INFO?
[image]

MattWestb, 2022-10-12:
First configure local quirks for ZHA in HA config: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

Then copy one quirk file in the folder and adding your device ID in the signature (one new line but best copy one working one and editing it then no extra spaces or other things is braking the files format). 

ZONNSMART TV01-ZG "type" is here
https://github.com/zigpy/zha-device-handlers/blob/687adffe438c4baf2678e248c521f78dc8c5686c/zhaquirks/tuya/ts0601_trv.py#L1676-L1686

SAS is here:
https://github.com/zigpy/zha-device-handlers/blob/687adffe438c4baf2678e248c521f78dc8c5686c/zhaquirks/tuya/ts0601_trv_sas.py#L233-L240

Only having one local quirk with your device ID in the folder so you is knowing witch is the working one !!

ZONNSMART is needing little more editing getting all working in ZHA but you is getting one standard TRV and it shall working if its the right one.
Restart HA and look if the system is loading the quirk.

Also adding little debug logging is great help if getting problems.

MattWestb, 2022-10-12:
I was too long at the doctor with me bad ear and was missing the train :-((((

MattWestb, 2022-10-12:
@MS-Turbo Then is being paired and is working so you is getting the battery and / or the thermostat OK taking the battery out and waiting one minute and potting it back and it shall reporting all known commands and attributes.

@javicalle is needing them for matching the right quirk or making one new.

The "train" was not one real one only getting home and commenting this issue.

The ear was emerge OP last evening and i can hearing around 5-10% more then for the after last weekend (Zero) so going better.

And i need more holidays weeks is south Spain !!!!! (hola)

MS-Turbo, 2022-10-12:
Good information to have!

Status: 
I took out the batteries and put them back in. Then the device startet a calibration process, but still I get little data from device. Have i done everything correct?

Se picture of device view and no __pycache__ folder in config
Quirk: ts0601_trv.ZonnsmartTV01_ZG

[image]
[image]

MattWestb, 2022-10-12:
Hola !!

The first gen was not needing repairing and some times ZHA was not loading the quirk and must starting more times (perhaps the bug is away 1.5 year later).

Then having the device up and running also with wrong quirk changing to ZONNSMART and restarting HA and then repower the device and all DP shall coming in the log for you.

MS-Turbo, 2022-10-12:
Tried two restarts and removing device and re-adding it

MattWestb, 2022-10-12:
Add https://www.home-assistant.io/integrations/zha/#debug-logging and restart HA and look in the log if the device is sending somthing interesting then doing changes on it locally.

javicalle, 2022-10-12:
It seems that the working Z2M version could be this one:
* https://github.com/twhittock/avatto_me167/blob/main/me167.js

The DPs seems very diferents from the current Tuya TRV quirks.
That would need a few work to make it operational.

cannabarman, 2022-10-15:
Hi guys,
I have exact same thermostat valves model and I'm struggling with it too. 
Is any of these 2 quirks working for you @MS-Turbo ?
I understand that we have to wait for somebody to integrate working existant js code into our python quirk ?

MS-Turbo, 2022-10-15:
So far, non of the suggested qirks are working adequate

challs, 2022-10-15:
Hello everyone, I just received 6 of these bought from [this listing](https://www.aliexpress.com/i/1005003468882163.html).

@MattWestb [wrote](https://github.com/zigpy/zha-device-handlers/issues/1818#issuecomment-1276434589):

> @MS-Turbo Then is being paired and is working so you is getting the battery and / or the thermostat OK taking the battery out and waiting one minute and potting it back and it shall reporting all known commands and attributes.

I have done this as suggested. Here is the log: [TZE200_bvu2wnxz TS0601reporting.log](https://github.com/zigpy/zha-device-handlers/files/9793189/TZE200_bvu2wnxz.TS0601reporting.log)

challs, 2022-10-15:
Set point temperature in degrees C, multiplied by 10. Minimum is 5 degrees:
```
[zhaquirks.tuya] [0x3835:1:0xef00] Received value [0, 0, 0, 250] for attribute 0x0204 (command 0x0002)
```

Current temperature * 10:
```log
[zhaquirks.tuya] [0x3835:1:0xef00] Received value [0, 0, 0, 210] for attribute 0x0205 (command 0x0002)
```

Mode (probably 0=auto, 1=?, 2=off):
```log
[zhaquirks.tuya] [0x3835:1:0xef00] Received value [1] for attribute 0x0402 (command 0x0002)
```

challs, 2022-10-16:
> It seems that the working Z2M version could be this one:
> 
>     * https://github.com/twhittock/avatto_me167/blob/main/me167.js
> 
> 
> The DPs seems very diferents from the current Tuya TRV quirks. That would need a few work to make it operational.

Actually, it does seem to be the same attribute list. If you convert an individual dataPoint to hex, it corresponds to the lower part of the attribute in hex.

For example,

`me167HeatingSetpoint: 4` -> value [0, 0, 0, 250] for attribute 0x02**04** (command 0x0002)
`me167Schedule1: 28` (28 decimal = 1c hex) -> [zhaquirks.tuya] [0x3835:1:0xef00] Received value [1, 6, 0, 0, 120, 8, 0, 0, 160, 18, 0, 0, 220, 20, 0, 0, 200] for attribute 0x00**1c** (command 0x0002)

challs, 2022-10-16:
I have started an implementation on this branch: https://github.com/challs/zha-device-handlers/tree/ts0601_me167_trv

The current room temperature is correctly sent back to Homeassistant, but the temperature setpoint and mode setting is not yet working properly. Maybe there is someone watching this who might want to comment on the implementation. I see that there are some very different styles of implementation, and it wasn't clear to me which is the current best practice style to use.

jacekk015, 2022-10-16:
Blakadder identified it as below:
https://zigbee.blakadder.com/Avatto_TRV06.html

Above DPs from Z2M and @challs help should be sufficient.
Will do the quirk probably tomorrow.

Only question is:
    me167FrostGuard: 36,
    me167AntiScaling: 39,
Are they switches to enable function or should they be just binary sensors??

0x0402
Mode (probably 0=auto, 1=?, 2=off):
Someone need to check which is which.
Z2M says:
0 - schedule
1 - manual
2 - off

javicalle, 2022-10-17:
> the temperature setpoint and mode setting is not yet working properly

I'm not familiar with all the TRV implementations, but I believe that first you need to uncomment/implement the `def map_attribute(self, attribute, value):` part. Add logs to validate that is working your code.
If this part is working as expected, and the reading part is working but none of the 'write' part, can be a `NO_MANUFACTURER_ID` issue.
If it was the case I would overwrite the `write_attributes(self, attributes, manufacturer=None):` method with something like:
```python
    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )
```

challs, 2022-10-17:
> 0x0402 Mode (probably 0=auto, 1=?, 2=off): Someone need to check which is which. Z2M says: 0 - schedule 1 - manual 2 - off

[0] auto [1] heat [2] off - according to me167.js.

I'm not sure that the Z2M entry is correct - the [entry](https://zigbee.blakadder.com/Avatto_TRV06.html) for the signature `TZE200_bvu2wnxz` which we know belongs to the me167 says it is an [Avatto TV02](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html), which appears to be a different model.

challs, 2022-10-17:
Looks like the temperature setting part is working after all already; :)

[image]

jacekk015, 2022-10-17:
First version - can contain BUGs
[ts0601_trv_me167.py.zip](https://github.com/zigpy/zha-device-handlers/files/9804860/ts0601_trv_me167.py.zip)

To achieve functionality of Temperature Calibration and Error State[which gives us battery low alarm] debug logs are needed.
HA logs from pairing process and 10 minutes after, not cut anywhere.
We need to match DP 47 -> HEX 2F and DP 35 -> HEX 23 with some attributes.

MS-Turbo, 2022-10-18:
@jacekk015 , I added me167 using your first version above and i got a switch controller that when turned ON it puts the thermostat i child lock mode (CL)

[image]

jacekk015, 2022-10-18:
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```

MS-Turbo, 2022-10-18:
Do you want me to filter out spesific words? Lots of other zigbee stuff in the log..

jacekk015, 2022-10-18:
Better not. I can filter it out easily

MS-Turbo, 2022-10-18:
[10min logs from inclusion prosess.txt](https://github.com/zigpy/zha-device-handlers/files/9813504/10min.logs.from.inclusion.prosess.txt)

MS-Turbo, 2022-10-19:
@jacekk015 , did you get a look at he logs?

jacekk015, 2022-10-20:
That looks like Temperature correction
`2022-10-18 18:40:38.974 DEBUG (MainThread) [zhaquirks.tuya] [0xe757:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x022f (command 0x0002)`

That should be Error state
`2022-10-18 18:40:38.623 DEBUG (MainThread) [zhaquirks.tuya] [0xe757:1:0xef00] Received value [0] for attribute 0x0523 (command 0x0002)`

challs, 2022-10-20:
@jacekk015 how should we collaborate on this? I had already solved some of the problems still in your implementation. I can port them over to your code but we'll need to keep in sync. 

I have [a git branch](https://github.com/zigpy/zha-device-handlers/compare/dev...challs:zha-device-handlers:ts0601_me167_trv) for this so you can see my state there.

challs, 2022-10-20:
I have fixed some problems in the mode and state reporting. [This version](https://raw.githubusercontent.com/zigpy/zha-device-handlers/d9506180eceae37cf5a76c1e398ee85c557d328b/zhaquirks/tuya/ts0601_trv_me167.py) is now able to report temperature, mode and heating state properly.

[ts0601_trv_me167.zip](https://github.com/zigpy/zha-device-handlers/files/9832629/ts0601_trv_me167.zip)

jacekk015, 2022-10-21:
- bugs corrected
- implemented: temperature correction + battery level as empty alarm
[ts0601_trv_me167.py.zip](https://github.com/zigpy/zha-device-handlers/files/9842895/ts0601_trv_me167.py.zip)

cannabarman, 2022-10-22:
I just tested it, everything seems OK, great job guys !!!
Just one thing, battery level is always 0%, is this normal ?

jacekk015, 2022-10-22:
Probably battery line code need to be corrected. That TRV doesn't send battery level, just warns when empty.

MS-Turbo, 2022-10-22:
@challs and @jacekk015, great work!
Is this git branch from challs to be seen as the master from both your work? IT would be very practical for everyone 🙂(https://github.com/zigpy/zha-device-handlers/compare/dev...challs:zha-device-handlers:ts0601_me167_trv)

jacekk015, 2022-10-22:
Battery level corrected
[ts0601_trv_me167.py.zip](https://github.com/zigpy/zha-device-handlers/files/9844315/ts0601_trv_me167.py.zip)

ma4er, 2022-10-22:
Hi, i had the same Problems with missing Entities. Your solution works! THX !
But one thing doesn´t work.
I can turn the power off, it works fine.   But on the other side i can´t turn it on. If i push the Power button, nothing happens.
Anyone else the same Problems?

MS-Turbo, 2022-10-22:
Using the latest file from @jacekk015, this is the how the thermostat is shown. 
Any variations from this in you guys HA view? 
[image]

ma4er, 2022-10-22:
Yes, i´m using the latest file from  @jacekk015
This is my view.
 
[image]

MS-Turbo, 2022-10-22:
Thanx! I see your sensor havac has heating state, but mine has state unknown. I will try to rejoin the device.

MattWestb, 2022-10-22:
Then doing changes in the local quirk is important deleting the  `__pycache__ ` directory before restarting HA so the system is not reusing some old code that is changed / deleted.

jacekk015, 2022-10-22:
As for battery level - could someone check in Device Cluster Management
[image]
Cluster: ME167ManufCluster
Attribute: battery_state
Click Read Attribute
and paste here screenshot??

and also:
[image]
Cluster: TuyaPowerConfigurationCluster
Attribute: battery_percentage_remaining
Click Read Attribute

MS-Turbo, 2022-10-22:
> Then doing changes in the local quirk is important deleting the `__pycache__ ` directory before restarting HA so the system is not reusing some old code that is changed / deleted.

Where can i find the local __pycache__ ? There is no other file in /config/custom_zha_quirks

[image]

challs, 2022-10-22:
@MS-Turbo wrote:
> Is this git branch from challs to be seen as the master from both your work?

@jacekk015 isn't completely working off my branch at the moment but I am attempting to track his changes in there.

I have imported the latest changes from @jacekk015, great work thank you!

For me the battery status was showing in the interface as 127%, so I have added a further commit to change that to 100% instead.

Version with fixed battery percentage:

[ts0601_trv_me167.zip](https://github.com/zigpy/zha-device-handlers/files/9844950/ts0601_trv_me167.zip)

My device appears to expect the power calibration in units of 1, not tenths. With the current version, setting the calibration to +4.9C ends up with the device reporting that the current temperature is 49 degrees higher:

[image]

challs, 2022-10-22:
@MS-Turbo 
> Where can i find the local pycache ? There is no other file in /config/custom_zha_quirks

I don't seem to have a __pycache__ folder under the zha custom folder. I do have one under custom_components though. As far as I can see, the custom zha python files aren't being cached for some reason.

MattWestb, 2022-10-22:
I just tested and its one hidden folder and the HA file editor is not showing it.
If using SSH to the HA you can finding it here `/usr/share/hassio/homeassistant/custom_zha_quirks/__pycache__/`
Also in core  container is mapped `/config/custom_zha_quirks/__pycache__/` if you can assessing the container CLI.

MS-Turbo, 2022-10-22:
> @MS-Turbo wrote:
>
>
> @jacekk015 isn't completely working off my branch at the moment but I am attempting to track his changes in there.
>
> I have imported the latest changes from @jacekk015, great work [...]

I have the same issue when calibration temperature
[image]

jacekk015, 2022-10-22:
@MS-Turbo Logs are needed.
I need to see data of attribute 0x022F
First change it from zero to some value.

MS-Turbo, 2022-10-22:
> @MS-Turbo Logs are needed. I need to see data of attribute 0x022F First change it from zero to some value.

[image]

dmulcahey, 2022-10-22:
@jacekk015 are you going to PR the changes from all of these zip files I see in my notifications?

MS-Turbo, 2022-10-22:
Int the last six lines i adjusted the temperature offset slider in tenths 0,1 - 0,2 - 0,3 in the thermostats 

2022-10-22 22:29:01.433 ERROR (MainThread) [zigpy.zcl] [0x7F37:1:0xef00] Failed to convert attribute 0x022F from -44.0 (<class 'float'>) to type <class 'zigpy.types.basic.uint32_t'>: -44 is not an unsigned 32 bit integer
2022-10-22 22:29:10.207 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x022f (command 0x0002)
2022-10-22 22:29:33.507 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 1] for attribute 0x022f (command 0x0002)
2022-10-22 22:29:51.921 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 9] for attribute 0x022f (command 0x0002)
2022-10-22 22:30:05.302 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 1] for attribute 0x022f (command 0x0002)
2022-10-22 22:32:19.567 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 26] for attribute 0x022f (command 0x0002)
2022-10-22 22:33:13.142 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 1] for attribute 0x022f (command 0x0002)
2022-10-22 23:48:22.026 ERROR (MainThread) [zigpy.zcl] [0xF557:1:0xef00] Failed to convert attribute 0x022F from -24.0 (<class 'float'>) to type <class 'zigpy.types.basic.uint32_t'>: -24 is not an unsigned 32 bit integer
2022-10-22 23:48:29.830 DEBUG (MainThread) [zhaquirks.tuya] [0xf557:1:0xef00] Received value [0, 0, 0, 1] for attribute 0x022f (command 0x0002)
2022-10-22 23:48:45.643 DEBUG (MainThread) [zhaquirks.tuya] [0xf557:1:0xef00] Received value [0, 0, 0, 2] for attribute 0x022f (command 0x0002)
2022-10-22 23:49:32.487 DEBUG (MainThread) [zhaquirks.tuya] [0x7f37:1:0xef00] Received value [0, 0, 0, 2] for attribute 0x022f (command 0x0002)
2022-10-22 23:50:15.353 DEBUG (MainThread) [zhaquirks.tuya] [0x6615:1:0xef00] Received value [0, 0, 0, 1] for attribute 0x022f (command 0x0002)
2022-10-22 23:51:01.002 DEBUG (MainThread) [zhaquirks.tuya] [0x6615:1:0xef00] Received value [0, 0, 0, 2] for attribute 0x022f (command 0x0002)
2022-10-22 23:51:49.547 DEBUG (MainThread) [zhaquirks.tuya] [0x6615:1:0xef00] Received value [0, 0, 0, 3] for attribute 0x022f (command 0x0002)
2022-10-22 23:52:05.405 DEBUG (MainThread) [zhaquirks.tuya] [0xf557:1:0xef00] Received value [0, 0, 0, 3] for attribute 0x022f (command 0x0002)

jacekk015, 2022-10-22:
@MS-Turbo can you, on the TRV device, do correction in tenths or full degrees?
Adjust it on the TRV only and post the logs.

Z2M code looks like they coded only full degrees in calibration
[BecaWindowDetection_bus](https://github.com/twhittock/avatto_me167/blob/main/me167.js#L303)

MS-Turbo, 2022-10-22:
Sorry, but is currently in my cabin in the mountains and not back at the apartment before Tuesday

MattWestb, 2022-10-22:
@dmulcahey Normally in the end its one working quirk then @jacekk015 have getting all things in place but one the road its very much try and error for getting understanding how the TRV /tuya MCU is doing all the things and "translating" it to Zigbee standard.

With brave user and good logs is normally fast done if the device is not doing extreme strange things.

dmulcahey, 2022-10-22:
> @dmulcahey Normally in the end its one working quirk then @jacekk015 have getting all things in place but one the road its very much try and error for getting understanding how the TRV /tuya MCU is [...]

I understand. I just want to make sure we don’t lose all of his work!

cannabarman, 2022-10-23:
> @MS-Turbo can you, on the TRV device, do correction in tenths or full degrees? Adjust it on the TRV only and post the logs.
> 
> Z2M code looks like they coded only full degrees in calibration https://github.com/twhittock/avatto_me167/blob/main/me167.js#L303

I'm not sure we can adjust it directly on the TRV, I tried but without instructions manual, I didn't succeed... However, all adjustments I saw on the TRV are in full degrees.

[128 more replies]

closed 2025-01-16 by TheJulianJES