zigpy/zha-device-handlers #1821: [Device Support Request] TS0001 _TZ3000_x3ewpzyr & TS0002 _TZ3000_irrmjcgi

Der-Nax, 2022-10-12:
**Is your feature request related to a problem? Please describe.**
The problem is that current is in A, devices provide information in mA.

**Describe the solution you'd like**
Please make correction A on mA or divide on 1000 to save in A.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_irrmjcgi",
  "model": "TS0002",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2022-10-18:
Have you tried to modify the `ac_current_divisor (id: 0x0603)` attribute?
From the device view --> Manage clusters --> select cluster `ElectricalMeasurement (Endpoint id: 1, Id: 0x0b04, Type: in)` --> select attribute `ac_current_divisor (id: 0x0603)`

[image]

You can first get the current value and then try to update to x1000. Once updated, get again the value to verify if setted.

Der-Nax, 2022-10-18:
Thank you for idea. Unfortunately it does not work.I tried to set value 1000. Then I try to get value - receive NONE.
-------- Исходное сообщение --------От: javicalle ***@***.***> Дата: 18.10.2022  21:55  (GMT+03:00) Кому: zigpy/zha-device-handlers ***@***.***> Копия: Der-Nax ***@***.***>, Author ***@***.***> Тема: Re: [zigpy/zha-device-handlers] [Device Support Request] TS0001 _TZ3000_x3ewpzyr & TS0002 _TZ3000_irrmjcgi (Issue #1821) 
Have you tried to modify the ac_current_divisor (id: 0x0603) attribute?
From the device view --> Manage clusters --> select cluster ElectricalMeasurement (Endpoint id: 1, Id: 0x0b04, Type: in) --> select attribute ac_current_divisor (id: 0x0603)

You can first get the current value and then try to update to x1000. Once updated, get again the value to verify if setted.

Der-Nax, 2022-10-20:
Thank you! I'll try it on weekend.

Der-Nax, 2022-10-22:
I tested it. It's work! Thank you!
I did't - remove and pair you device again. I make reconfiguration. 
How do I make refresh Energy count?

MattWestb, 2022-10-22:
I think (= not knowing) that resetting the device and repairing it is setting the counters to zero.

Testing and you is not thinking then you is knowing :-))

closed 2022-10-23 by Der-Nax

cloudbr34k84, 2022-11-17:
@javicalle i have this switch in the 2 gang with power monitoring, do i need to change or add anything to this quirk??

cloudbr34k84, 2022-11-17:
@MattWestb how are you going with this device. I'm finding that its not that responsive when using it with HA??

MattWestb, 2022-11-17:
I only having one `TS0004 _TZ3000_ltt60asa` its without power metering but is working OK for plant lights.

cloudbr34k84, 2022-11-17:
> I only having one `TS0004 _TZ3000_ltt60asa` its without power metering but is working OK for plant lights.

Cheers! this quick has been applied to other devices and they are not very responsive atm

PR #1860 (2022-10-25): Add new TS000X metering devices
New quirks for the TS000X metering devices.

Fixes: #1821
Fixes: #1853