zigpy/zha-device-handlers #1822: [BUG] Blitzwolf BW-SHP15 plugs power outage memory.

Charisketi, 2022-10-12:
Blitzwolf BW-SHP15 plugs are in "off" state after a power outage by default. This can be configured to be 'ON' or last known state.
This can be done in z2m ,how can we configure this in ZHA?

javicalle, 2022-10-13:
In ZHA you must edit the device attribute `power_on_state` from `TuyaZBOnOffAttributeCluster` cluster.

You can do it from the device view, 'Manage clusters' option.
Select the `TuyaZBOnOffAttributeCluster` and then select the `power_on_state` attribute.
You can get the current value (probably something like `PowerOnState.off`). 
Just set the desired value as a number 0/1/2:
* https://github.com/zigpy/zha-device-handlers/blob/687adffe438c4baf2678e248c521f78dc8c5686c/zhaquirks/tuya/__init__.py#L869-L875

dmulcahey, 2022-10-13:
We can add configuration entities to zha for this. They are fairly easy to add. See here: https://github.com/home-assistant/core/blob/ca4c4774ca9cf4efc9578cd9b2a9d9bd39ed9b31/homeassistant/components/zha/select.py#L219

javicalle, 2022-10-13:
I know, I know... 
I have look at the code and is a babygarden task. 😓 
I want to think in some approach for the Tuya manufacturer cluster (vs multiple device types, vs multiples optional not always supported attributes)

dmulcahey, 2022-10-13:
ZHA will handle the not supported attributes OOTB. If you wanna talk through it hit me up on discord and I’ll help. (Not tonight though 😀)

Charisketi, 2022-10-13:
@javicalle thank you this fix the issue. Is there a way to have entities with this specific setting? As for example I have with a gledopto device;

[image]

javicalle, 2022-10-15:
> Is there a way to have entities with this specific setting?

Yes, that's what dmulcahey suggested.

As this issue is already resolved, do you mind to close the ticket?

If you want to track the other issue, please open another issue with this topic.

Thanks in advance.

closed 2022-10-15 by Charisketi

javicalle, 2022-11-01:
FYI: https://github.com/home-assistant/core/pull/81311