zigpy/zha-device-handlers #1823: [BUG]Blitzwolf SHP15 not working properly

haralake, 2022-10-13:
A while ago I received 3 Blitzwolf shp15 switches knowing that they are fully compatible with home assistant zha. I have noticed that while they are working properly they are behaving strangely. When I operate them from the same device, their status does not change in the home assistant dashboard. If I handle them from the home assistant, everything works normally.

Device name: TS011F

Device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_u5u4cakc",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

These switches are very well known in the zigbee community and I would like to order more, but this particular problem is preventing me. Is there a way to fix this problem without needing a custom fix?

javicalle, 2022-10-13:
Duplicate of: #1787

javicalle, 2022-10-13:
There are info about a possible fix:
* https://github.com/zigpy/zha-device-handlers/issues/1787#issuecomment-1276845969

yfands, 2022-10-13:
The above fix doesn't work for me... Same Issue, Regards Frank

The strange thing is, why does it show up like TS011F in HA, while the Blitzwolf SHP 13 shows up as TS0121 and is working fine...

javicalle, 2022-10-14:
Let's continue the conversation on #1787

closed 2022-10-14 by javicalle